espressif/esp-idf #2652: Native Ota example not working

Mr-Techtron, 2018-11-01:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ✔] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ✔] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ✔] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for DevKitC): [v1]
- Core (if using chip or module): [ESP-WROOM-32D]
- IDF version : bd90f534f
- Development Env: [Eclipse]
- Operating System:	[Windows]
- Power Supply: [USB]

## Problem Description

//Detailed problem description goes here.
When running the native ota example code in my esp32, it's able to connect and fetch the hello-world.bin file but it fails in esp_ota_end function giving the invalid segment length 0xffffffff error.

Mr-Techtron, 2018-11-01:
D (7252) HTTP_CLIENT: Write header[3]: GET /hello-world.bin HTTP/1.1
User-Agent: ESP32 HTTP Client/1.1
Host: 192.168.10.20
Content-Length: 0

D (7264) HTTP_CLIENT: on_message_begin
D (7265) HTTP_CLIENT: HEADER=Content-type:text/plain
D (7265) HTTP_CLIENT: http_on_headers_complete, status=200, offset=45, nread=45
D (7272) HTTP_CLIENT: http_on_body 467
D (7275) HTTP_CLIENT: content_length = -1
I (7280) native_ota_example: Writing to partition subtype 16 at offset 0x110000
I (7293) native_ota_example: esp_ota_begin succeeded
D (7294) HTTP_CLIENT: is_data_remain=1, is_chunked=1
D (7294) HTTP_CLIENT: need_read=557, byte_to_read=512, rlen=213, ridx=467
D (7300) HTTP_CLIENT: http_on_body 213
D (7304) HTTP_CLIENT: is_data_remain=1, is_chunked=1
E (7310) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
D (7316) HTTP_CLIENT: need_read=344, byte_to_read=344, rlen=0, ridx=680
I (7323) native_ota_example: Written image length 680
D (7327) HTTP_CLIENT: is_data_remain=1, is_chunked=1
E (7332) TRANS_SSL: esp_tls_conn_read error, errno=Socket is not connected
D (7339) HTTP_CLIENT: need_read=1024, byte_to_read=512, rlen=0, ridx=0
I (7346) native_ota_example: Connection closed,all data received
I (7353) native_ota_example: Total Write binary data length : 680
D (7359) esp_image: reading image header @ 0x110000
D (7365) esp_image: image header: 0xe9 0x0c 0x02 0x02 40080ef0
V (7370) esp_image: loading segment header 0 at offset 0x110018
V (7376) esp_image: segment data length 0x7390 data starts 0x110020
D (7381) phy_init: wifi mac time delta: 87016
V (7383) esp_image: segment 0 map_segment 1 segment_data_offs 0x110020 load_addr 0x3f400020
I (7395) esp_image: segment 0: paddr=0x00110020 vaddr=0x3f400020 size=0x07390 ( 29584) map
D (7405) esp_image: free data page_count 0x0000003e
V (7432) esp_image: loading segment header 1 at offset 0x1173b0
V (7433) esp_image: segment data length 0xffffffff data starts 0x1173b8
E (7434) esp_image: invalid segment length 0xffffffff
E (7440) native_ota_example: err:1503, esp_ota_end failed!
E (7450) native_ota_example: Exiting task due to fatal error...

chegewara, 2018-11-01:
Are you sure that you image length is 680 bytes?
`I (7353) native_ota_example: Total Write binary data length : 680`

jitin17, 2018-11-01:
Can you refer this issue #2625? It appears to be similar to yours.

Mr-Techtron, 2018-11-01:
Thank you Jitin, it seems that there was some problem with openssl but with python http server it works flawlessly

closed 2018-11-01 by Mr-Techtron