espressif/esp-idf #2653: ESP32 WIFI_MODE_APSTA failed after sometime while computer is still connected to the SOFT_AP

KrrishnaShah, 2018-11-01:
I'm using my esp32 as Station+Soft AP.
The continuous ping from my computer failed after sometime (random intervals), while my computer is still connected to the esp32 soft_AP. When I disconnect my computer from esp32 soft AP and reconnect it again, it works fine for some time.

KrrishnaShah, 2018-11-01:
/******************************* network.c ********************************/
	#include <stdio.h>
	#include <string.h>

	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"
	#include "freertos/event_groups.h"
	#include "freertos/queue.h"

	#include "esp_system.h"
	#include "esp_err.h"
	#include "esp_event_loop.h"
	#include "esp_event.h"
	#include "esp_attr.h"
	#include "esp_log.h"
	#include "esp_wifi.h"
	#include "rom/ets_sys.h"
	#include "tcpip_adapter.h"
	#include "lwip/dns.h"
	#include "lwip/ip4_addr.h"

	#include "debug.h"

	const int WIFI_CONNECTED_BIT = BIT0;
	static EventGroupHandle_t wifi_event_group;

	extern void wifi_nw_init(void);

	void wifi_wait_for_connection(void);
	static esp_err_t __event_handler(void *ctx, system_event_t *event);

	void network_init(void)
	{
		uint8_t mac[6];

		wifi_event_group = xEventGroupCreate();
		ESP_ERROR_CHECK( esp_event_loop_init(__event_handler, NULL) );

		ESP_ERROR_CHECK( esp_efuse_mac_get_default(mac) );
		ESP_ERROR_CHECK( esp_base_mac_addr_set(mac) );

		tcpip_adapter_init();

		wifi_nw_init();
	}

	/* wifi event handler */
	static esp_err_t __event_handler(void *ctx, system_event_t *event)
	{
		switch(event->event_id) 
		{

	#if 1	/* Stataion mode events */
			case SYSTEM_EVENT_STA_START:
				esp_wifi_connect();
			break;

			case SYSTEM_EVENT_STA_GOT_IP:
				xEventGroupSetBits(wifi_event_group, WIFI_CONNECTED_BIT);
			break;

			case SYSTEM_EVENT_STA_DISCONNECTED:
				esp_wifi_connect();
				xEventGroupClearBits(wifi_event_group, WIFI_CONNECTED_BIT);
			break;
	#endif

	#if 1	/* AP mode events */
			case SYSTEM_EVENT_AP_START:
				trace_imp("Station Connected ...");
			break;

			case SYSTEM_EVENT_AP_STOP:
				trace_imp("Station Connected ...");
			break;

			case SYSTEM_EVENT_AP_STACONNECTED:
				trace_imp("Station Connected ...");
			break;

			case SYSTEM_EVENT_AP_STADISCONNECTED:
				trace_imp("Station Dis-connected ...");
			break;

			case SYSTEM_EVENT_AP_STAIPASSIGNED:
				trace_imp("Station Connected ...");
			break;

			case SYSTEM_EVENT_AP_PROBEREQRECVED:
				trace_imp("Station Connected ...");
			break;
	#endif

			default:
			break;
		}
		return ESP_OK;
	}

	void wifi_wait_for_connection(void)
	{
		xEventGroupWaitBits(wifi_event_group, WIFI_CONNECTED_BIT, false, true, portMAX_DELAY);
	}

/****************************** wifi.c ******************************************/
	#include "esp_system.h"
	#include "esp_wifi.h"
	#include "esp_wifi_types.h"
	#include "esp_event_loop.h"
	#include "tcpip_adapter.h"

	#include "freertos/FreeRTOS.h"
	#include "freertos/task.h"
	#include "freertos/event_groups.h"

	#include "lwip/ip4_addr.h"
	#include "string.h"
	#include "apps/dhcpserver.h"
	#include "apps/dhcpserver_options.h"
	#include "settings.h"

	#include "esp_log.h"

	#if 1
		#define WIFI_SSID   "WSPKR"
		#define WIFI_PASS   "RTSR&DWCS"

		#define STA_STATIC_IPv4_ADDR     "192.168.1.32"
		#define STA_STATIC_IPv4_GW       "192.168.1.1"
		#define STA_STATIC_IPv4_MASK     "255.255.255.0"
	#endif

	#if 1

		#define AP_STATIC_IPv4_ADDR     "172.168.0.1"
		#define AP_STATIC_IPv4_GW       "172.168.0.1"
		#define AP_STATIC_IPv4_MASK     "255.255.0.0"
		#define AP_STATIC_IPv4_DNS      "8.8.8.8"

	#endif

	void wifi_nw_init(void)
	{

		/*  from settings.h
			ip4addr_aton("172.168.1.1", 	&settings_info.network_info.ap.ip.ip);
			ip4addr_aton("255.255.255.0", 	&settings_info.network_info.ap.ip.netmask);
			ip4addr_aton("172.168.1.1",   	&settings_info.network_info.ap.ip.gw);
			ip4addr_aton("8.8.8.8", 		&settings_info.network_info.ap.dns.u_addr.ip4);

			settings_info.network_info.ap.wifi.ap.authmode 			= WIFI_AUTH_WPA2_PSK;
			settings_info.network_info.ap.wifi.ap.beacon_interval	= 100;
			settings_info.network_info.ap.wifi.ap.channel 			= 11;
			settings_info.network_info.ap.wifi.ap.max_connection 	= 4;
			settings_info.network_info.ap.wifi.ap.ssid_hidden 		= 0;
			settings_info.network_info.ap.wifi.ap.ssid_len 			= 0;
		*/
		
		_S_AP_INFO * _ap = settings_get_network_ap_info();

		ESP_ERROR_CHECK( tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP) );
		tcpip_adapter_set_ip_info(TCPIP_ADAPTER_IF_AP, &_ap->ip); //set static IP

		ESP_ERROR_CHECK( tcpip_adapter_dhcps_start(TCPIP_ADAPTER_IF_AP) ); // start dhcp server firstly

		uint8_t dns_offer = 0x02;
		dhcps_set_option_info(DOMAIN_NAME_SERVER, &dns_offer, 1);
		dhcps_dns_setserver(&_ap->dns);

		wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();

		ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
		ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );

		esp_wifi_set_bandwidth(ESP_IF_WIFI_AP, WIFI_BW_HT20);
		wifi_config_t wifi_station_config = {
			.sta = {
				.ssid = "WSPKR",
				.password = "RTSR&DWCS",
			}
		};

		int8_t max_tx_power = 0;

		ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_APSTA) );
		ESP_ERROR_CHECK( esp_wifi_set_ps(WIFI_PS_NONE) );
		ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_station_config) );
		ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_AP, &_ap->wifi) );
		ESP_ERROR_CHECK( esp_wifi_start() );

		ESP_ERROR_CHECK( esp_wifi_get_max_tx_power(&max_tx_power) );
		trace_imp("max TX_POWER: %d", max_tx_power);
	}

GeorgeFlorian, 2019-02-15:
Have you found a fix to this ?

KrrishnaShah, 2019-02-16:
I did not find any core solutions for this issue. 
But running the wifi in only soft AP mode is somehow good ( does not disconnect as frequently as in STA_AP ).

GeorgeFlorian, 2019-02-16:
> I did not find any core solutions for this issue.
> But running the wifi in only soft AP mode is somehow good ( does not disconnect as frequently as in STA_AP ).
 
I understand ! But I need it to behave like a SoftAP, serve a webpage, introduce SSID and Password to access a wireless network, then switch it to a Station and server another page.

liuzfesp, 2019-02-17:
Hi @GeorgeFlorian, need you help to describe:

> The continuous ping from my computer failed after sometime

in more detailed? Does the ping continuously fail, or it just fail one time and recoveries immediately? About how long time to replicate this issue? It will be helpful if you can paste the log here.

Our engineer is now investigating this issue.

GeorgeFlorian, 2019-02-17:
> Hi @GeorgeFlorian, need you help to describe:
>
>
> in more detailed? Does the ping continuously fail, or it just fail one time and recoveries immediately? About how long time to replicate this issue? [...]

I didn't try WIFI_MODE_APSTA. I posted here because I was curious if there was a fix to it, because I wanted to give it a shot.

@KrrishnaShah Is the one having troubles, and he posted this.

KrrishnaShah, 2019-02-18:
> Hi @GeorgeFlorian, need you help to describe:
>
>
> in more detailed? Does the ping continuously fail, or it just fail one time and recoveries immediately? About how long time to replicate this issue? [...]

It runs normally (OK) for sometimes (duration is random) and then ping fails. Once the ping fails the station needs to be reconnected to the SoftAP and then it continuous again. In this case, some times the number of connected station increments (although the same only one station is reconnected).

liuzfesp, 2019-03-18:
HI @KrrishnaShah we have some fixes about WIFI_MODE_APSTA, could you help to check whether this issue still exists in latest release (lastest master/v3.2/v3.1/v3.0 contains the fixes).

KrrishnaShah, 2019-03-19:
I will try it very soon.
Thanks for the update ..

Alvin1Zhang, 2019-07-25:
@KrrishnaShah Thanks for reporting the issue, will close the issue due to short of feedback. Feel free to reopen the issue if it still exists. Thanks.

closed 2019-07-25 by Alvin1Zhang