espressif/esp-idf #2656: esp_https_ota panic

markfink, 2018-11-02:
when I trigger the OTA update I get a core panic (see below).

I checked:

* use current esp-idf version (pulled this morning).
* partition table: factory app, two OTA definitions
* blink-demo.bin works
* aws_s3_server.pem is correct
* download url is correct and works
* wifi connection is ok

still panics... I am stuck - ideas on what else to check / to get more info would be much appreciated

code:

``` c
    ESP_LOGI(TAG, "Connect to Wifi ! Starting download....");

    esp_http_client_config_t http_config;
    http_config.event_handler = _http_event_handler;

#if defined(CONFIG_EXAMPLE_EMBEDDED_CERTS)
    http_config.cert_pem = (const char *)aws_s3_server_pem_start;
#elif defined(CONFIG_EXAMPLE_FILESYSTEM_CERTS)
    http_config.cert_pem = S3_SERVER_PATH;
#endif

    http_config.url = "https://farm-iot-bin.s3.amazonaws.com/blink-demo.bin?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAJXB4Q3GGMEMHQ6DA%2F20181102%2Feu-central-1%2Fs3%2Faws4_request&X-Amz-Date=20181102T064442Z&X-Amz-Expires=3600&X-Amz-SignedHeaders=host&X-Amz-Signature=a72c503db8e9883506f6ac55d925bfc330f8fb30cb5340b71a44b1a615570d48";

    // download the binary from the presigned_url and run the update
    //http_config.url = presigned_url;
    esp_err_t rc = esp_https_ota(&http_config);
    if (rc == ESP_OK) {
        esp_restart();
    } else {
        ESP_LOGE(TAG, "Firmware upgrade failed code: %d", rc);
    }
    while (1) {
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
```

``` txt
I (19234) ESP_IOT: Connect to Wifi ! Starting download....
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014dc  PS      : 0x00060c30  A0      : 0x80001459  A1      : 0x3ffc76c0  
A2      : 0x00000001  A3      : 0xfffffffd  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x8008c66a  A9      : 0x3ffc7690  
A10     : 0x3ffc5fec  A11     : 0x00000012  A12     : 0x0000003c  A13     : 0x3ffd6944  
A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400014dc:0x3ffc76c0 0x40001456:0x3ffc76d0 0x40001445:0x3ffc76f0 0x401504f7:0x3ffc7710 0x40150ede:0x3ffc7740 0x4011cbb1:0x3ffc7760 0x400d2d26:0x3ffc77a0 0x400e7b75:0x3ffc7820 0x400e7d51:0x3ffc7850 0x400e7ee0:0x3ffc78a0 0x400e796c:0x3ffc78c0 0x400e7a32:0x3ffc78f0 0x400d2f03:0x3ffc7910 0x4008be85:0x3ffc85c0
0x401504f7: _set_config at /home/mark/devel/10_esp32/esp-idf/components/esp_http_client/esp_http_client.c:298

0x40150ede: esp_http_client_init at /home/mark/devel/10_esp32/esp-idf/components/esp_http_client/esp_http_client.c:298

0x4011cbb1: esp_https_ota at /home/mark/devel/10_esp32/esp-idf/components/esp_https_ota/src/esp_https_ota.c:43

0x400d2d26: iot_subscribe_callback_handler at /home/mark/devel/10_esp32/esp32/ota_mqtt_example/main/ota_update.c:214

0x400e7b75: _aws_iot_mqtt_internal_deliver_message at /home/mark/devel/10_esp32/esp-idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_common_internal.c:520

0x400e7d51: _aws_iot_mqtt_internal_handle_publish at /home/mark/devel/10_esp32/esp-idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_common_internal.c:555

0x400e7ee0: aws_iot_mqtt_internal_cycle_read at /home/mark/devel/10_esp32/esp-idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_common_internal.c:624

0x400e796c: _aws_iot_mqtt_internal_yield at /home/mark/devel/10_esp32/esp-idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_yield.c:206

0x400e7a32: aws_iot_mqtt_yield at /home/mark/devel/10_esp32/esp-idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client_yield.c:295

0x400d2f03: aws_ota_update_task at /home/mark/devel/10_esp32/esp32/ota_mqtt_example/main/ota_update.c:335

0x4008be85: vPortTaskWrapper at /home/mark/devel/10_esp32/esp-idf/components/freertos/port.c:403

Rebooting...
```

----------------------------- Delete below -----------------------------

mahavirj, 2018-11-02:
@markfink Probably `esp_http_client_config_t http_config` is missing zero initialization. Could you please try doing that?

markfink, 2018-11-02:
@mahavirj I do not understand what you mean by that... initialization has event_handler, cert_pem and url.

negativekelvin, 2018-11-02:
He means `esp_http_client_config_t http_config = {0};`

markfink, 2018-11-02:
@mahavirj @negativekelvin that fixes it. thank you so much!

closed 2018-11-02 by markfink