espressif/esp-idf #2658: Sink connect works no more

Elektrik1, 2018-11-02:
Since last week commits esp_a2d_sink_connect() does not connect to BT source.

On cold boot of esp, calling esp_a2d_sink_connect() immediately returns this:
```
I (5343) BTDM_INIT: BT controller compile version [7c8c9b8]
I (5343) system_api: Base MAC address is not set, read default base MAC address
from BLK0 of EFUSE
I (5503) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 2
E (5843) BT_APPL: reset flags
W (5843) BT_SDP: SDP - Rcvd conn cnf with error: 0x9  CID 0x40
I (5843) BT_AV: A2DP connection state: Connecting, [08:21:ef:a4:97:eb]
W (5853) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2
I (5853) BT_AV: A2DP connection state: Disconnected, [08:21:ef:a4:97:eb]
```

however if I connect and disconnect from my phone (or whatever device) then sink_connect starts to work with such log:
```
E (61133) BT_APPL: reset flags
I (61133) BT_AV: A2DP connection state: Connecting, [08:21:ef:a4:97:eb]
E (61183) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (61263) BT_APPL: new conn_srvc id:19, app_id:0
I (61263) BT_AV: A2DP connection state: Connected, [08:21:ef:a4:97:eb]
```

Code used for test in main.c:
```
esp_bd_addr_t what_to_connect = { 0x08, 0x21, 0xef, 0xa4, 0x97, 0xeb};
esp_a2d_sink_connect(what_to_connect);
```

blueMoodBHD, 2018-11-05:
@Elektrik1 According to your log, the reason for the connection failure is beyond max ACL connections.
Please run  `make menuconfig` , `Component config --->`, `Bluetooth --->`, `Bluetooth controller -->`,  set `BR/EDR ACL Max Connections` to 2 or more. (You may set it to 1, before).  Because page scan should use one.

Elektrik1, 2018-11-06:
@blueMoodBHD Thank you, that was the issue, however now BT connects but disconnects after ~ seconds with such error:

> E (6913) BT_APPL: reset flags
> I (6913) BT_AV: A2DP connection state: Connecting, [08:21:ef:a4:97:eb]
> E (8373) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
> E (11703) BT_APPL: bta_dm_pm_btm_status hci_status=12
> W (41763) BT_AVCT: ccb 1 not allocated
> I (41763) BT_AV: A2DP connection state: Disconnected, [08:21:ef:a4:97:eb]

sometimes those as well:

> E (880753) BT_APPL: reset flags
> I (880763) BT_AV: A2DP connection state: Connecting, [08:21:ef:a4:97:eb]
> E (881683) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
> E (885033) BT_APPL: bta_dm_pm_btm_status hci_status=12
> W (900383) BT_APPL: new conn_srvc id:19, app_id:1
> I (900383) BT_AV: A2DP audio state: Started
> E (915103) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x0
> W (915103) BT_L2CAP: L2CAP - no LCB for L2CA_SetAclPriority
> W (915103) BT_L2CAP: WARNING L2CA_SetFlushTimeout No lcb for bd_addr [...;a497eb]
> E (915113) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=7
> W (915123) BT_AVCT: ccb 1 not allocated
> I (915123) BT_AV: A2DP audio state: Stopped
> I (915133) BT_AV: A2DP connection state: Disconnected, [08:21:ef:a4:97:eb]

blueMoodBHD, 2018-11-07:
Hi,@Elektrik1,  I tried to reproduce the  new issue but failed. So could you please tell me the commit id of ESP-IDF and your code.
Thanks.

Alvin1Zhang, 2019-03-15:
@Elektrik1 Hi, would you help provide the commit ID of ESP-IDF and your code? Thanks.

Elektrik1, 2019-04-20:
Hello, can't tell the commit ID, but it is still broken, tested with latest pull, just added in main.c:
```
/* set discoverable and connectable mode, wait to be connected */
esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
		
esp_bd_addr_t what_to_connect = { 0x08, 0x21, 0xef, 0xa4, 0x97, 0xeb};
esp_a2d_sink_connect(what_to_connect);
		
break;
```

and occasionally results in log: https://pastebin.com/RJ08m2EN

Tested with Samsung S7 (android 8.0) and Samsung S6 (Android 7.0). Mostly occurs in cold start (power on). Issue goes away after connecting manually from phone to ESP, and then perform proper disconnect. Wonder if that's phone firmware denying connection after no proper BT disconnect procedure done, but a year ago that feature worked like charm with same hardware and code...

blueMoodBHD, 2019-04-24:
@Elektrik1,  commit 791209d0eb3548b53bddf333f3cc6b5ab58a04e0 may fix this problems,  Please update esp-idf and test again. Thanks.

Alvin1Zhang, 2019-05-17:
@Elektrik1 Hi, would you help try commit 791209d which may fix this problems and share the updates? Thanks.

Elektrik1, 2019-05-21:
Hello, sorry for the delay, been abroad.

I have just fetched last commits, and after recompile, couldn't get disconnects (without application of libcoexist.a file manually). 
Not sure if this fix was applied in main branch already, and it fixed the issue, or some other commit caused the fix ?

crespum, 2019-07-23:
@blueMoodBHD @Alvin1Zhang  I can confirm this is still an issue after applying 791209d. If the phone initiates the connection it works fine, however if the ESP is the initiator (through `esp_a2d_sink_connect(...)`), the connection is lost at some point. The logs differ depending on the IDF version, but overall they point at the same files:
```
E (312326) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x0
W (312326) BT_L2CAP: L2CAP - no LCB for L2CA_SetAclPriority
W (312326) BT_L2CAP: WARNING L2CA_SetFlushTimeout No lcb for bd_addr [...;ebecb8]
E (312336) BT_APPL: bta_dm_pm_sniff BTM_SetPowerMode() returns ERROR status=7
W (312346) BT_AVCT: ccb 1 not allocated
I (312356) BT_AV: A2DP audio state: Stopped
I (312356) BT_AV: A2DP conn state: Disconnected, c8:1e:e7:eb:ec:b8
```

blueMoodBHD, 2019-07-24:
Hi, I have test is with Huawei P10 and Iphone 7, it works well. What is the phone did you use? And is there other tips to reproduce it?

Alvin1Zhang, 2019-10-12:
@Elektrik1 Sorry for the very slow turnaround. Would you please help share the phone you used? And are there other tips to reproduce it? Thanks.

Elektrik1, 2019-10-14:
@Alvin1Zhang I have tested with Samsung S7 (android 8.0) and Samsung S6 (Android 7.0). Wonder what phone did @crespum used since he confirmed the issue as well.

crespum, 2019-10-15:
It's working for me now. It's been a while, but I think the problem was the ESP was initiating the connection while there was still another Bluetooth operation running (e.g. a disconnection).

dordnung, 2019-10-15:
I have exactly the same problem with a Samsung Galaxy S9 (Android 9).
If I connect with the phone there is no problem, but when I connect with "esp_a2d_sink_connect" to the phone it disconnects after a few seconds:

`
I (55597) BT_APP: Connect to device
E (55607) BT_APPL: reset flags
I (55607) BT_A2DP: A2DP connection state: Connecting
E (56737) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
D (56807) BT_A2DP: A2DP audio stream configuration; codec type 0
W (56887) BT_APPL: new conn_srvc id:19, app_id:0
I (56887) BT_A2DP: A2DP connection state: Connected
E (60087) BT_APPL: bta_dm_pm_btm_status hci_status=12
W (90147) BT_AVCT: ccb 1 not allocated
I (90147) BT_A2DP: A2DP connection state: Disconnected, reason 1
`

Elektrik1, 2019-10-16:
Steps to reproduce:

Take a2dp_sink from examples, in main.c after 

>/* set discoverable and connectable mode, wait to be connected */
        esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);

on line 197, before 

> break;

add:

```
esp_bd_addr_t what_to_connect = { 0x08, 0x21, 0xef, 0xa4, 0x97, 0xeb}; //your bt address here
esp_a2d_sink_connect(what_to_connect);
```

Flash to ESP32, pair and connect from the phone. Log:
```
ets Jun  8 2016 00:22:57����8�&���0�'&&��SH�HH����Ő    �]I�N���UQ�,b����3�i�%�F�*�e1�SH��ꊕjflash��V�Aerr,�LCQ��main.c��S�

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6848
ho 0 tail 12 room 4
load:0x40078000,len:14204
load:0x40080400,len:4292
entry 0x400806e4
I (75) boot: Chip Revision: 1
I (75) boot_comm: mismatch chip revision, expect 1, found 0
I (40) boot: ESP-IDF v4.1-dev-437-gd2ad0f077 2nd stage bootloader
I (40) boot: compile time 23:15:21
I (40) boot: Enabling RNG early entropy source...
I (46) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (54) boot: SPI Flash Size : 2MB
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: mismatch chip revision, expect 1, found 0
I (102) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1bb10 (113424) map
I (152) esp_image: segment 1: paddr=0x0002bb38 vaddr=0x3ffbdb60 size=0x02c30 ( 11312) load
I (157) esp_image: segment 2: paddr=0x0002e770 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/Espressif/esp-idf/components/freertos/xtensa_vectors.S:1778

I (159) esp_image: segment 3: paddr=0x0002eb78 vaddr=0x40080400 size=0x01498 (  5272) load
I (170) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x708c4 (460996) map
0x400d0018: _stext at ??:?

I (343) esp_image: segment 5: paddr=0x000a08e4 vaddr=0x40081898 size=0x11b94 ( 72596) load
0x40081898: coex_bt_request_wrapper at D:/Espressif/esp-idf/components/esp32/esp_adapter.c:475

I (386) boot: Loaded app from partition at offset 0x10000
I (386) boot: Disabling RNG early entropy source...
I (386) cpu_start: Pro cpu up.
I (390) cpu_start: Application information:
I (395) cpu_start: Project name:     a2dp_sink
I (400) cpu_start: App version:      1
I (404) cpu_start: Compile time:     Oct 16 2019 23:14:59
I (410) cpu_start: ELF file SHA256:  81bd31d94a408369...
I (416) cpu_start: ESP-IDF:          v4.1-dev-437-gd2ad0f077
I (423) cpu_start: Starting app cpu, entry point is 0x400812b0
0x400812b0: call_start_cpu1 at D:/Espressif/esp-idf/components/esp32/cpu_start.c:276
[... 143 lines cut ...]
W (3132) BT_APPL: new conn_srvc id:19, app_id:0
I (3132) BT_AV: A2DP connection state: Connected, [08:21:ef:a4:97:eb]
I (3262) RCCT: AVRC conn_state evt: state 1, [08:21:ef:a4:97:eb]
I (3262) RCTG: AVRC conn_state evt: state 1, [08:21:ef:a4:97:eb]
I (3262) RCCT: AVRC remote features 25b, TG features 51
I (3262) RCTG: start volume change simulation
I (3272) RCTG: AVRC remote features 25b, CT features 2
I (3282) RCCT: remote rn_cap: count 7, bitmask 0xf26
I (3332) RCCT: AVRC metadata rsp: attribute id 0x1,
I (3342) RCCT: AVRC metadata rsp: attribute id 0x2,
I (3342) RCCT: AVRC metadata rsp: attribute id 0x4,
I (3342) RCCT: AVRC metadata rsp: attribute id 0x20,
I (3362) RCTG: AVRC register event notification: 13, param: 0x0
I (4582) RCCT: AVRC event notification: 5
I (4582) BT_AV: Play position changed: 0-ms
E (6342) BT_APPL: bta_dm_pm_btm_status hci_status=12
I (13282) RCTG: Volume is set locally to: 3%
I (23282) RCTG: Volume is set locally to: 7%
I (33282) RCTG: Volume is set locally to: 11%
W (36422) BT_AVCT: ccb 1 not allocated
I (36422) BT_AV: A2DP connection state: Disconnected, [08:21:ef:a4:97:eb]
I (36422) RCCT: AVRC conn_state evt: state 0, [00:00:00:00:00:00]
I (36432) RCTG: AVRC conn_state evt: state 0, [00:00:00:00:00:00]
I (36432) RCTG: Stop volume change simulation
```

wasmac, 2020-09-17:
Anyone has a working fix for that? I got the same problem at windows 10 (msi ge62 6qf apache pro laptop).

Wth-Esp, 2020-10-29:
This log indicates that A2DP SRC profile has not been initialized when SNK is producing inquiry scan.
`W (5853) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2`
Please refer to the a2dp demo of IDF to fix this issue in app layer.

nikolindima, 2020-10-31:
Have same problem. I have board LyraT-V4.3 and BeatsSolo Wireless headphones. I use ESP board as BLUETOOTH_A2DP_SOURCE. At first day Headphones connect normally and I have no problems. But on next day I started to receive error - 
`I (13829) BLUETOOTH_SERVICE: Discovery started.
I (14229) BLUETOOTH_SERVICE: Scanned device: 04:88:e2:6e:18:c5
I (14229) BLUETOOTH_SERVICE: --Class of Device: 0x240418
I (14229) BLUETOOTH_SERVICE: --RSSI: -46
I (14229) BLUETOOTH_SERVICE: --Name: BeatsSolo Wireless
I (14239) BLUETOOTH_SERVICE: Found a target device, address 04:88:e2:6e:18:c5, name BeatsSolo Wireless
I (14249) BLUETOOTH_SERVICE: Cancel device discovery ...
I (14259) BLUETOOTH_SERVICE: Device discovery stopped.
I (14259) BLUETOOTH_SERVICE: a2dp connecting to peer: BeatsSolo Wireless
W (14269) BT_SDP: SDP - Rcvd conn cnf with error: 0x9  CID 0x40

W (14279) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2`

Nothing helps to fix this. I was thinking that the problem in ESP32 module. I change ESP32-WROVER-B on LyraT board and still have save Error.
Headphones connecting normally to different ESP board and to any phone.
Please help me

Wth-Esp, 2020-11-02:
> Have same problem. I have board LyraT-V4.3 and BeatsSolo Wireless headphones. I use ESP board as BLUETOOTH_A2DP_SOURCE. At first day Headphones connect normally and I have no problems. But on next [...]

From the log, the device discovery procedure (inquiry) is doing its work. But in the specific time duration, the A2DP SNK cannot be discovered. So first, please check the headphones inquiry open or not; second, please change the time duration parameter in `esp_bt_gap_start_discovery()` to a relatively long time to make sure the headphone can be discovered.

nikolindima, 2020-11-02:
@Wth-Esp I fix the problem with full reinstall esp-idf esp-adf. Now headphones connecting with no problems

Alvin1Zhang, 2020-12-22:
Thanks for sharing the updates, will close now, feel free to reopen.

closed 2020-12-22 by Alvin1Zhang