espressif/esp-idf #2661: [cmake buildtoolchain] cmake functions improvement

callalilychen, 2018-11-03:
Hi,

I tried to use cmake buildtoolchain from you. It is really very nice of you to provide it. 

I would like to recommend to change the provided cmake functions, e.g. `register_component()`
from:

```
set(COMPONENT_SRCS "foo.c")
set(COMPONENT_ADD_INCLUDEDIRS "include")
register_component()

```
to

```
register_component(
    SRCS "foo.c"
    INCLUDEDIRS "include"
)
```
cmake provides the built-in argument parsing function: [cmake_parse_arguments](https://cmake.org/cmake/help/latest/command/cmake_parse_arguments.html#command:cmake_parse_arguments) to realize it.

projectgus, 2019-03-06:
Hi @callalilychen ,

Thanks for being patient while someone got back to you, sorry we didn't get back earlier.

We are going to make some changes to the CMake-based build system in IDF V4.0 which are similar to what you suggest. Originally the idea was to make it very similar to the GNU Make build system for users who migrate their projects, but the CMake style is better in a number of ways.

This issue will be updated again once that supports lands in master.

projectgus, 2019-06-04:
Hi @callalilychen ,

The latest IDF CMake updates are in master now, and recommend a more CMake-like approach. We're having some issues with the docs build on RTD, so the docs.espressif.com has not updated yet (it should soon) but you can see the latest in the source files on GitHub:
https://github.com/espressif/esp-idf/blob/master/docs/en/api-guides/build-system-cmake.rst#minimal-component-cmakelists

If you have any additional requests for changes, please let us know. Otherwise, we'll probably close this issue.

projectgus, 2019-10-14:
Closing as we've adopted the suggested style.

@callalilychen if you have any other suggestions for CMake API improvements, please feel free to open new issues. :)

closed 2019-10-14 by projectgus