espressif/esp-idf #2663: heap assert failed when call heap_caps_print_heap_info(MALLOC_CAP_SPIRAM)

chegewara, 2018-11-04:
I am trying to build app: I2S + DMA with BLE. As for now its not complicated app yet, simple camera app in one task and simple BLE app in other task. If i run only camera task or only ble task it works fine, but when i am trying to run both i am getting assert multi_heap error. Usually i can find the reason, but this time i am getting this error when i am trying to print heap info from SPIRAM with this command: `		heap_caps_print_heap_info(MALLOC_CAP_SPIRAM);` even before any task is running, literally its first command in app_main. This is log:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6212
load:0x40078000,len:10084
ho 0 tail 12 room 4
load:0x40080400,len:6612
entry 0x40080764
I (30) boot: ESP-IDF v3.2-dev-1429-g7efbe8178-dirty 2nd stage bootloader
I (30) boot: compile time 13:39:45
I (31) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00330000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x533f0 (340976) map
I (215) esp_image: segment 1: paddr=0x00063418 vaddr=0x3ff80000 size=0x00000 (     0) load
I (215) esp_image: segment 2: paddr=0x00063420 vaddr=0x3ff80000 size=0x00000 (     0) load
I (221) esp_image: segment 3: paddr=0x00063428 vaddr=0x3ffbdb60 size=0x02d84 ( 11652) load
I (235) esp_image: segment 4: paddr=0x000661b4 vaddr=0x3ffc08e4 size=0x00000 (     0) load
I (239) esp_image: segment 5: paddr=0x000661bc vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (249) esp_image: segment 6: paddr=0x000665c4 vaddr=0x40080400 size=0x09a4c ( 39500) load
I (274) esp_image: segment 7: paddr=0x00070018 vaddr=0x400d0018 size=0xa2f58 (667480) map
0x400d0018: _flash_cache_start at ??:?

I (508) esp_image: segment 8: paddr=0x00112f78 vaddr=0x40089e4c size=0x0dea4 ( 56996) load
0x40089e4c: r_lld_evt_schedule at ??:?

I (532) esp_image: segment 9: paddr=0x00120e24 vaddr=0x400c0000 size=0x00000 (     0) load
I (532) esp_image: segment 10: paddr=0x00120e2c vaddr=0x50000000 size=0x00000 (     0) load
I (538) esp_image: segment 11: paddr=0x00120e34 vaddr=0x50000000 size=0x00000 (     0) load
I (561) boot: Loaded app from partition at offset 0x10000
I (561) boot: Disabling RNG early entropy source...
I (563) spiram: Found 64MBit SPI RAM device
I (566) spiram: SPI RAM mode: flash 40m sram 40m
I (572) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (579) cpu_start: Pro cpu up.
I (582) cpu_start: Starting app cpu, entry point is 0x40081548
0x40081548: call_start_cpu1 at D:/msys32/home/imper/esp/esp-idf/components/esp32/cpu_start.c:245

I (0) cpu_start: App cpu up.
I (1464) spiram: SPI SRAM memory test OK

0x40080000: _WindowOverflow4 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

D (1534) memory_layout: Reserved memory range 0x40080000 - 0x40097cf0
0x40080000: _WindowOverflow4 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (1748) heap_init: Initializing. RAM available for dynamic allocation:
D (1755) heap_init: New heap initialised at 0x3ffaff10
I (1760) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
D (1766) heap_init: New heap initialised at 0x3ffb6388
I (1771) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
D (1778) heap_init: New heap initialised at 0x3ffb9a20
I (1783) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1789) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
D (1795) heap_init: New heap initialised at 0x3ffc3848
I (1800) heap_init: At 3FFC3848 len 0001C7B8 (113 KiB): DRAM
I (1807) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1813) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (1820) heap_init: New heap initialised at 0x40097cf0
I (1825) heap_init: At 40097CF0 len 00008310 (32 KiB): IRAM
I (1831) cpu_start: Pro cpu start user code
I (1836) spiram: Adding pool of 4089K of external SPI memory to heap allocator
D (1875) clk: RTC_SLOW_CLK calibration value: 15264218
D (92) intr_alloc: Connected src 46 to int 2 (cpu 0)
D (93) intr_alloc: Connected src 57 to int 3 (cpu 0)
D (93) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (97) cpu_start: Starting scheduler on PRO CPU.
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (4) cpu_start: Starting scheduler on APP CPU.
D (121) heap_init: New heap initialised at 0x3ffe0440
D (127) heap_init: New heap initialised at 0x3ffe4350
I (132) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
D (140) spiram: Allocating block of size 32768 bytes
Heap summary for capabilities 0x00000400:
assertion "next > (intptr_t)block" failed: file "D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap
.c", line 124, function: get_next_block
abort() was called at PC 0x400d7867 on core 0
0x400d7867: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newl
ib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x400950d0:0x3ffbb070 0x40095301:0x3ffbb090 0x400d7867:0x3ffbb0b0 0x40094e16:0x3ffbb0e0 0x400d703
a:0x3ffbb100 0x400d747f:0x3ffbb140 0x400d306a:0x3ffbb170 0x40092a79:0x3ffbb190
0x400950d0: invoke_abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:680

0x40095301: abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:680

0x400d7867: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newl
ib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x40094e16: get_next_block at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:377
 (inlined by) multi_heap_get_info_impl at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:724

0x400d703a: heap_caps_print_heap_info at D:/msys32/home/imper/esp/esp-idf/components/heap/heap_caps.c:389

0x400d747f: app_main at D:/esp32_camera/ble_project/ESP32_CAMERA_QR/main/app_main.cpp:95

0x400d306a: main_task at D:/msys32/home/imper/esp/esp-idf/components/esp32/cpu_start.c:493

0x40092a79: vPortTaskWrapper at D:/msys32/home/imper/esp/esp-idf/components/freertos/port.c:435

CPU halted.
```

projectgus, 2018-11-05:
Hi @chegewara ,

This looks like a bug, the assert indicates heap corruption somehow. Changing the malloc() threshold maybe works around it by causing something that was (incorrectly) allocated in SPIRAM to be allocated into internal RAM.

Can you please try making it crash again and then try calling `heap_caps_dump(MALLOC_CAP_SPIRAM);` instead of `heap_caps_print_info()`, and show us what it dumps before it crashes?

Angus

chegewara, 2018-11-05:
I have to change the story a little bit. All i described is correct except that issue "has been fixed" when i changed this parameter in menuconfig:
` [*] Allow .bss segment placed in external memory` not `Maximum malloc() size, in bytes, to always put in internal memory` (i changed both but Maximum malloc size is not important).

Logs from `heap_caps_dump`:
```
Heap start 0x3f806bf4 end 0x3fbffff4
First free block 0x3f806bf9
Block 0x3f806bf4 data size 0x00000000 bytes next block 0x3f806bf8 FREE. Next free 0x3f806bf9
assertion "next > (intptr_t)block" failed: file "D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c", line 124, function: get_next_block
abort() was called at PC 0x400d77ff on core 0
0x400d77ff: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x400951b8:0x3ffbb090 0x400953e9:0x3ffbb0b0 0x400d77ff:0x3ffbb0d0 0x40094de7:0x3ffbb100 0x400d7033:0x3ffbb120 0x400d7432:0x3ffbb140 0x400d304e:0x3ffbb170 0x40092a91:0x3ffbb190
0x400951b8: invoke_abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:680

0x400953e9: abort at D:/msys32/home/imper/esp/esp-idf/components/esp32/panic.c:680

0x400d77ff: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x40094de7: get_next_block at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:124
 (inlined by) multi_heap_dump at D:/msys32/home/imper/esp/esp-idf/components/heap/multi_heap.c:689

0x400d7033: heap_caps_dump at D:/msys32/home/imper/esp/esp-idf/components/heap/heap_caps.c:441
```

projectgus, 2018-11-05:
@chegewara So it crashes when "Allow .bss segment placed in external memory" is disabled, but enabling this option it doesn't crash. Is that correct?

Any chance that you are able to send us the .elf file for this app, please?

chegewara, 2018-11-05:
@projectgus Enabling this option causing crash, with disabled is all ok.

projectgus, 2019-03-01:
Hi @chegewara ,

Sorry I never followed up on this. If you still have the app and the ELF file, please share them. Thanks.

Angus

chegewara, 2019-03-01:
Sorry, i dont have any resources now. I will close it and reopen when i have any news.

closed 2019-03-01 by chegewara