espressif/esp-idf #2664: [question] characteristic value length

chegewara, 2018-11-04:
@Weijian-Espressif 
Hi, at the moment have constant length of characteristic value with length 600. In most cases it is not bad, but i have two opposite situations at the moment. One is request where user needs to create about 80 characteristics in project, hell a lot, i know. In this case 80 characteristics with over 600 bytes each is waste of heap and there is no option to change it except changing this value for that particular project:
https://github.com/espressif/esp-idf/blob/12b2268ee56dac79988b75dbb49a453629ed8022/components/bt/bluedroid/api/include/api/esp_gatt_defs.h#L290

Now i have project where i need characteristic length 30-300kB which i can take control from my app and possibly its best way to do it because of this comment `ESP_GATT_MAX_ATTR_LEN   600 //as same as GATT_MAX_ATTR_LEN`.

My question is, will be possible to implement somehow dynamic size of characteristic value in bt stack, or best way is to do it from app?
Thanks.

Weijian-Espressif, 2018-11-06:
@chegewara  I will change it to dynamic allocation in stack.  I have a lot of tasks to do this month, i try to finish next month. I have a question, your characteristic length is 30-300kB,  our ESP32 memory is not enough, how do you plan to solve it?

chegewara, 2018-11-06:
@Weijian-Espressif Thanks for changing it.

Im thinking about creating 2 simple app (one for android), to send jpeg base64 encoded image from esp32 to android device and display it as image. Thats why i need so big characteristic. I am aware that it cant be achieved with regular ram, so i am thinking about using spiram to do it. If i could not allocate characteristic value in spiram then another idea is to pass pointer to characteristic and then send data as very long value. I dont know yet if android can handle this, but its only proof concept project, so no rush.

Weijian-Espressif, 2018-11-06:
@chegewara , if you want to send an image to android,  you can use indication,  manual segmented transmission.

chegewara, 2018-11-06:
@Weijian-Espressif I know, but to do this i still have to set value of characteristic to 30-300kB length, or read chunks of data and send indications. Thanks for idea.
Like i said, in my situation it will be better to handle this in app level, but for 80 characteristic with descriptors in each one its a lot memory allocated and wasted since client claims that characteristic and descriptor value will be 20-ish byte length max.

Weijian-Espressif, 2018-11-06:
@chegewara , Value of characteristic is dynamic allocation in IDF.

chegewara, 2018-11-06:
@Weijian-Espressif Maybe this is noob question, but how? I see that characteristic value is uint8_t array with length 600:
https://github.com/espressif/esp-idf/blob/master/components/bt/bluedroid/api/include/api/esp_gatt_defs.h#L369

Weijian-Espressif, 2018-11-07:
@chegewara,  it is the max length of response,  you can add char with value length over 600 bytes.

Weijian-Espressif, 2018-11-19:
@chegewara, Do you have any question?

chegewara, 2018-11-19:
@Weijian-Espressif Not at the moment, thanks.

closed 2018-11-19 by chegewara