espressif/esp-idf #2665: esp-idf V3.0 how to config the component.mk file ?

1XiaSheng, 2018-11-05:
[image]
[image]

1XiaSheng, 2018-11-05:
Great God, please see the above figure, what configuration is required for each component.mk file in the directory structure?

The correct guidelines are not available at https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html.

大神,请看上面的图,目录结构中的各个component.mk文件需要填入什么配置?

在https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html 中无法得到正确指引.

igrr, 2018-11-05:
Hi @1XiaSheng,

ESP-IDF build system does not support nested components. 

What you can do instead is: create a `components` directory in your project, and add `cfg`, `hdw`, and `usr` as component directories. Each of these directories can then include component.mk file. If you want your source files to be in `src` subdirectory and include files in `include` subdirectory of each component, then the component.mk will files look like:
```
COMPONENT_ADD_INCLUDEDIRS := include
COMPONENT_SRCDIRS := src
```

1XiaSheng, 2018-11-05:
thanks, i will try it now,see you later, i will reply you when i have test it in your way

1XiaSheng, 2018-11-19:
its working good! thanks

closed 2018-11-19 by 1XiaSheng

woiptcs, 2020-05-19:
Hello , 
I'm having troubles adding new components to my project.
I've allready added a component directory using CMakelists.txt the esp-idf-libs for asd111x but now I cannot add more libs ... 
If I use an absolute path , than I cannot use relatives ? 
I can add multiple paths? separed by " " space ? 
No combination of CMakeLists.txt and Component.mk is valid for my I think I tried everything posible ..