espressif/esp-idf #2667: smp_process_pairing_public_key, Invalid Public key

chegewara, 2018-11-05:
@Weijian-Espressif @Yulong-espressif Could you point me direction please.
I am assuming it is my fault that i am getting this error message, just i cant figure out what i do that i am changing keys value. The story is i am trying to pair with windows 10 and all has been working few weeks ago, now it complains about invalid pairing key. esp-idf gatt_secure example dont have this problem.
`E (119013) BT_SMP: smp_process_pairing_public_key, Invalid Public key.`
https://github.com/espressif/esp-idf/blob/master/components/bt/bluedroid/stack/smp/smp_act.c#L739

```
I (118853) BT_BTM: BTM_InqDbRead: bd addr [80a58982f5d2]

I (118873) BT_GATT: GATT_GetConnIdIfConnected status=1

I (118883) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 80a58982f5d2
I (118883) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (118883) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (118903) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 80a58982f5d2
I (118923) BLEDevice: ESP_GAP_BLE_SEC_REQ_EVT
I (118933) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0006  BDA: 80a58982f5d2
I (118933) BT_L2CAP: L2CA_EnableUpdateBleConnParams - BD_ADDR 80a58982f5d2 enable 0 current upd state 0x0c
I (118943) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 80a58982f5d2
E (119013) BT_SMP: smp_process_pairing_public_key, Invalid Public key.
I (119013) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0006  BDA: 80a58982f5d2
I (119013) BT_L2CAP: L2CA_EnableUpdateBleConnParams - BD_ADDR 80a58982f5d2 enable 1 current upd state 0x0c
I (119023) BT_L2CAP: L2CA_RemoveFixedChnl()  CID: 0x0006  BDA: 80a58982f5d2
I (119033) BT_BTM: BTM_InqDbRead: bd addr [80a58982f5d2]

I (119033) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 80a58982f5d2

I (119043) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 80a58982f5d2
W (119053) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

I (119063) BT_GATT: GATT_GetConnIdIfConnected status=0

E (119063) BT_BTM: Device not found

I (119073) BT_BTM: BTM_IsAclConnectionUp: RemBdAddr: 80a58982f5d2

I (119073) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
W (119083) BT_APPL: BTA got unregistered event id 31

W (119093) BT_APPL: BTA got unregistered event id 31

I (119103) BLEDevice: ESP_GAP_BLE_AUTH_CMPL_EVT
```

Weijian-Espressif, 2018-11-06:
@hegewara Can you print the values of` peer_publ_key.x` and `peer_publ_key.y` in `smp_process_pairing_public_key()`.  ESP32 will check whether the x and y coordinates are on the given elliptic curve.

chegewara, 2018-11-06:
@Weijian-Espressif For some reason `peer_bul_key.y` is not initialized and x is not fully initialized:
```
W (17538) SMP_ACT X:: 0c d7 0f 90 45 14 e5 68 31 47 7d 22 4a 3d fe 05
W (17538) SMP_ACT X:: 6e 2d 17 3b 82 79 00 00 00 00 00 00 00 00 00 00
W (17548) SMP_ACT Y:: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
W (17548) SMP_ACT Y:: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
This is from android device:
```
W (37638) SMP_ACT X:: 07 6e 9f c4 79 7b 1c 71 4a 09 a9 6a 3e d0 d0 73
W (37638) SMP_ACT X:: 1b 89 9b 8a 24 2d e5 ef 0d ab 2d 42 49 50 b4 4b
W (37638) SMP_ACT Y:: 99 c1 c0 2a 0e 29 43 c8 1d 82 95 79 98 1a 71 6e
W (37648) SMP_ACT Y:: 75 54 49 62 54 8a 8f e2 f6 9f 77 5a 29 16 e7 29
```
What is even more disturbing, when iPhone is exchanging keys `smp_process_pairing_public_key` is not even called:
```
W (24748) SampleHIDDevice: custom gatts event handler, event: 14, length: 0
W (24748) SampleHIDDevice: custom gatts event handler, event: 4, length: 0
W (24828) SampleHIDDevice: custom gap event handler, event: 20
E (25338) BT_GATT: GATT_INSUF_ENCRYPTION

W (25398) SampleHIDDevice: custom gap event handler, event: 10
W (25458) SampleHIDDevice: custom gap event handler, event: 20
W (27048) SampleHIDDevice: custom gap event handler, event: 9
W (27058) SampleHIDDevice: custom gap event handler, event: 9
W (27258) SampleHIDDevice: custom gatts event handler, event: 1, length: 5467
W (27258) SampleHIDDevice: custom gap event handler, event: 9
W (27258) SampleHIDDevice: custom gap event handler, event: 9
W (27268) SampleHIDDevice: custom gap event handler, event: 20
W (27268) SampleHIDDevice: custom gap event handler, event: 8
W (27278) SampleHIDDevice: custom gatts event handler, event: 21, length: 0
W (27558) SampleHIDDevice: custom gatts event handler, event: 1, length: 5467
W (27558) SampleHIDDevice: custom gatts event handler, event: 21, length: 21346
W (27678) SampleHIDDevice: custom gatts event handler, event: 1, length: 5467
W (27688) SampleHIDDevice: custom gatts event handler, event: 21, length: 21346
W (27918) SampleHIDDevice: custom gatts event handler, event: 1, length: 5467
W (27918) SampleHIDDevice: custom gatts event handler, event: 21, length: 21346
W (27978) SampleHIDDevice: custom gatts event handler, event: 1, length: 5467
W (27978) SampleHIDDevice: custom gatts event handler, event: 21, length: 21346
W (28068) SampleHIDDevice: custom gatts event handler, event: 2, length: 5467
W (28068) SampleHIDDevice: custom gatts event handler, event: 21, length: 21346
W (28368) SampleHIDDevice: custom gatts event handler, event: 2, length: 5467
W (28368) SampleHIDDevice: custom gatts event handler, event: 21, length: 21346
W (28438) SampleHIDDevice: custom gap event handler, event: 20
W (28438) SampleHIDDevice: custom gatts event handler, event: 1, length: 5467
W (28448) SampleHIDDevice: custom gatts event handler, event: 21, length: 21346
```

chegewara, 2018-11-06:
Here is my wireshark file (renamed to upload):

[win10.pcapng.txt](https://github.com/espressif/esp-idf/files/2554326/win10.pcapng.txt)

Yulong-espressif, 2018-11-07:
@chegewara From the log you gave, it is because your y coordinate is not initialized. If the y coordinate is not initialized, it means that your public key is not on the given elliptic curve. In this case, when your private key is the base, then your DHkey is calculated to be equal to the Public key; when your private key is even, DHkey is 0.