espressif/esp-idf #2668: WPA2 PEAP connection disconnect after 30 minutes - reason 23

PaulFreund, 2018-11-05:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v3.2-beta1-34-gfb7ba1baa
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 1.22.0-80-g6c4433a) 5.2.0
- Operating System:     Linux
- Power Supply:         external 3.3V

## Problem Description

After being connected and exchanging data for about 20-40 minutes (not a reliable number) to a Enterprise Cisco AP the ESP32 is disconnected with reason 23:

    E (76330858) DeviceManager: STA Disconnected event, reason(23)

I can not be 100% sure but it looks like this is from the log of the Cisco AP:

    *dot1xMsgTask: Oct 30 12:11:24.731: %DOT1X-3-WPA_SEND_STATE_ERR: [PA]1x_kxsm.c:1693 Unable to send EAPOL-key msg  - invalid WPA state (0) - client 84:0d:8e:xx:xx:xx

### Expected Behavior

ESP32 stays connected to AP

### Actual Behavior

ESP32 gets disconnected

### Steps to repropduce

1. Connect to AP
2. Wait
3. See that the ESP32 got disconnected

mikaelkanstrup, 2018-12-17:
This is likely caused by esp32 not responding to eap reauthentication initiated by AP.
I opened an issue about this a few months back. https://github.com/espressif/esp-idf/issues/2324

PaulFreund, 2018-12-17:
You are right @mikaelkanstrup, sure sounds like it. I hope this will be implemented soon!

sagb2015, 2019-10-12:
@PaulFreund Apologies fo the delay!  #2324 is now fixed.  Can you please check if this addresses your issue as well?

PaulFreund, 2019-10-14:
Hi, as written in #2324 this does unfortunately not solve my issue. I have yet to find enough time to dig deeper.

sagb2015, 2019-10-14:
@PaulFreund - Yes, I missed the discussion on the other thread. Let us use this tracker for the issue. Can you help with the sniffer capture? It may be because of the channel conditions.  Quoting my reply on the other thread - "In the failure case, the router did not get message 2/4 from ESP32. So it is retrying by sending 1/4 to ESP32 again. ".

PaulFreund, 2019-10-14:
I will do as soon as possible but I need to find some Equipment first. 

About the missed message it has to be a logical error and not a transmission. We have about 200 ESP32 equippped boards that all disconnect after 30 minutes with a quote of 100% (we log all reconnects). I will try to make a log as soon as possible but it will most likely be noisy.

kanstrup, 2019-10-14:
@PaulFreund @sagb2015 I just retested #2324 and realized I didn't check the sniffer logs before confirming the issue as solved. Now that I retested and captured sniffer logs I realize that the fix is not OK. This is most likely the problem Paul sees. Due to #3956 I was never able to verify in a production environment and instead used a local setup with hostapd that apparently did not verify incoming frames properly.

What I realize now when investigating the sniffer logs is that during EAP reauthentication the 4-way handshake messages (2/4 and 4/4) are sent unencrypted from ESP32, not encrypted with current unicast key. I will reopen #2324 and add both sniffer and debug logs.

mikaelkanstrup, 2019-10-14:
@sagb2015 logs and sniffer logs attached to #2324. I was not able to re-open the issue so please do if. It's not solved correctly.

sagb2015, 2019-12-16:
Hi @PaulFreund & @mikaelkanstrup - I am closing this thread as well as #2324 is fixed. Feel free to re-open in case of issues.

closed 2019-12-16 by sagb2015