espressif/esp-idf #2671: ESP32的MQTT问题打印No PING_RESP, disconnected

Freepom, 2018-11-06:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2018-11-06:
The translation of the issue description is: When MQTT is in use, it suddenly lost connection and then printed No. PING_RESP, disconnected. Do you have any idea what caused this issue? @Freepom Please help correct if the translation is not correct. Thanks.

david-cermak, 2018-11-06:
Hi @Freepom ,

Generally this is caused by the MQTT broker not responding to pings. By default the client keep-alive setting is to 120s, so the issue indicates there was no communication server-client for this period.

If your connection is stable and your broker is online, this might also be an issue with mqtt-client not correctly processing some messages and thus not correctly sending pings (so the broker does not send ping responses). It is not easy to tell what is wrong in this case without deep understanding  of your application (plus the cause might have happened more than 120s before no ping response).

Can you please share, what is your application doing, what kind of data publishing. Also might be helpful to setup "verbose" log level (from menuconfig -> Component config -> Log output -> Verbose), so you can see more details what is going on. Example of such  log is here:
```
I (3012) MQTT_EXAMPLE: MQTT_EVENT_SUBSCRIBED, msg_id=45992
D (63022) MQTT_CLIENT: Sent PING successful
D (63022) MQTT_CLIENT: PING sent
D (63182) MQTT_CLIENT: msg_type=13, msg_id=0
D (63182) MQTT_CLIENT: MQTT_MSG_TYPE_PINGRESP
```
As you can see, the output contains also 'D(ebug)' logs about sending PINGs, msg types, etc.

Thanks, David.

david-cermak, 2018-12-12:
Hi @Freepom, 
any update from your side? Have you succeeded in connecting?

Thanks for keeping us posted! David

Alvin1Zhang, 2019-03-15:
@Freepom  Any updates? Thanks.

Alvin1Zhang, 2019-05-16:
@Freepom Thanks for reporting the issue. Will close the issue for now due to lack of updates and details. Feel free to reopen the issue, and please help share updates then. Thanks.

closed 2019-05-16 by Alvin1Zhang

controllermaster, 2019-11-15:
Below are the logs I am receiving while there is a random disconnection. 
Publish frequency is one min and publish data length is around 200bytes 
The network is stable and broker is online (as other devices are publishing successfully at same time ) 
Keepalive time is 30sec.

```
D (480354) MQTT_CLIENT: mqtt_enqueue id: 33373, type=3 successful
D (495464) MQTT_CLIENT: Sent PING successful
D (495464) MQTT_CLIENT: PING sent
E (511464) MQTT_CLIENT: No PING_RESP, disconnected
D (511464) MQTT_CLIENT: Reconnect after 10000 ms
I (511464) MQTT_EVENT: MQTT_EVENT_DISCONNECTED
D (526464) MQTT_CLIENT: Reconnecting...
I (531464) MQTT_EVENT: MQTT_EVENT_BEFORE_CONNECTED
D (544744) MQTT_CLIENT: Transport connected to mqtts://192.168.1.10:8883
I (544744) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
D (544754) MQTT_CLIENT: Connected
I (544754) MQTT_EVENT: MQTT_EVENT_CONNECTED
D (544754) MQTT_CLIENT: mqtt_enqueue id: 0, type=1 successful
I (544764) publish_data: 
```
Please help with this.
Thanks

Alex-Smart, 2020-04-14:
So Ive got even worse situation, constant **No PING_RESP, disconnected**
Ive added some debug lines in mqtt_client.c, 
`ESP_LOGE(TAG, "\r\n in mqtt_write_data() \r\n Platform: %llu,client_tick: %llu, keepalive: %d,\r\n", platform_tick_get_ms(),client->keepalive_tick, (client->connect_info.keepalive* 1000)/2);`
and the reason as I see it is **platform_tick_get_ms()** does not work properly
```
E (4451) MQTT_CLIENT: 
 in esp_mqtt_client_init() 
 Platform: 0,client_tick: 0, keepalive: 60000 

E (4461) MQTT_EXAMPLE:  -------------------- MQTT CLIENT CREATED -------------------- 
I (4471) MQTT_EXAMPLE: Other event id:7
I (4481) MQTT_CLIENT: Connecting to: mqtt://192.168.198.151:1883
I (4501) MQTT_CLIENT: Sending MQTT CONNECT message, type: 1, id: 0000
I (4521) MQTT_EXAMPLE: MQTT_EVENT_CONNECTED
E (5521) MQTT_CLIENT: 
 in esp_mqtt_task() 
 Platform: 1072513,client_tick: 0, keepalive: 60000 

E (5521) MQTT_CLIENT: 
 in mqtt_write_data() 
 Platform: 18446741927103497129,client_tick: 18446741927103497129, keepalive: 60000,

E (5531) MQTT_CLIENT: 
 in esp_mqtt_task() 
 Platform: 1072513,client_tick: 18446741927103497129, keepalive: 60000 

E (5541) MQTT_CLIENT: 
 in mqtt_write_data() 
 Platform: 18446741927103497129,client_tick: 18446741927103497129, keepalive: 60000,

E (5551) MQTT_CLIENT: 
 in esp_mqtt_task() 
 Platform: 1072513,client_tick: 18446741927103497129, keepalive: 60000 

E (5571) MQTT_CLIENT: 
 in mqtt_write_data() 
 Platform: 18446741927103497129,client_tick: 18446741927103497129, keepalive: 60000,

E (5581) MQTT_CLIENT: 
 in esp_mqtt_task() 
 Platform: 1072513,client_tick: 18446741927103497129, keepalive: 60000 
```

eric-Kiaofarming, 2023-08-17:
Hi everyone

I have the same problem,
The information captured by ESP-IDF is as follows:
```
D (106239) transport_base: poll_read: select - Timeout before any socket was ready!
D (106239) transport_base: poll_read: select - Timeout before any socket was ready!
D (106239) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read(): call timed out before data was ready!
D (106249) mqtt_client: Sent PING successful
D (106259) mqtt_client: mqtt_message_receive: first byte: 0xd0
D (106259) mqtt_client: mqtt_message_receive: read "remaining length" byte: 0x0
D (106269) mqtt_client: mqtt_message_receive: total message length: 2 (already read: 2)
D (106279) mqtt_client: mqtt_message_receive: transport_read():2 2
D (106279) mqtt_client: msg_type=13, msg_id=0
D (106289) mqtt_client: MQTT_MSG_TYPE_PINGRESP
D (107299) transport_base: poll_read: select - Timeout before any socket was ready!
D (107299) transport_base: poll_read: select - Timeout before any socket was ready!
D (107299) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read(): call timed out before data was ready!
D (108319) transport_base: poll_read: select - Timeout before any socket was ready!
D (108319) transport_base: poll_read: select - Timeout before any socket was ready!
D (108319) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read(): call timed out before data was ready!
D (108329) mqtt_client: Sent PING successful
D (109349) transport_base: poll_read: select - Timeout before any socket was ready!
D (109349) transport_base: poll_read: select - Timeout before any socket was ready!
D (109349) mqtt_client: esp_mqtt_handle_transport_read_error: transport_read(): call timed out before data was ready!
E (109359) mqtt_client: No PING_RESP, disconnected
D (109369) mqtt_client: Reconnect after 10000 ms
D (109369) event: running post MQTT_EVENTS:2 with handler 0x400d51d8 and context 0x3ffdcf6c on loop 0x3ffdce50
D (109379) myMQTT: Event dispatched from event loop base=MQTT_EVENTS, event_id=2
I (109389) myMQTT: MQTT_EVENT_DISCONNECTED 1
D (119389) mqtt_client: Reconnecting...
D (119389) event: running post MQTT_EVENTS:7 with handler 0x400d51d8 and context 0x3ffdcf6c on loop 0x3ffdce50
D (119389) myMQTT: Event dispatched from event loop base=MQTT_EVENTS, event_id=7
I (119399) myMQTT: MQTT_EVENT_BEFORE_CONNECT
D (119399) esp-tls: host:192.168.31.26: strlen 13
D (119409) esp-tls: [sock=51] Resolved IPv4 address: 192.168.31.26
D (119409) esp-tls: [sock=51] Connecting to server. HOST: 192.168.31.26, Port: 1883
D (119429) mqtt_client: Transport connected to mqtt://192.168.31.26:1883
D (119429) mqtt_client: Sending MQTT CONNECT message, type: 1, id: 0000
D (119499) mqtt_client: mqtt_message_receive: first byte: 0x20
D (119509) mqtt_client: mqtt_message_receive: read "remaining length" byte: 0x2
D (119509) mqtt_client: mqtt_message_receive: total message length: 4 (already read: 2)
D (119519) mqtt_client: mqtt_message_receive: read_len=2
D (119519) mqtt_client: mqtt_message_receive: transport_read():4 4
D (119529) mqtt_client: Connected
D (119529) event: running post MQTT_EVENTS:1 with handler 0x400d51d8 and context 0x3ffdcf6c on loop 0x3ffdce50
D (119539) myMQTT: Event dispatched from event loop base=MQTT_EVENTS, event_id=1
I (119549) myMQTT: MQTT_EVENT_CONNECTED
D (119549) mqtt_client: mqtt_enqueue id: 31646, type=8 successful
D (119559) outbox: ENQUEUE msgid=31646, msg_type=8, len=17, size=17
D (119569) mqtt_client: Sent subscribe topic=Y5mJL/PING, id: 31646, type=8 successful
D (119599) mqtt_client: mqtt_message_receive: first byte: 0x90
D (119599) mqtt_client: mqtt_message_receive: read "remaining length" byte: 0x3
D (119599) mqtt_client: mqtt_message_receive: total message length: 5 (already read: 2)
D (119609) mqtt_client: mqtt_message_receive: read_len=3
D (119609) mqtt_client: mqtt_message_receive: transport_read():5 5
D (119619) mqtt_client: msg_type=9, msg_id=31646
D (119619) mqtt_client: pending_id=31646, pending_msg_count = 1
D (119629) outbox: DELETED msgid=31646, msg_type=8, remain size=0
D (119639) mqtt_client: deliver_suback, message_length_read=5, message_length=5
D (119639) event: running post MQTT_EVENTS:3 with handler 0x400d51d8 and context 0x3ffdcf6c on loop 0x3ffdce50
D (119649) myMQTT: Event dispatched from event loop base=MQTT_EVENTS, event_id=3
I (119659) myMQTT: MQTT_EVENT_SUBSCRIBED, msg_id=31646
D (119669) mqtt_client: mqtt_enqueue id: 48336, type=3 successful
D (119669) outbox: ENQUEUE msgid=48336, msg_type=3, len=20, size=20
I (119679) MyAgri: mqtt_event_handler => dataRefresh
D (119689) mqtt_client: mqtt_enqueue id: 27541, type=3 successful
D (119689) outbox: ENQUEUE msgid=27541, msg_type=3, len=73, size=93
I (119699) MyAgri: wf8098 00000000,0.000,0,2,0,info,M207,2,1692253651,unsaved[58]
D (119699) mqtt_client: mqtt_message_receive: first byte: 0x40
[... 72 lines cut ...]
I (92) boot:  4 spiffs           Unknown data     01 82 006b0000 00140000
I (99) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=330e8h (209128) map
I (188) esp_image: segment 1: paddr=00043110 vaddr=3ffbdb60 size=051a4h ( 20900) load
I (196) esp_image: segment 2: paddr=000482bc vaddr=40080000 size=07d5ch ( 32092) load
I (210) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=b629ch (746140) map
I (479) esp_image: segment 4: paddr=001062c4 vaddr=40087d5c size=0ed30h ( 60720) load
I (516) boot: Loaded app from partition at offset 0x10000
[... cut ...]
I (549) cpu_start: Pro cpu start user code
I (549) cpu_start: cpu freq: 160000000 Hz
I (550) cpu_start: Application information:
I (554) cpu_start: Project name:     controller
I (559) cpu_start: App version:      18b9876-dirty
I (565) cpu_start: Compile time:     Aug 17 2023 13:56:24
I (571) cpu_start: ELF file SHA256:  81f5676df903c66e...
I (577) cpu_start: ESP-IDF:          5.0.2
I (582) cpu_start: Min chip rev:     v0.0
I (586) cpu_start: Max chip rev:     v3.99 
I (591) cpu_start: Chip rev:         v3.0
```

LHYYSY, 2024-01-05:
[image]
Hello, these errors will appear after I disconnect the link, but I can reconnect back.