espressif/esp-idf #2672: Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

xgenesys, 2018-11-06:
Hi @igrr ,

I am using the mqtt/tcp example to publish some data to mqtt broker (use WIFI), and I try to scan BLE packages also.

The scanning duration is infinite, any beacon package found, I will forward it to MQTT broker.

I got the issue as below:

```
ASSERT_PARAM(0 4), in lld_evt.c at line 1749
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40084729  PS      : 0x00060b34  A0      : 0x800863da  A1      : 0x3ffbe260
0x40084729: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x000e0001  A5      : 0x60008054
A6      : 0x3ffbdcd0  A7      : 0x60008050  A8      : 0x80084729  A9      : 0x3ffbe240
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  Q3     : 0xffffffff
Q4     : 0x00000000  Q5     : 0xfffffffb  SAR     : 0x00000004  EXCCQSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40084661  LEND    : 0x40084668  LCOUNT  : 0x00000000
0x40084661: r_assert_param at ??:?

0x40084668: r_assert_param at ??:?

Core 0 was running in ISR context:
EPC1    : 0x40090967  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40084729
0x40090967: vTaskEnterCritical at C:/msys32/home/thanh-IOT/esp/esp-idf/components/freertos/tasks.c:3564

0x40084729: r_assert_param at ??:?

Backtrace: 0x40084729:0x3ffbe260 0x400863d7:0x3ffbe280 0x40086fa6:0x3ffbe2a0 0x40087ee2:0x3ffbe2c0 0x4008901b:0x3ffbe2e0 0x40082609:0x3ffbe300 0x4000bfed:0x00000000
0x40084729: r_assert_param at ??:?

0x400863d7: r_lld_evt_end at ??:?

0x40086fa6: r_lld_evt_end_isr at ??:?

0x40087ee2: r_rwble_isr at ??:?

0x4008901b: r_rwbtdm_isr_wrapper at intc.c:?

0x40082609: _xt_lowint1 at C:/msys32/home/thanh-IOT/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

Core 1 register dump:
PC      : 0x400d3996  PS      : 0x00060334  A0      : 0x8009106c  A1      : 0x3ffbc350
0x400d3996: esp_vApplicationIdleHook at C:/msys32/home/thanh-IOT/esp/esp-idf/components/esp32/freertos_hooks.c:88

A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060021  A7      : 0x00000000  A8      : 0x3ffc8aa0  A9      : 0x3ffc8a64
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x00000000  Q3     : 0x00000001
Q4     : 0x00000003  Q5     : 0x00060023  SAR     : 0x00000000  EXCCQSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x400d3996:0x3ffbc350 0x40091069:0x3ffbc370 0x40091a15:0x3ffbc390
0x400d3996: esp_vApplicationIdleHook at C:/msys32/home/thanh-IOT/esp/esp-idf/components/esp32/freertos_hooks.c:88

0x40091069: prvIdleTask at C:/msys32/home/thanh-IOT/esp/esp-idf/components/freertos/tasks.c:3564

0x40091a15: vPortTaskWrapper at C:/msys32/home/thanh-IOT/esp/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6200
load:0x40078000,len:9404
load:0x40080400,len:7128
entry 0x400807a0
```

Could you please tell me where is the problem come from? I am new to ESP32.

Thank you very much,

negativekelvin, 2018-11-06:
You did not post your idf version

Alvin1Zhang, 2018-11-07:
Would you please help check whether the issue you reported is the same as #2530 [https://github.com/espressif/esp-idf/issues/2530](url) ?

sysizlayan, 2018-11-13:
I got Core panic errors when I try to reach a null pointer. Could you check if there is an uninitialized pointer?

gerekon, 2018-11-22:
@xgenesys Could you try steps described in [#2530](https://github.com/espressif/esp-idf/issues/2530#issuecomment-441006665)?

Alvin1Zhang, 2019-03-15:
@xgenesys Hi, could you help share if any updates for the issue? Thanks.

o-marshmallow, 2021-11-17:
Closing this issue. If the problem is still relevant, please open a new issue.

closed 2021-11-17 by o-marshmallow