espressif/esp-idf #2673: ssl_tls.c:5875 alloc(16717 bytes) failed

zhaochengwei, 2018-11-06:
[image]
As you can see, 16384 is TLS maximum message content length.But I met this problem, So what's the problem?
[image]

esp-idf v3.1

zhaochengwei, 2018-11-07:
> You probably ran out of free heap. You need 40k to do ssl with full length buffers.

So, how can I enlarge my full length buffers?

negativekelvin, 2018-11-07:
No, it means there is not enough memory available.  You can try some of the menuconfig option for asymmetric buffers or reduce your app memory usage or use wrover/psram.

zhaochengwei, 2018-11-07:
> No, it means there is not enough memory available. You can try some of the menuconfig option for asymmetric buffers or reduce your app memory usage or use wrover/psram.

OK，I solved the problem thank you !

closed 2018-11-07 by zhaochengwei

Jigar3690, 2020-11-18:
@zhaochengwei How did you resolved it? I am facing the same issue with exact same number of bytes!
`mbedtls: ssl_tls.c:6893 alloc(16717 bytes) failed` I did enough memory leak analysis already. Tried increasing the stack size as well. But no change.

VitoLeo92, 2022-11-22:
> > No, it means there is not enough memory available. You can try some of the menuconfig option for asymmetric buffers or reduce your app memory usage or use wrover/psram.
> 
> OK，I solved the problem thank you !

**How did you resolved it?**

HamzaHajeir, 2023-06-11:
I'm having the exact same issue,
the number is a preprocessor-defined.
My heap is very large at the moment of the call:

```log
                     > altcp_tls: TCP_WND is smaller than the RX decrypion buffer, connection RX might stall!
04:10:41.028 > H4AT:1: H=143412 M=110580 S=4684
04:10:41.029 > IDF/components/mbedtls/mbedtls/library/ssl_tls.c:3857: alloc(16717 bytes) failed
04:10:41.030 > mbedtls_ssl_setup failed
04:10:41.031 > tcp_close: closing in State: CLOSED
```

hitecSmartHome, 2024-02-21:
Facing the same issue. I also have enough max contigous heap block but calloc fails with mbedtls