espressif/esp-idf #2674: [request] option to turn on/off slave connection intervals in advertising packet

chegewara, 2018-11-06:
@Weijian-Espressif @Yulong-espressif 
At the moment we have all the time advertised `Slave connection interval range`  (0x12). As far as i can say not many devices advertising this packet and it takes 5 bytes. Can we have at least option similar to include_name and include_txpower to turn it off from advertising packet? Thanks.

lucashutchinson, 2018-11-06:
You can manually construct the advertising and scan response data and set this if you are wanting to optimise the data that you are sending.
I have done this in my application and it works great!

I create two arrays like:
```
uint8_t raw_adv_data[ESP_BLE_ADV_DATA_LEN_MAX];
uint8_t raw_scan_rsp_data[ESP_BLE_SCAN_RSP_DATA_LEN_MAX];
```
And then manually fill them with data like:
```
0x02,
ESP_BLE_AD_TYPE_FLAG,
(ESP_BLE_ADV_FLAG_GEN_DISC | ESP_BLE_ADV_FLAG_BREDR_NOT_SPT), // advertising flags
```

Then set the raw advertising and scan response data using:
`esp_ble_gap_config_adv_data_raw` and `esp_ble_gap_config_scan_rsp_data_raw`

The data fields that I use are:
```
ESP_BLE_AD_TYPE_FLAG
ESP_BLE_AD_MANUFACTURER_SPECIFIC_TYPE
ESP_BLE_AD_TYPE_INT_RANGE
ESP_BLE_AD_TYPE_NAME_CMPL
```

This allows me to have a name with a length of 23 bytes, and 24 bytes of manufacturer data (after the manufacturer ID)

Hope this helps

chegewara, 2018-11-06:
@lucashutchinson I am working with Kolban's c++ library to easy building applications, so creating raw advertising packet only because something is not what i want. This library is widely used by arduino community, do you see option that arduino programmers will build raw adwertising packets?

lucashutchinson, 2018-11-06:
Since it is using c++ im sure there could be some sort of class or template that could be created to allow users of the lib to create more custom adv data, using these functions in the back ground.

Weijian-Espressif, 2018-11-09:
@chegewara, You do not want to set `Slave connection interval range`, am i right? please set `min_interval` and `max_interval ` to zero.

chegewara, 2018-11-09:
@Weijian-Espressif I want to set it, but i dont want to have it included in advertising packet, or at least have control that it is included in advertising packet but excluded in scan response. Unless the logic requires to advertise it if its > 0. In this case i know what to do. Thanks.

chegewara, 2018-11-09:
Ok, i think i see the point. Thanks for answer.

closed 2018-11-09 by chegewara