espressif/esp-idf #2675: [feature request] mDNS support alias name

outermanle, 2018-11-07:
## Environment
- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Core (if using chip or module): [ESP-WROOM-32D]
- IDF version (3276a1316): 
- Development Env:		[Make]
- Operating System:		[Ubuntu]
- Power Supply:         [USB]

## Problem Description
mDNS not support alias name.

It will benefit for device management in the local network if supported.
We can access device with both unify name(company.local) and alias name(company_macSuffix.local) in different scene.

It is hard to distinguish many devices in the same local network.
For example, there are three devices in the local network.
If I want to login with http://company.local/ then I maybe login the random device(maybe power on firstly).
But if I  can login with another alias name(http://company_macSuffix.local/) then I can login the specify device.

### Expected Behavior
mdns support alias name which can be searched by others.
sethostname("company.local");
setaliasname("company_macSuffix.local"); // feature request

david-cermak, 2018-11-07:
Hi @outermanle 

Although, It is possible (and in fact quite easy) to add another A/AAA answer in the response  packet and thus having an alias hostname resolved, it may not conform to mDNS standards. @me-no-dev  Can you please comment?

Maybe you can address the issue by setting up some aliases locally using `avahi-alias` service? Having your hostname `company_macSuffix.local` and create an alias for each macSuffix to `company.local`?

outermanle, 2018-11-08:
@david-cermak 
Q: Having your hostname company_macSuffix.local and create an alias for each macSuffix to company.local?
A: Yes, it is convenient for us to manage devices in local network.

outermanle, 2018-11-09:
@david-cermak 
Any plan to support this feature?

david-cermak, 2018-11-09:
@outermanle 
No, we most likely will not implement this, have started an internal discussion about mDNS with multiple A records, but that does not seem to fit into esp32 mdns component.

More importantly, your use case is clearly against the spec, as it imposes multiple non-unique host names on local network; mdns employs mechanisms to solve these conflicts (see https://tools.ietf.org/html/rfc6762 -- see chapter 8 -- probing, conflicts; perhaps still acceptable after conflicts resolve...?)

Generally having multiple address records might not be directly in conflict with the RFC, this however is not a clean solution and not a way how mdns should be used (my view).
@me-no-dev Please comment!

outermanle, 2018-11-10:
@david-cermak 
I got it, Thanks a lot!

closed 2018-11-10 by outermanle

me-no-dev, 2018-11-11:
@outermanle This functionality is really out of spec and could cause issues on the network. Our mDNS server is required to pass Apple Bonjour certification, which is a really rigorous test. Sorry but we can not implement this feature and break specs.

phatpaul, 2021-04-29:
I have the same feature request for my product.  Why would it violate the spec if the alias is also subject to the same collision check as the hostname already is?  (Then only one of the devices would get the company.local, while the others get company-1.local, etc.)

Another way to look at it is to run 2 instances of the mDNS service on the same host with different hostnames.  Would that violate the spec?  Is that possible on esp32?

david-cermak, 2021-05-06:
@phatpaul We're not planning on supporting mdns alias names, but will support hostname/service delegation, which could be used  to delegate another hostname for its own address.

The mdns API would look like
```cpp
esp_err_t mdns_delegate_hostname_add(const char * hostname, const mdns_ip_addr_t *address_list);
```

phatpaul, 2021-05-07:
@david-cermak thanks for your reply.
hostname/service delegation sounds interesting.  How does delegation differ from an alias?  Can you point me to any description, example, or spec?  I didn't see anything in the RFC.

david-cermak, 2021-06-28:
@phatpaul This feature is already on master:

https://github.com/espressif/esp-idf/blob/master/components/mdns/include/mdns.h#L118-L133

> How does delegation differ from an alias?

Delegation is publishing a host/service for another device. This could be (mis)used as alias, too, as we could delegate a new host-name to our own IP. We'd have to register the got/lost IP event handlers to update the IP whenever it changes. 
Usage would be something similar to the example below, just publishing our own address.

https://github.com/espressif/esp-idf/blob/master/examples/protocols/mdns/main/mdns_example_main.c#L64-L70

> I didn't see anything in the RFC.

The specs (both RFC 6762 and RFC 6763) don't forbid advertising/answering A/AAAA records for another host and both `avahi` and `dns-sd` support this (Note, that my previous comment https://github.com/espressif/esp-idf/issues/2675#issuecomment-437355750 (2.5 yrs back) refers to the original author's use-case being out of spec)