espressif/esp-idf #2676: make issue ! while trying make for - esp-idf/examples/system/ulp_adc program.

conduire, 2018-11-07:
Hello I am a beginner in esp32 programming. When i am trying to perform make for the ulp and ulp_adc program on the ulp of my esp32 pico d4. I am getting the error "
 error: implicit declaration of function 'esp_deep_sleep_disable_rom_logging' [-Werror=implicit-function-declaration]
     esp_deep_sleep_disable_rom_logging(); // suppress boot messages". 
 shown in the screenshot below. Is there something I might be doing wrong ? I am not able to understand why. Could you provide some suggestions.

I have attached the error screenshot below.

[image]

mahavirj, 2018-11-08:
@conduire What is ESP-IDF version that you are trying this build with? Regarding build failure, you may refer (or cross-check) https://github.com/espressif/esp-idf/blob/master/components/esp32/sleep_modes.c#L699

chegewara, 2018-11-08:
https://github.com/espressif/esp-iot-solution/blob/master/documents/low_power_solution/esp32_ulp_co-processor_and_assembly_environment_setup_en.md

conduire, 2018-11-09:
Thank you for the reply @mahavirj  and @chegewara . It was the issue with the version, just installed esp-idf from the link by chegwara and the issue was resolved.

closed 2018-11-09 by conduire