espressif/esp-idf #2678: ADC Calibration Documentation Incomplete

vonnieda, 2018-11-07:
In the ESP-IDF docs at https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/adc.html it says "There are currently three sources of calibration values. The availability of these calibration values will depend on the type and production date of the ESP32 chip/module." but there is no information that describes which chips or modules have this, or how to get them. Is there any other information about this available?

I have asked in the forum to no response: https://esp32.com/viewtopic.php?f=2&t=7900

And others have asked as well: https://www.esp32.com/viewtopic.php?f=12&t=2334&start=10

Alvin1Zhang, 2019-03-22:
@vonnieda Sorry to get back to you so late and thanks for reporting this.

Dazza0, 2021-09-23:
@vonnieda The [documentation](https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/peripherals/adc.html#calibration-values) should be updated with a reference regarding how to read the date code on an ESP32 chip. But as a summary:

- ESP32 chips with Two Point calibration are for the most part rare. For most cases, users can assume that their chip will not have two point calibration
- ESP32 chips manufactured from 2018 onwards will have VREF calibration in eFuse by default

I'm closing this issue, feel free to reopen if you have further inquiries. Alternatively, you can also [contact our sales team](https://www.espressif.com/en/contact-us/sales-questions) directly for more specifics about chips and their supported eFuse versions.

closed 2021-09-23 by Dazza0