espressif/esp-idf #2679: Bluetooth Classic A2DP disconnecting after ~20 seconds

markingle, 2018-11-08:
## Environment

- Development Kit:      SparkFun ESP32 Thing
- Kit version: N/A
- Module or chip used:  ESP32-DOWD
- IDF version:  v3.2-dev-1425-gbd90f534f
- Build System:        Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:    macOS
- Power Supply:         USB

## Problem Description
I am working to get the Bluetooth Classic A2DP profile setup between the ESP32 and pair of JVC BT headset speakers (ear buds).  After standing up the BT stack and GAP successfully the two devices connect but eventually the A2DP connection is disconnected.  This does not occur with my desktop BT speaker.   This code is based on the A2DP Source example.

### Expected Behavior
This the log output when connecting to a desktop BT speaker (ANKER).  The A2DP connection does not drop.

I (9250) BT_AV: Found a target device, address 00:e0:4c:b9:19:a0, name SoundCore mini
I (9260) BT_AV: Cancel device discovery ...
I (9260) BT_AV: Device discovery stopped.
I (9260) BT_AV: a2dp connecting to peer: SoundCore mini
I (9270) BT_AV: bt_app_gap_cb A2DP SOURCE CONNECTION SUCCESS

E (9280) BT_APPL: reset flags
I (9280) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
E (12280) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (12640) BT_L2CAP: L2CAP - rcvd conn req for unknown PSM: 27
W (12700) BT_APPL: new conn_srvc id:18, app_id:0
I (12700) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
I (12700) BT_AV: a2dp connected
E (21850) BT_APPL: bta_dm_pm_btm_status hci_status=12
E (30850) BT_APPL: bta_dm_pm_btm_status hci_status=12
E (39850) BT_APPL: bta_dm_pm_btm_status hci_status=12
....
...
..

### Actual Behavior
I (35320) BT_AV: Found a target device, address c4:45:67:21:26:d2, name JVC HA-FX9BT
I (35330) BT_AV: Cancel device discovery ...
I (35330) BT_AV: Device discovery stopped.
I (35340) BT_AV: a2dp connecting to peer: JVC HA-FX9BT
I (35340) BT_AV: bt_app_gap_cb A2DP SOURCE CONNECTION SUCCESS

E (35350) BT_APPL: reset flags
I (35350) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
E (36590) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (36750) BT_APPL: new conn_srvc id:18, app_id:0
I (36760) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
I (36760) BT_AV: a2dp connected
E (43980) BT_APPL: bta_dm_pm_btm_status hci_status=12
W (78050) BT_AVCT: ccb 0 not allocated
E (78050) BT_APPL: bta_av_rc_create found duplicated handle:0
I (78050) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x0
I (78050) BT_AV: a2dp disconnected

The A2DP disconnects after about 20 seconds and just after the ".....state 5, eat 0x0"

### Steps to repropduce

1. Compile code after changing the following line....
if (strcmp((char *)peer_bdname, "JVC HA-FX9BT") != 0) {     //JVC HA-FX9BT   SoundCore mini

### Code to reproduce this issue

I am not sure where the issue might be occurring in the code but I will gladly paste any function(s) that are needed

## Debug Logs
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6072
load:0x40078000,len:10084
load:0x40080400,len:6552
entry 0x40080764
I (31) boot: ESP-IDF v3.2-dev-1425-gbd90f534f 2nd stage bootloader
I (31) boot: compile time 20:05:30
I (39) boot: Enabling RNG early entropy source...
I (40) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00180000
I (82) boot:  3 storage          Unknown data     01 82 00190000 00100000
I (90) boot: End of partition table
I (94) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2a824 (174116) map
I (164) esp_image: segment 1: paddr=0x0003a84c vaddr=0x3ff80000 size=0x00000 (     0) load
I (164) esp_image: segment 2: paddr=0x0003a854 vaddr=0x3ff80000 size=0x00000 (     0) load
I (170) esp_image: segment 3: paddr=0x0003a85c vaddr=0x3ffbdb60 size=0x02c90 ( 11408) load
I (184) esp_image: segment 4: paddr=0x0003d4f4 vaddr=0x3ffc07f0 size=0x00000 (     0) load
I (188) esp_image: segment 5: paddr=0x0003d4fc vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/user/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (198) esp_image: segment 6: paddr=0x0003d904 vaddr=0x40080400 size=0x0270c (  9996) load
I (210) esp_image: segment 7: paddr=0x00040018 vaddr=0x400d0018 size=0x8e6a8 (583336) map
0x400d0018: _flash_cache_start at ??:?

I (419) esp_image: segment 8: paddr=0x000ce6c8 vaddr=0x40082b0c size=0x0fe2c ( 65068) load
0x40082b0c: get_cached_log_level at /Users/user/esp/esp-idf/components/log/log.c:121
 (inlined by) esp_log_write at /Users/user/esp/esp-idf/components/log/log.c:199

I (447) esp_image: segment 9: paddr=0x000de4fc vaddr=0x400c0000 size=0x00000 (     0) load
I (447) esp_image: segment 10: paddr=0x000de504 vaddr=0x50000000 size=0x00000 (     0) load
I (453) esp_image: segment 11: paddr=0x000de50c vaddr=0x50000000 size=0x00000 (     0) load
I (473) boot: Loaded app from partition at offset 0x10000
I (473) boot: Disabling RNG early entropy source...
I (474) cpu_start: Pro cpu up.
I (478) cpu_start: Starting app cpu, entry point is 0x4008106c
0x4008106c: call_start_cpu1 at /Users/user/esp/esp-idf/components/esp32/cpu_start.c:245

I (0) cpu_start: App cpu up.
I (488) heap_init: Initializing. RAM available for dynamic allocation:
I (495) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (501) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (507) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (513) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (519) heap_init: At 3FFCF220 len 00010DE0 (67 KiB): DRAM
I (526) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (532) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (538) heap_init: At 40092938 len 0000D6C8 (53 KiB): IRAM
I (545) cpu_start: Pro cpu start user code
I (227) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (270) BTDM_INIT: BT controller compile version [b4633da]
I (270) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (350) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 1, 0
I (860) BT_AV: Starting device discovery...
I (860) BT_AV: Discovery started.
I (870) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 1| Intr:3 
I (870) gpio: GPIO[22]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (1430) BT_AV: Scanned device: b8:81:98:5c:92:0b
I (1430) BT_AV: --Class of Device: 0xa010c
I (1430) BT_AV: --RSSI: -81
I (6320) BT_AV: Scanned device: 6c:56:97:0a:45:c9
I (6320) BT_AV: --Class of Device: 0x2c0414
I (6320) BT_AV: --RSSI: -91
I (6730) BT_AV: Scanned device: b8:81:98:5c:92:0b
I (6730) BT_AV: --Class of Device: 0xa010c
I (6730) BT_AV: --RSSI: -82
I (7060) BT_AV: Scanned device: c4:45:67:21:26:d2
I (7060) BT_AV: --Class of Device: 0x240404
I (7060) BT_AV: --RSSI: -45
I (7060) BT_AV: Found a target device, address c4:45:67:21:26:d2, name JVC HA-FX9BT
I (7070) BT_AV: Cancel device discovery ...
I (7080) BT_AV: Device discovery stopped.
I (7080) BT_AV: a2dp connecting to peer: JVC HA-FX9BT
I (7090) BT_AV: bt_app_gap_cb A2DP SOURCE CONNECTION SUCCESS

E (7090) BT_APPL: reset flags
I (7100) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
E (9330) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
W (9540) BT_APPL: new conn_srvc id:18, app_id:0
I (9540) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
I (9540) BT_AV: a2dp connected
E (16780) BT_APPL: bta_dm_pm_btm_status hci_status=12
W (50830) BT_AVCT: ccb 0 not allocated
E (50830) BT_APPL: bta_av_rc_create found duplicated handle:0
I (50830) BT_AV: bt_app_av_sm_hdlr state 5, evt 0x0
I (50840) BT_AV: a2dp disconnected

## Other items if possible

blueMoodBHD, 2018-11-09:
Hi @markingle, This issue seems like a bug we had fixed. Please update your ESP-IDF and test again. Thanks.

markingle, 2018-11-16:
I will be testing this weekend to confirm that latest ESP-IDF fixes this issue.

markingle, 2018-12-03:
Updating the ESP-IDF did not fix this issue.  I will keep searching the for issue

yunling-cn, 2019-06-23:
Hello, how do you implement the connection to the Bluetooth headset? I use ESP32 to connect to the Bluetooth headset and always stuck in "esp_a2d_source_connect". I just modified the official example a2dp_source. But connecting a computer can be, that is, connecting a Bluetooth headset does not work. Sometimes it suddenly crashes and then restarts and suddenly connects.

aadnesd, 2019-06-28:
I got my headset connected by just cloning the example and changing the name of my headset to "ESP_SPEAKER". Now the only problem is finding out how to play files through in stead of just a random signal. Please let me know if you have any idea.

mohnazemi, 2019-09-14:
@blueMoodBHD 
I am facing the same exact issue with a Bluetooth speaker even though I am using the stable IDF version 3.3. Any ideas ??

I (178) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (246) BTDM_INIT: BT controller compile version [e736f10]
I (246) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (346) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 0
I (886) BT_AV: Starting device discovery...
I (906) BT_AV: Discovery started.
I (4346) BT_AV: Scanned device: 8a:7a:0a:cf:3e:14
I (4346) BT_AV: --Class of Device: 0x240404
I (4346) BT_AV: --RSSI: -49
I (4596) BT_AV: Scanned device: 8a:7a:0a:cf:3e:14
I (4596) BT_AV: --Class of Device: 0x240404
I (4596) BT_AV: --RSSI: -49
I (4596) BT_AV: Found a target device, address 8a:7a:0a:cf:3e:14, name DY22
I (4606) BT_AV: Cancel device discovery ...
I (4606) BT_AV: Device discovery stopped.
I (4616) BT_AV: a2dp connecting to peer: DY22

E (4616) BT_APPL: reset flags

I (4626) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
I (10886) BT_AV: bt_app_av_sm_hdlr state 4, evt 0xff00
I (20886) BT_AV: bt_app_av_sm_hdlr state 4, evt 0xff00
I (30886) BT_AV: bt_app_av_sm_hdlr state 3, evt 0xff00
I (30886) BT_AV: a2dp connecting to peer: 8a:7a:0a:cf:3e:14
I (40886) BT_AV: bt_app_av_sm_hdlr state 4, evt 0xff00
W (41466) BT_SDP: SDP - Rcvd conn cnf with error: 0x8  CID 0x40

W (41466) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2

I (41466) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
I (50886) BT_AV: bt_app_av_sm_hdlr state 3, evt 0xff00
I (50886) BT_AV: a2dp connecting to peer: 8a:7a:0a:cf:3e:14
I (50886) BT_AV: bt_app_av_sm_hdlr state 4, evt 0x0
I (60886) BT_AV: bt_app_av_sm_hdlr state 4, evt 0xff00

Gabriel-Gardin, 2020-04-03:
Same problem here... Any fix? I get this error with esp-idf version 4.1 and also with 3.2.2

Wth-Esp, 2020-06-19:
Hi all,

Please pull the latest master and try to reproduce this issue. If it still exist, we will look into this as soon as possible.

Thanks
Weitianhua

KirillYatsenko, 2020-09-18:
> I got my headset connected by just cloning the example and changing the name of my headset to "ESP_SPEAKER". Now the only problem is finding out how to play files through in stead of just a random signal. Please let me know if you have any idea.

@aadnesd if you still interesting in playing songs - I managed to do this 
https://github.com/KirillYatsenko/mp3-bluetooth

Wth-Esp, 2021-01-21:
Hi all

We have tested the speakers we have, but no such issue happens.
So, please provide the model of your speaker devices.

We cannot find a **JVC HA-FX9BT** now. And does **DY22** is a speaker model's name?

Thanks

Alvin1Zhang, 2021-01-28:
Thanks for reporting, feel free to reopen.

closed 2021-01-28 by Alvin1Zhang

xiongweichao, 2021-01-28:
Hi all

We have tested the DY22 speaker and repropduce the issue in ESP-IDF v3.2.2.  We have fixed this issue.  Please update your ESP-IDF and test again.

Thanks