espressif/esp-idf #2685: no congest event triggered

chegewara, 2018-11-08:
Hi @Weijian-Espressif 
i dont know if this normal, weird or unexpected (issue) behavior. I am trying for test purpose to trigger congest event but instead i am getting btuT and btc_Task watchdog. My procedure is as follows:
- create esp32 peripheral server app,
- on connect with central send `esp_ble_gap_update_conn_params` with min and max set to 0x30 (60ms), it works i can confirm with nRF connect,
- on trigger start sending notifications in very tight loop (set value(16 bytes) -> send notification 
->vTaskDelay(3ms)),
- with connection parameters set to 60ms i would expect to trigger congestion event almost immediately,
- i am also priniting logs from every gatts and gap event, here is log:
```
W (101488) SampleServer: custom gatts event handler, GAP event: 5
W (101495) SampleServer: custom gatts event handler, GAP event: 5
W (101502) SampleServer: custom gatts event handler, GAP event: 5
W (101508) SampleServer: custom gatts event handler, GAP event: 5
W (101515) SampleServer: custom gatts event handler, GAE (101520) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (101520) task_wdt:  - IDLE0 (CPU 0)
E (101520) task_wdt: Tasks currently running:
E (101520) task_wdt: CPU 0: Btc_task
E (101520) task_wdt: CPU 1: IDLE1
P event: 5
W (101550) SampleServer: custom gatts event handler, GAP event: 5
W (101556) SampleServer: custom gatts event handler, GAP event: 5
W (101563) SampleServer: custom gatts event handler, GAP event: 5
W (101570) SampleServer: custom gatts event handler, GAP event: 5
W (101577) SampleServer: custom gatts event handler, GAP event: 5
```

chegewara, 2018-11-08:
I just want to add that indicate works perfect. With connection params set to 60 ms i am able to send indicate every 120ms (60ms send, 60ms receive confirm) even if again i have tight loop 3ms. Only thing that bothers me is conf_evt being triggered by bt stack when indication is sent, but no event triggered when peer device sends indication confirmation:
```
W (111554) handleGATTServerEvent: m_handle = 42, conf->handle = 0
W (111554) SampleServer: custom gatts event handler, event: 5
W (111559) LOOP: send indicate

W (111674) handleGATTServerEvent: m_handle = 42, conf->handle = 0
W (111674) SampleServer: custom gatts event handler, event: 5
W (111679) LOOP: send indicate

W (111794) handleGATTServerEvent: m_handle = 42, conf->handle = 0
W (111794) SampleServer: custom gatts event handler, event: 5
W (111799) LOOP: send indicate
```

chegewara, 2018-11-14:
@Weijian-Espressif I am a little bit confused. I can accept your response but I am sending indication and getting conf_evt in very same moment(the same ms), There is no time for congest because my next indication is sent after 3 ms, also there is too early to get confirmation from peer device. 

Again, sorry for being pain in eye and thanks fr support.

Weijian-Espressif, 2018-11-14:
@chegewara,  you can refer to `bta_gatts_indicate_handle()` and `gatts_process_value_conf()`.

If  you send a notification , `ESP_GATTS_CONF_EVT` is triggered in `bta_gatts_indicate_handle()`.
If you send a  indication ,  `ESP_GATTS_CONF_EVT` is triggered in `gatts_process_value_conf()` when device receive the confirmation.

chegewara, 2018-11-14:
I will, but again, there is no time to receive conf from peer device in this case:
`If you send an indication , ESP_GATTS_CONF_EVT is triggered in gatts_process_value_conf() when device receive the confirmation.`

chegewara, 2018-11-18:
@Weijian-Espressif Its me again. Maybe i am wrong, but as i understand we have connection interval which is basically min time between two massages send from one ble device to other. Min allowed connection interval is 7.5 ms. So, no matter what there should not be possible to send indications, notification read or write requests more often than 7.5 ms. Can anyone explain me this situation:
```
nRF Connect, 2018-11-18
UART (30:AE:A4:43:A1:72)
V	16:15:27.432	Updating server database...
D	16:15:27.432	server.clearServices()
I	16:15:27.456	[Server] Database changed
V	16:15:27.457	[Server] No services
V	16:16:44.654	Updating server database...
D	16:16:44.654	server.clearServices()
D	16:16:44.681	server.addService(6e400001-b5a3-f393-e0a9-e50e24dcca9e)
D	16:16:44.701	[Server callback] Service 6e400001-b5a3-f393-e0a9-e50e24dcca9e added with status: 0
I	16:16:44.740	[Server] Database changed
V	16:16:44.756	Nordic UART Service (6e400001-b5a3-f393-e0a9-e50e24dcca9e)
- RX Characteristic [W] (6e400002-b5a3-f393-e0a9-e50e24dcca9e)
- TX Characteristic [N] (6e400003-b5a3-f393-e0a9-e50e24dcca9e)
 Client Characteristic Configuration (0x2902)
D	16:18:33.524	gatt.close()
D	16:18:33.527	wait(200)
V	16:18:33.729	Connecting to 30:AE:A4:43:A1:72...
D	16:18:33.729	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	16:18:34.001	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	16:18:34.002	[Server] Device with address 30:AE:A4:43:A1:72 connected
D	16:18:34.013	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	16:18:34.014	Connected to 30:AE:A4:43:A1:72
D	16:18:34.038	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	16:18:34.055	Discovering services...
D	16:18:34.055	gatt.discoverServices()
I	16:18:34.462	Connection parameters updated (interval: 11.25ms, latency: 0, timeout: 10000ms)
I	16:18:34.574	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	16:18:34.707	[Callback] Services discovered with status: 0
I	16:18:34.707	Services discovered
V	16:18:34.731	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
 Client Characteristic Configuration (0x2902)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Central Address Resolution [R] (0x2AA6)
Nordic UART Service (6e400001-b5a3-f393-e0a9-e50e24dcca9e)
- TX Characteristic [N] (6e400003-b5a3-f393-e0a9-e50e24dcca9e)
 Client Characteristic Configuration (0x2902)
- RX Characteristic [W] (6e400002-b5a3-f393-e0a9-e50e24dcca9e)
D	16:18:34.732	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	16:18:34.739	gatt.setCharacteristicNotification(6e400003-b5a3-f393-e0a9-e50e24dcca9e, true)
I	16:18:34.775	Connection parameters updated (interval: 11.25ms, latency: 0, timeout: 10000ms)
I	16:18:34.862	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.862	"test" received
I	16:18:34.864	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.864	"test" received
I	16:18:34.865	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.865	"test" received
I	16:18:34.865	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.865	"test" received
I	16:18:34.866	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.866	"test" received
I	16:18:34.868	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.868	"test" received
I	16:18:34.869	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.869	"test" received
I	16:18:34.869	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.869	"test" received
I	16:18:34.870	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.870	"test" received
I	16:18:34.870	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.870	"test" received
I	16:18:34.871	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.871	"test" received
I	16:18:34.872	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.872	"test" received
[... 12 lines cut ...]
I	16:18:34.876	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.876	"test" received
I	16:18:34.876	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.876	"test" received
I	16:18:34.877	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.877	"test" received
I	16:18:34.879	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.879	"test" received
I	16:18:34.880	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.880	"test" received
I	16:18:34.880	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.880	"test" received
[... cut ...]
A	16:18:34.881	"test" received
I	16:18:34.881	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.881	"test" received
I	16:18:34.882	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
A	16:18:34.882	"test" received
I	16:18:34.882	Notification received from 6e400003-b5a3-f393-e0a9-e50e24dcca9e, value: (0x) 74-65-73-74, "test"
```
Most notifications is sent with every 1 ms and few is sent in the same ms.

chegewara, 2018-11-18:
In addition i performed test with conn interval set to 3990 ms, all read/write operations are long as hell, but notifications are sent like in above logs (i have 5 notivications with exactly the same time stamp in nRF).

Weijian-Espressif, 2018-11-19:
@chegewara, Notification do not need response.

chegewara, 2018-11-19:
@Weijian-Espressif I know, im not saying about response anywhere here. Maybe my understanding is wrong and connection interval parameter does not matter when we send notifications and notifications can be flooded as fast as device can send.

Weijian-Espressif, 2018-11-19:
[image]

BLE can send multiple packets in a connection interval.

chegewara, 2018-11-19:
@Weijian-Espressif Thank You.

closed 2018-11-19 by chegewara