espressif/esp-idf #2687: Core panic on OTA over BLE (Cache disabled but cached memory region accessed)

dsa-t, 2018-11-09:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    v3.2-beta1-34-gfb7ba1b
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    1.22.0-80-g6c4433a5
- Operating System:     [Linux]
- Power Supply:         [external 3.3V]

## Problem Description

I started using btstack library with esp32 and encountered a crash which happens about 1/2 times during OTA upgrade over BLE.

I am using custom board with ESP32-D0WDQ6, 4M flash and stable power supply.

Here's the test project file:
[BTStackCrash.zip](https://github.com/espressif/esp-idf/files/2565804/BTStackCrash.zip)

Before compiling you have to extract btstack component into your esp-idf components:
[btstack.zip](https://github.com/espressif/esp-idf/files/2565809/btstack.zip)

Then it can be compiled as usual (bin and elf files are also already included).
To run the OTA script, you need to have Python 3 and bluepy.
Note that bluepy only supports Linux.

After the device has started, find out the device address and run the script:
`python3 blecontrol.py <device address> otaupgrade build/btstack-crash-test.bin`

The OTA upgrade will start and sometimes esp32 will crash during the process. About 1/2 of OTA upgrades are successful.

The crash happens when the following loop is executed in cache_utils.c:
```
    while (!s_flash_op_complete) {
        // busy loop here and wait for the other CPU to finish flash operation
    }
```

The issue look similar to #1697

## Debug Logs

```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400f54bc  PS      : 0x00060534  A0      : 0x80044715  A1      : 0x3ffbe0d0  
0x400f54bc: r_lld_get_mode at ??:?

A2      : 0x00000002  A3      : 0x3ffcffc4  A4      : 0x00000001  A5      : 0x3ffcfebe  
A6      : 0x3ffcfebc  A7      : 0x00000000  A8      : 0x8004297a  A9      : 0x3ffae70c  
A10     : 0x00000000  A11     : 0x400f54bc  A12     : 0x3ff98af0  A13     : 0x00000021  
0x400f54bc: r_lld_get_mode at ??:?

A14     : 0x00000074  A15     : 0x3ffbe858  SAR     : 0x0000001e  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core 0 was running in ISR context:
EPC1    : 0x4008202f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400f54bc
0x4008202f: portENTER_CRITICAL_NESTED at /home/alex/esp/esp-idf/components/freertos/port.c:403
 (inlined by) xPortInIsrContext at /home/alex/esp/esp-idf/components/freertos/port.c:338

0x400f54bc: r_lld_get_mode at ??:?

Backtrace: 0x400f54bc:0x3ffbe0d0 0x40044712:0x3ffbe110 0x40087822:0x3ffbe140 0x4008612d:0x3ffbe170 0x40086d82:0x3ffbe190 0x40087ac6:0x3ffbe1b0 0x400888ab:0x3ffbe1d0 0x400823d5:0x3ffbe1f0 0x4008334e:0x00000000
0x400f54bc: r_lld_get_mode at ??:?

0x40087822: r_lld_pdu_rx_handler at ??:?

0x4008612d: r_lld_evt_end at ??:?
[... 153 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
8Kv7P1Cq+z/Yq/s/
UKr7P3Cr+z/Ozs7O7Hn7P7wU/D/wq/s/tBT8PwEAAAAAf/s/AH/7P/Cr+z/4fvs/
GAAAANyj+z9pcGMxAM7Ozs7Ozs7Ozs4AAQAAANir+z8AAAAAIQAGAAMAAADOzs7O
GAAAAAAAAAAAAAAAAAAAAAAAAACwZfs/GGb7P4Bm+z8AAAAAAAAAAAEAAAAAAAAA
d9hAPwAAAABIHQBAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAM7Ozg==
hCEIQLjACEAwAgYA+h0IgBCr+z/Ufvs/AAAAACR/+z8AAAAA1H77PwAAAAC4wAiA
8Kr7PwEAAAD/////MBf8PwEAAABE7Ps/AAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAH0jCEABAAAAQOAIQCC1+T8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAEAAAABAAAAq6sAAOCj+z8AAAAAUKv7P0wU/D8AAAAA
AAAAABQAAAAAAAAA/////wAAAABQq/s/AAAAAAgAAABMfvs/AAAAAAEAAAAAAAAA
AAAAAHCr+z8AAAAAAAAAADAzCEABAAAAAAAAAAAAAAD6EQiAgH3+PygAAAAoAAAA
AAAAAAAAAAB8q/s/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAA=
================= CORE DUMP END =================
E (11888) esp_core_dump: Skipped 1 tasks with bad TCB!
E (11888) esp_core_dump: Crashed task has been skipped!
CPU halted.
```

chegewara, 2018-11-09:
https://github.com/espressif/esp-idf/issues/2512

chegewara, 2018-11-09:
@dsa-t I know there is GATT service related to firmware update (at least i think i saw one), but i cant find it now. Could de give me a hint what is the UUID? Or you are using random UUID's?

dsa-t, 2018-11-09:
I am using Custom Characteristics:

OTA write (and current block read) Characteristic: 
8880f7bd-c968-41bb-9977-3cb3d7e204f6

Send Command Characteristic: 
67f35d72-6a95-4072-a8b6-6a28ec9159c6

Service:
4fafc201-1fb5-459e-8fcc-c5c9c331914b

Weijian-Espressif, 2018-11-14:
@dsa-t , We have fixed the bug,  please update to the latest master and try. we will merge it into v3.2 this week.

dsa-t, 2018-11-15:
Unfortunately I cannot compile the latest master of https://github.com/espressif/esp32-bt-lib:
```
.../esp-idf/components/bt/lib/libbtdm_app.a(dbg_task.o): In function `hci_dbg_common_echo_cmd_handler':
dbg_task.c:(.text+0x54): undefined reference to `dbg_default_state'
collect2: error: ld returned 1 exit status
...
```

chegewara, 2018-11-15:
@dsa-t I am running app compiled about hour ago with master. Before i compiled app i did git pull, but ive been already up to date with master. Stupid question but did you try `git submodule update --init` in esp-idf and `make clean` in your app?

dsa-t, 2018-11-15:
The master of https://github.com/espressif/esp-idf links to https://github.com/espressif/esp32-bt-lib/tree/16f95952efc63017167c863a7eadd2e8dfd27a61
which is old.
If I update the bt lib with `git submodule update --remote components/bt/lib`, I get the above error.
I always do `make clean`

Weijian-Espressif, 2018-11-16:
@dsa-t, Sorry, our CI has a problem, the code is not synced to GitHub yet.  If the problem is solved, I will notify you in time.

dsa-t, 2018-11-17:
I added `int dbg_default_state = 0;` into my app and I was able to compile.
Left it upgrading in a loop for a night with this one: https://github.com/espressif/esp32-bt-lib/tree/b2c7769cac97ffda9f30e33b89bcce098dd9efb2
It upgraded successfully at least 40 times in a row without a crash.
Thanks! But the problem with dbg_default_state is still there.

dsa-t, 2018-11-27:
Closing as I haven't encountered this bug ever since

closed 2018-11-27 by dsa-t

chegewara, 2018-11-29:
Hi @Weijian-Espressif 
can it be that my crash is related to this one?

Not because its also from my own OTA over BLE app, but because guru meditation. My app works good so far, but this crash is after ota update and restart esp32. When i ota flash hello-world example i dont have crash, but this crash is when i ota flash the same binary that is flashed with USB cable and which is for sure working:
```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x40173da4  PS      : 0x00060f34  A0      : 0x800466c6  A1      : 0x3ffbe230
0x40173da4: r_platform_reset at ??:?

A2      : 0x00000000  A3      : 0x40173da4  A4      : 0x00000000  A5      : 0x00000010
0x40173da4: r_platform_reset at ??:?

A6      : 0x3ffb93cc  A7      : 0x3ffb8360  A8      : 0x80019fb8  A9      : 0x00002068
A10     : 0xf2f2f2f2  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000022
A14     : 0x000015fc  A15     : 0x00000000  SAR     : 0x00000012  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x4008ff5b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40173da4
0x4008ff5b: portENTER_CRITICAL_NESTED at D:/msys32/home/imper/esp/esp-idf/components/freertos/port.c:435
 (inlined by) xPortInIsrContext at D:/msys32/home/imper/esp/esp-idf/components/freertos/port.c:338

0x40173da4: r_platform_reset at ??:?

Backtrace: 0x40173da4:0x3ffbe230 0x400466c3:0x3ffbe260 0x40088aa2:0x3ffbe280 0x400896aa:0x3ffbe2a0 0x4008a3e6:0x3ffbe2c0 0x4008b1cb:0x3ffbe2e0 0x40084d
bd:0x3ffbe300 0x400862e4:0x00000000
0x40173da4: r_platform_reset at ??:?

0x40088aa2: r_lld_evt_end at ??:?

0x400896aa: r_lld_evt_end_isr at ??:?

0x4008a3e6: r_rwble_isr at ??:?

0x4008b1cb: r_rwbtdm_isr_wrapper at intc.c:?

0x40084dbd: _xt_lowint1 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

0x400862e4: spi_flash_op_block_func at D:/msys32/home/imper/esp/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1)

CPU halted.
```

Weijian-Espressif, 2018-11-30:
@chegewara, Can you help me do some test? here is a new bt lib, please update to the latest IDF and replace the bt lib and try. please give me the crash log, thanks.
[2018-11-30_115027.zip](https://github.com/espressif/esp-idf/files/2631923/2018-11-30_115027.zip)

chegewara, 2018-11-30:
@Weijian-Espressif Sorry, but this lib is broken I did the same workaround as @dsa-t (thanks)):
```
D:/msys32/home/imper/esp/esp-idf/components/bt/lib\libbtdm_app.a(dbg_task.o): In function `hci_vendor_ble_long_adv_cmd_handler':
dbg_task.c:(.text+0x140): undefined reference to `dbg_default_state'
collect2.exe: error: ld returned 1 exit status
make: *** [/d/msys32/home/imper/esp/esp-idf/make/project.mk:436: /home/imper/ble_test/build/hello-world.elf] Error 1
```
This is log with new lib and workaround (esp-idf fb7ba1baa068a2267ef54b7105cf3c8a004fe4eb):
```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x400976e9  PS      : 0x00060f34  A0      : 0x80086f4d  A1      : 0x3ffbe1d0
0x400976e9: strcmp at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../..
/./newlib/libc/machine/xtensa/strcmp.S:472

A2      : 0x3f439a4c  A3      : 0x3ffbf44d  A4      : 0x00000001  A5      : 0x00000001
A6      : 0x00060f21  A7      : 0x00000000  A8      : 0x000000ad  A9      : 0x0000006c
A10     : 0x00000003  A11     : 0x00060f23  A12     : 0x00060f21  A13     : 0x40089628
0x40089628: r_lld_evt_end_isr at ??:?

A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000012  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x40090063  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400976e9
0x40090063: portENTER_CRITICAL_NESTED at D:/msys32/home/imper/esp/esp-idf/components/freertos/port.c:435
 (inlined by) xPortInIsrContext at D:/msys32/home/imper/esp/esp-idf/components/freertos/port.c:338

0x400976e9: strcmp at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../..
/./newlib/libc/machine/xtensa/strcmp.S:472

Backtrace: 0x400976e9:0x3ffbe1d0 0x40086f4a:0x3ffbe1e0 0x400870fd:0x3ffbe200 0x40019fb5:0x3ffbe220 0x40089fa7:0x3ffbe250 0x40088aa9:0x3ffbe280 0x400896
e2:0x3ffbe2a0 0x4008a4ee:0x3ffbe2c0 0x4008b2d3:0x3ffbe2e0 0x40084dbd:0x3ffbe300 0x400862d6:0x00000000
0x400976e9: strcmp at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../..
/./newlib/libc/machine/xtensa/strcmp.S:472

0x40086f4a: r_assert_param at ??:?

0x400870fd: r_platform_reset at ??:?

0x40089fa7: r_lld_pdu_rx_handler at ??:?

0x40088aa9: r_lld_evt_end at ??:?

0x400896e2: r_lld_evt_end_isr at ??:?

0x4008a4ee: r_rwble_isr at ??:?

0x4008b2d3: r_rwbtdm_isr_wrapper at intc.c:?

0x40084dbd: _xt_lowint1 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_vectors.S:1105

0x400862d6: spi_flash_op_block_func at D:/msys32/home/imper/esp/esp-idf/components/spi_flash/cache_utils.c:81

CPU halted.
```

chegewara, 2018-11-30:
I found it something else. After ota flash and SW reset i have crash described above, but when i do HW reset then app is running without crash:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6304
ho 0 tail 12 room 4
load:0x40078000,len:10280
ho 0 tail 12 room 4
load:0x40080400,len:6612
entry 0x40080764
I (32) boot: ESP-IDF v3.2-beta1-34-gfb7ba1baa-dirty 2nd stage bootloader
I (32) boot: compile time 08:57:08
I (34) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : DIO
I (47) boot: SPI Flash Size : 4MB
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (70) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (77) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  3 ota_0            OTA app          00 10 00010000 001c2000
I (92) boot:  4 ota_1            OTA app          00 11 001e0000 001c2000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=0x001e0020 vaddr=0x3f400020 size=0x518a0 (333984) map
I (179) esp_image: segment 1: paddr=0x002318c8 vaddr=0x3ff80000 size=0x00000 (     0) load
I (179) esp_image: segment 2: paddr=0x002318d0 vaddr=0x3ff80000 size=0x00000 (     0) load
I (186) esp_image: segment 3: paddr=0x002318d8 vaddr=0x3ffbdb60 size=0x03400 ( 13312) load
I (197) esp_image: segment 4: paddr=0x00234ce0 vaddr=0x3ffc0f60 size=0x00000 (     0) load
I (204) esp_image: segment 5: paddr=0x00234ce8 vaddr=0x40080000 size=0x00400 (  1024) load
I (213) esp_image: segment 6: paddr=0x002350f0 vaddr=0x40080400 size=0x0af20 ( 44832) load
I (231) esp_image: segment 7: paddr=0x00240018 vaddr=0x400d0018 size=0xca338 (828216) map
I (395) esp_image: segment 8: paddr=0x0030a358 vaddr=0x4008b320 size=0x0cbf4 ( 52212) load
I (407) esp_image: segment 9: paddr=0x00316f54 vaddr=0x400c0000 size=0x00000 (     0) load
I (407) esp_image: segment 10: paddr=0x00316f5c vaddr=0x50000000 size=0x00000 (     0) load
I (414) esp_image: segment 11: paddr=0x00316f64 vaddr=0x50000000 size=0x00000 (     0) load
I (427) boot: Loaded app from partition at offset 0x1e0000
I (429) boot: Disabling RNG early entropy source...
E (436) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (442) cpu_start: Failed to init external RAM; continuing without it.
I (449) cpu_start: Pro cpu up.
I (453) cpu_start: Starting app cpu, entry point is 0x400813e0
I (432) cpu_start: App cpu up.
D (463) memory_layout: Checking 12 reserved memory ranges:
D (468) memory_layout: Reserved memory range 0x3f800000 - 0x3fc00000
D (475) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (481) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10
D (488) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388
D (494) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20
D (500) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c
D (507) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffcc878
D (513) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (520) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (526) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (533) memory_layout: Reserved memory range 0x40078000 - 0x40080000
D (539) memory_layout: Reserved memory range 0x40080000 - 0x40097f11
D (545) memory_layout: Building list of available memory regions:
D (552) memory_layout: Available memory region 0x3ffaff10 - 0x3ffb0000
D (558) memory_layout: Available memory region 0x3ffb6388 - 0x3ffb8000
D (565) memory_layout: Available memory region 0x3ffb9a20 - 0x3ffbdb28
D (571) memory_layout: Available memory region 0x3ffbdb5c - 0x3ffbdb60
D (578) memory_layout: Available memory region 0x3ffcc878 - 0x3ffce000
D (585) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
D (591) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
D (598) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
[... 172 lines cut ...]
D (109) heap_init: New heap initialised at 0x3ffe4350
D (114) intr_alloc: Connected src 16 to int 12 (cpu 0)
I (1186) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 2
```
The only thing is, as you can see, some garbage in logs during boot. This garbage in logs is always after reset from ota flashed app:
```
eׁJ]
   8▒0L▒0:22:57

rst:0x1 (POWERON_RESET),boot:▒▒3BS▒}MQ}1M!}     ==Q▒
flash read err, 1000C▒▒_[▒ni▒371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FP*▒e1A*▒%=OU▒
                                                         ▒˙igsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6304
ho 0 tail 12 room 4
load:0x4007▒0bb▒▒▒10280
ho 0 tail 12 room 4
load:0x40080400,len:6612
entry 0x40080764
```

chegewara, 2018-11-30:
I have to fix few issues in my code first then i get back here.

chegewara, 2019-08-14:
https://github.com/espressif/esp-idf/issues/3923