espressif/esp-idf #2689: Program does not start on PICO V4

Reiner1210, 2018-11-11:
Hello,

my program works fin on ESP32 Core Board V2, but on PICO V4 it does not start automatically. After pressing the enable button the program starts. Any solution ?

costaud, 2018-11-12:
Do you mean you can successfully burn firmware to the chip, but failed to get it run ?

krzychb, 2018-11-12:
Hi @Reiner1210, also do you mean the issue described in https://www.esp32.com/viewtopic.php?t=5056 or it happens in different scenario? What is the sequence of steps to replicate?

Reiner1210, 2018-11-12:
Hello,

yes I have tried tried the 10k resistor between IO0 and +3.3V - no success.
I can burn the image, and after reset with make monitor it works. But only power it on 5V over USB does not start the firmware. After EN pressed it starts. So I think a reset logic is broken.

krzychb, 2018-11-12:
Hi @Reiner1210, I see the similar issue with my [ESP32-PICO-KIT V4](https://dl.espressif.com/dl/schematics/esp32-pico-kit-v4_schematic.pdf). Apparently IO0 is hold low too long by C6 on power up and the board enters application download boot mode.
* Adding 10k resistor between IO0 and +3.3V works for me.
* If it does not work for you, then you can try lowering this resistor to 5k
* Another option is removing C6, this is one of changes done in [ESP32-PICO-KIT V4.1](https://dl.espressif.com/dl/schematics/esp32-pico-kit-v4.1_schematic.pdf)

Reiner1210, 2018-11-12:
@krzychb  Thanks for this. A 4.7k resistor do the job. A 10k did not!

Reiner1210, 2018-11-13:
After a while without voltage - the same. It does not start.
I will use a normal dev board an put the Pico in the trash

Reiner1210, 2018-11-15:
Ok. Yesterday I removed C6 and I have still the 4.7k Ohm resistor - has worked. I will check if it works also today :-)

closed 2018-11-15 by Reiner1210