espressif/esp-idf #2691: Is there any way to disable automatic append of FCS for ethernet emac?

wayneuroda, 2018-11-12:
I wish to interface to an ethernet switch which uses tail tagging. An extra byte is inserted (by my software) just before the FCS, and the switch IC will remove it before it is sent to the LAN. This means that the FCS will be wrong, since it was computed with the tail tag present.

Is there a way I can disable automatic FCS inside the emac and instead compute and append my own?

Thanks

negativekelvin, 2018-11-12:
Check TRM 10.8.1 Transmit Descriptors

wayneuroda, 2018-11-14:
Thank you that looks like the information I need - stupidly I didn't even think to check the TRM.

closed 2018-11-14 by wayneuroda

suda-morris, 2018-11-14:
@wayneuroda Well， playing esp32 with switch phy sounds interesting. Enable/Disable checksum calculation and insertion when transmit by setting/reset CIC bits in TDES1。

wayneuroda, 2018-11-15:
It turns out that I didn't need to disable the inbuilt CRC and compute my own at all, the issue I was facing was due to frame padding - in effect I was adding the tail tag and then the emac hardware was adding padding, so the tag wasn't in the right place. I simply added the padding and tail tag before sending the packet to the emac and everything is working now. That was a hair pulling experience trying to figure that one out!