espressif/esp-idf #2694: Bug in emac_main.c => Packet loss

neoniousTR, 2018-11-12:
We are the makers of the neonious one microcontroller board. It uses an ESP32-WROVER and has an Ethernet port. The phy is the LAN8720.

Users were reporting very bad Ethernet connections.

With almost no traffic other the port, everything is fine. But when some packets go in and out, the desc0 of the emac receive descriptor gets weird frame length values. But still, the error flags of desc0 are not set. We read out the desc0 right before the calls of emac_config.emac_tcpip_input in emac_main.c.

First we thought it was an hardware problem, but it is a software problem because the problem disappears as soon as we enable CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE . Good for us, because everything is working now. But there is a bug, most probably in emac_main.c, which should be fixed.

suda-morris, 2018-11-13:
@neoniousTR  Could you tell me which idf version you are using? And do you have any idea that how to reproduce it?

neoniousTR, 2018-11-13:
The last commit we merged was 7efbe8178b9e25ac404c02cf6908474ab3918995 from Oct 29

Not sure how to reproduce the problem. It seemed to happen when lots of data was coming in from a single TCP socket over the Ethernet port, but we only consumed the data slowly (20 KBit/s).
But when the problem happens, it is easy to notice the problem: ping to the device via the Ethernet port start to get lost (and data over the socket stalls periodically).

closed 2019-03-15 by Alvin1Zhang