espressif/esp-idf #2697: BLE stack crash while repairing with PC

dhavalpatel0114, 2018-11-13:
I was running gatts_security_server example. It works fine with android devices but having some probleme with PC (windows).

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c

I can pair it once or twice after removing pairing device on PC. but some time during the pairing process BLE stack is crashing. 

And every time it crashing at same location [SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID ] while pairing with pc.

Below is attached log.

[1B][0;32mI (154767) SEC_GATTS_DEMO: c4 9d ed a8 cf 4b [1B][0m
[1B][0;32mI (154777) SEC_GATTS_DEMO: 50 7f af 53 e8 aa [1B][0m
[1B][0;32mI (159677) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13[1B][0m
[1B][0;32mI (159687) SEC_GATTS_DEMO: advertising start success[1B][0m
[1B][0;32mI (177357) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT[1B][0m
[1B][0;32mI (177927) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LENC[1B][0m
[1B][0;32mI (177927) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID[1B][0m
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x401284c0  PS      : 0x00060934  A0      : 0x80044715  A1      : 0x3ffbe1e0  
A2      : 0x00000000  A3      : 0x00000006  A4      : 0x3ffb30e4  A5      : 0x3ffb2fc6  
A6      : 0x3ffb2fc4  A7      : 0x3ffafd68  A8      : 0x80042786  A9      : 0x3ffae70c  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x3ff98af0  A13     : 0x00000021  
A14     : 0x000000b0  A15     : 0x3ffbe968  SAR     : 0x0000001a  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Core 0 was running in ISR context:
EPC1    : 0x4008d243  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x401284c0

Backtrace: 0x401284c0:0x3ffbe1e0 0x40044712:0x3ffbe220 0x400876fa:0x3ffbe250 0x40086005:0x3ffbe280 0x40086c5a:0x3ffbe2a0 0x4008799e:0x3ffbe2c0 0x40088783:0x3ffbe2e0 0x400825ad:0x3ffbe300 0x400621be:0x00000000

Rebooting...

Weijian-Espressif, 2018-11-14:
@dhavalpatel0114, we have fixed the bug in master,  please update to the latest master and try. we will merge it into v3.2 this week.

dhavalpatel0114, 2018-11-15:
I update to the latest master still having same problem.

[image]
 last time that code updated 2 months ago.

Weijian-Espressif, 2018-11-15:
@dhavalpatel0114, do you update bt lib ? Can you give me the bt version ? The latest Bt lib log is add `vendor hci cmd (4f362c59)`

dhavalpatel0114, 2018-11-15:
yes i am using the latest lib **[vendor hci cmd (4f362c59)]**  and updated example.

Weijian-Espressif, 2018-11-15:
@dhavalpatel0114, Can you give me the crash log and your example elf file?

dhavalpatel0114, 2018-11-15:
yes I am using latest bt lib **[vendor hci cmd (4f362c59)]** . It crashes when you remove and repair device from PC [in 2 -5 trials].

Weijian-Espressif, 2018-11-15:
@dhavalpatel0114, please  give me your crash log and your example elf file.

dhavalpatel0114, 2018-11-15:
yes 

please see attached
[sec_gatts_demo.zip](https://github.com/espressif/esp-idf/files/2583564/sec_gatts_demo.zip)
.

dhavalpatel0114, 2018-11-15:
crash log attached here

[crash log.txt](https://github.com/espressif/esp-idf/files/2583579/crash.log.txt)

Weijian-Espressif, 2018-11-15:
@dhavalpatel0114, from the crash log, `[1B][0;32mI (267) BTDM_INIT: BT controller compile version [bc0b51f]', your BT version is `bc0b51f`.  we have fix the bug in bt version `components/bt: Put function in isr into iram(1653746d)`. Our latest BT version is `add vendor hci cmd (4f362c59)`, please update to the latest BT lib. When you restart ESP32,  You can get the bt version in uart log.

dhavalpatel0114, 2018-11-16:
When I try to built with bt_lib `add vendor hci cmd (4f362c59)` . I got built error.
`dbg_task.c:(.text+0x54): undefined reference to `dbg_default_state'`

here is attached built error log.
[built_error_log.txt](https://github.com/espressif/esp-idf/files/2587584/built_error_log.txt)

dhavalpatel0114, 2018-11-17:
@Weijian-Espressif Thanks Let me know once it's available.