espressif/esp-idf #2698: Documentation: it's not clear where the WPA2 enterprise documentation is

detly, 2018-11-13:
## Environment

- Development Kit: Wemos Mini32
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it): v3.2-beta1-34-gfb7ba1baa
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 5.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

I'm looking at the example code for WPA2 enterprise connection in [`examples/wifi/wpa2_enterprise/`](https://github.com/espressif/esp-idf/tree/master/examples/wifi/wpa2_enterprise), and the documentation for [the Wifi API](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/wifi/index.html). However I can't find any documentation for the WPA2 enterprise API.

Issues are:

 * it is not clear where in the documentation hierarchy I should be navigating to (it is not under Wifi, Protocols, Peripherals)
 * it is not present in search results when searching for the function name or header file name

detly, 2018-11-13:
(See also this [forum question](https://www.esp32.com/viewtopic.php?f=13&t=7936).)

liuzfesp, 2018-11-20:
@detly we don't have document for WPA2 yet.

liuzfesp, 2018-11-21:
Sure, we will consider to add the WPA2 document in future. But we have our own priority, currently WPA2 document is in low priority.  Maybe we can add it in IDF v4.0 or v4.1, it should be half year later.

detly, 2018-11-21:
Thanks for letting me know. It's not urgent for me, but I'm sure it will help others.

closed 2021-01-05 by espressif-bot