espressif/esp-idf #2699: Documentation/examples: esp_event_loop_init() is used but not documented as part of the API

detly, 2018-11-14:
## Environment

- Development Kit: Wemos Mini32
- Module or chip used:  ESP32
- IDF version (run ``git describe --tags`` to find it): v3.2-beta1-34-gfb7ba1baa
- Build System: Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 5.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

Many of the examples register some sort of event handler with `esp_event_loop_init()` (eg. the softAP example). This function is in `esp_event_loop.c`. However, the documentation doesn't mention this function as part of the [event loop API](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/esp_event.html). In fact, it doesn't even seem to be in the documentation. The header file has a brief comment above it, but nothing on how this function fits in to the overall flow of the event system.

Does the documentation need to include it? Or is it part of a deprecated API and the example code needs to be updated? (Or at least just noted to avoid confusion?)

igrr, 2018-11-14:
In ESP-IDF up to (upcoming) version 3.2, `esp_event_loop_init` initializes the event loop used to dispatch event callbacks for Wi-Fi, IP, and Ethernet related events. It is indeed not documented, there only are a few Doxygen comments in https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_event_loop.h.

In ESP-IDF 3.3, we plan to start using the new event loop API to deliver Wi-Fi, Ethernet, and IP events. Existing code (which calls `esp_event_loop_init`) should keep working though, at least until IDF 4.0.

For now you need to keep using `esp_event_loop_init`, as it is demonstrated in Wi-Fi and Ethernet examples.

detly, 2018-12-20:
>In ESP-IDF 3.3, we plan to start using the new event loop API to deliver Wi-Fi, Ethernet, and IP events. 

Is this in `v3.3-beta1`?

igrr, 2018-12-20:
No, this has slipped due to some dependencies and will be merged after we start working on v4.0.

itay-grudev, 2019-02-18:
Could you please add some documentation in the mean time. At the very least possible return values. So we can expect errors and handle them appropriately.

igrr, 2019-08-21:
Some documentation got added, sorry for not updating earlier: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/event-handling.html. This applies to v4.0 and the master branch.

closed 2020-09-09 by igrr