espressif/esp-idf #2701: Possible memory leak in ethernetif_input (lwip/port/esp32/netif/ethernetif.c)

wayneuroda, 2018-11-15:
In the function `ethernetif_input`, I assume the passed in `buffer` must always eventually be freed with `esp_eth_free_rx_buf`, is this correct? (I can't see the code which calls it).

Right now, depending on the setting of `CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE` and other error conditions, there are code paths through `ethernetif_input` which don't eventually free the buffer.

Eg, 

        if(buffer== NULL || !netif_is_up(netif)) {
    #ifndef CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE
            if (buffer) {
                esp_eth_free_rx_buf(buffer);
            }
    #endif
            return;
        }

If the netif is down and the user has enabled `CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE`, then the buffer will be leaked?

Likewise, in the code which implements the `CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE` mode, a pbuf is allocated and the owner is set to NULL which is correct. Then the `buffer` is copied to the pbuf, but the `buffer` is not free afterwards. Is this a bug?

I don't use the `CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE` mode in my app, but somebody might, and I assume pretty bad things will happen eventually?

The only way I can see it isn't a bug is if the library code (the code which eventually calls `ethernetif_input`) doesn't require buffer to be freed when compiled with `CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE` set. And maybe that is the case, in which case, never mind!

suda-morris, 2018-11-15:
@wayneuroda `esp_eth_free_rx_buf` only need to be called when CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE is not set. And this function will be called in the pbuf_free function. What this function mainly does is clean the rx descriptor so that the `dma buffer` can be used again. If `CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE` is enabled and the netif is down, it won't cause any memory leak. Because in this mode the descriptor clean work is done in the `emac_process_rx` function. And since we don't have the chance to malloc a new pbuf to deliver the DMA Rx buffer, it's OK that we just return from this function. Maybe you're confused by the buffer which feeded to ethernetif_input function, actually this is just a pointer to a DMA rx buffer,it should not free during the life cycle of ethernet driver.

wayneuroda, 2018-11-15:
Thanks, I just made an incorrect assumption - all good!

closed 2018-11-15 by wayneuroda