espressif/esp-idf #2703: flex: can't open zconf - macOS Mojave

benjenq, 2018-11-15:
I follow this documents to setup ESP-IDF and got a error when test section "hello_world"
( https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-esp-idf )

```
$ cd ~/esp/hello_world
$ make menuconfig
cc -c  -DCURSES_LOC="<curses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -MD  /Users/administrator/esp/esp-idf/tools/kconfig/mconf.c -o mconf.o
flex -L -P zconf -o zconf.lex.c /Users/administrator/esp/esp-idf/tools/kconfig/zconf.l
flex: can't open zconf
make[1]: *** [zconf.lex.c] Error 1
Excelsius-5M:hello_world administrator$ 
```
How could I fix it? It seem an error similar like https://github.com/espressif/esp-idf/issues/1281#issuecomment-438414807

igrr, 2018-11-23:
Hi @benjenq sorry for the late response. Could you please try the following:

```
cd $IDF_PATH/tools/kconfig
flex -L -Pzconf -ozconf.lex.c zconf.l
```
 
and see if this runs without the "can't open" error?

benjenq, 2019-02-14:
Hi @igrr ,
It works!  Message "can't open" was disappear.

benjenq, 2019-02-14:
Hi @igrr ,
Thanks ! It works! Message "can't open" was disappear.
Sorry late to report.

igrr, 2019-02-14:
Thank you for the confirmation. Which OS are you using, and what is the version of flex (`flex --version)?

benjenq, 2019-02-19:
Hi @igrr 
macOS: 10.14.3 (18D109）Mojave
flex version 2.5.4

igrr, 2019-02-19:
It's odd that you have such an old version of flex on Mojave — I've got `flex 2.5.31 Apple(flex-31)` on the same version of OS X. In any case, the fix should be merged into IDF soon.

benjenq, 2019-02-21:
Hi @igrr 
I found there is an old version flex (2.5.4) exist in `/Applications/XAMPP/xamppfiles/bin/` which was  included in `$PATH` (found in `.bash_profile` file).

Flex ver will change to `flex 2.5.35 Apple(flex-31)` if I remove `/Applications/XAMPP/xamppfiles/bin/` from `$PATH`, or change $PATH order.

So, thanks for your information very much.

closed 2019-02-22 by igrr