espressif/esp-idf #2704: Requirements for esp_wifi_internal_tx() - new Stack

jhinklejr, 2018-11-15:
I removed lwip as the stack over in ESP8266 RTOS V3 IDF and ported in a different stack.

The function that bridged the stack and the internal wifi transmit required an additional 36 bytes preceding the data (used for wifi headers) and need to be allocated from the DRAM heap (IRAM would cause a crash).

I have removed lwif from the ESP32 IDF and am porting in the new stack -- I have questions.

Q1: It appears ESP32 does NOT require the additional preceding 36 bytes for wifi header as was required for ESP8266) -- is this correct? LWIF has PBUF_LINK_ENCAPSULATION_HLEN default to ZERO (0).

Q2: If no wifi header space is required for the data being passed in -- are you coping it into your own internal buffer?

Q3: Is there an alignment and heap CAP requirement associated with the data buffer being passed in? If you are coping the data from my buffer into an internal transmit buffer -- then there might not be a requirement associated with my data.

Thanks.

liuzfesp, 2018-11-20:
Hi @jhinklejr, my answers:

- Correct.
- Yes, WiFi driver makes a copy of high layer data.
- No,  WiFi driver has no alignment requirement.

jhinklejr, 2018-11-20:
Thanks

Please close

closed 2018-11-21 by Alvin1Zhang