espressif/esp-idf #2708: Firmware crash at tcp.c on excess of http requests

dude-mansaa, 2018-11-17:
I was trying to debug my code and found this core dump.

I was trying to send continuous requests on my iot device that can be around more than 1 per second. So for few seconds the code worked but after that the firmware crashed.

Firmware crashes in the cases when my client application sends commands either through web socket or http requests.

The logs for the core dump are mentioned below:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
[Nov 17 19:02:15.776] . Exception was unhandled.
[Nov 17 19:02:15.777] Core 0 register dump:
[Nov 17 19:02:15.779] PC      : 0x4019eb27  PS      : 0x00060f30  A0      : 0x800e7c92  A1      : 0x3ffd6fa0
[Nov 17 19:02:15.787] A2      : 0x00000000  A3      : 0x0000011c  A4      : 0x3ffd6144  A5      : 0x00000941
[Nov 17 19:02:15.797] A6      : 0x3fff55bc  A7      : 0x00000014  A8      : 0x801a8d54  A9      : 0x3ffd6f70
[Nov 17 19:02:15.804] A10     : 0x3ffd6148  A11     : 0x00000000  A12     : 0x3ffd619c  A13     : 0x00000000
[Nov 17 19:02:15.810] A14     : 0x3ffd61c4  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
[Nov 17 19:02:15.819] EXCVADDR: 0x00000038  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

```

```
Loaded core dump from last snippet in  /core
Remote debugging using 127.0.0.1:1234
tcp_recved (pcb=0x0, len=284)
    at /opt/Espressif/esp-idf/components/lwip/core/tcp.c:652
652       LWIP_ASSERT("don't call tcp_recved for listen-pcbs",
#0  tcp_recved (pcb=0x0, len=284)
    at /opt/Espressif/esp-idf/components/lwip/core/tcp.c:652
#1  0x400e7c92 in tcp_recved_tcpip (arg=0x3ffdac70)
    at common/platforms/lwip/mg_lwip_net_if.c:576
#2  0x4019e09e in tcpip_thread (arg=<optimized out>)
    at /opt/Espressif/esp-idf/components/lwip/api/tcpip.c:171

```

I need to know how to fix this issue. Please provide the solution.

liuzfesp, 2018-11-20:
Hi @dude-mansaa, the system is crashed because it is asserted by LWIP, this is probably caused by the bug in your code. Could you help to provide your code and detailed reproduce steps?

Alvin1Zhang, 2019-03-15:
Feel free to reopen the issue, thanks. @dude-mansaa

closed 2019-03-15 by Alvin1Zhang