espressif/esp-idf #2709: SPI no MISO & MOSI phase

michprev, 2018-11-17:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- IDF version: v3.1.1-6-g2aa9a2118
- Build System:         Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:     Linux

## Problem Description

After setting zero length of tx part of spi_transaction_t SPI behaves incorrectly.

Expected: 0x8F 0x00 0xAB 0xCD (CS goes high)

[image]

### Code to reproduce this issue

https://gist.github.com/michprev/594d5005d55f77a7cfdf991623fb098b

ginkgm, 2018-11-19:
hi @michprev ,
Could you please try set tx_buffer and rx_buffer of the transaction descriptor to ``NULL`` by either
``memset(&t, 0, sizeof(t));``
or 
```
spi_transaction_t t = {
.cmd = 0x8f,
...
};//unused fields are set to NULL
```
**every time** you configure a transaction?

according to the programming guide: https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#spi-transactions
The MOSI and MISO are skipped only when these buffers are NULL. If these phase exist, but the length is set to 0, it is regarded as -1 (quite long). That's why you cannot find the CS inactive edge.

michprev, 2018-11-19:
You are right.
It is not very intuitive but if you define like so...

ginkgm, 2018-11-19:
@michprev We design this at first for the convenience that you can leave rxlength=0 when the tx and rx use the same length, the driver will automatically fill this field for you.

closed 2018-12-13 by michprev