espressif/esp-idf #2710: SPI_CS_HOLD needless?

michprev, 2018-11-18:
What is the purpose of SPI_CS_HOLD bit in SPI_USER_REG register?
Zero hold delay of CS signal can be easily achieved just by setting SPI_HOLD_TIME in SPI_CTRL2_REG register to 0 (even while having SPI_CS_HOLD enabled).

projectgus, 2019-03-12:
Hi @michprev,

Thanks for being patient while someone got back to you about this. For future reference, the best place for this kind of hardware-oriented question is probably the https://esp32.com forums.

I'm not aware of any behavioral difference in the hardware between setting SPI_HOLD_TIME to 0 and 0 SPI_CS_HOLD, at least for "SPI_USR" commands.

This seemingly redundant setting may be included for use when the SPI peripheral block is used internally for other command types against the internal flash chip (ie "SPI_CMD" modes other than USR). In this mode, certain register values are manipulated internally depending on the command selection.

I'll check with the hardware team for confirmation and get back to you.

costaud, 2019-03-13:
```
T = 1 spi clock cycle
#t_hold is the real hold time on the wire

if spi clock == 80Mhz:
    if spi_cs_hold ==0: 
        t_hold = 0.5 * T
    else:
        t_hold = (1.5 + spi_hold_time) * T
else:
    if spi_cs_hold == 0:
        t_hold = T
    else:
        t_hold = (1 + spi_hold_time) * T
```

closed 2019-08-01 by projectgus