espressif/esp-idf #2712: esp_http_client getting redirection location / headers

pathob, 2018-11-18:
## Environment

- Development Kit: Other
- Module or chip used: ESP32-WROOM-32
- IDF version: v3.2-beta1
- Build System: Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: Linux
- Power Supply: USB

## Problem Description

Hi, I started working with the esp_http_client and generally like the simplicity of that client. But certainly, the client has technical problems and is not thought to end. These are my problems:

1. Location + Length

I tried to follow some redirections (HTTP status 302) and neither with using the automatic approach (`esp_http_client_perform`) nor the manual approach (stream reader) I am currently able to get the location I was redirected to. I noticed that in `esp_http_client.c` the location I was redirected to is stored in `client->location`.

Locally, I implemented a method for getting this location. But what I also noticed while working with very long URLs from Amazon AWS (> 500 characters) is that the stored location is truncated to a maximum of 300 characters.

2. Headers

I would be able to get the URL I was redirected to if I was able to get the header information. Generally, there is a function called `esp_http_client_get_header` but on the one hand this function is only working for request headers, so there should also be something like `esp_http_client_get_response_header` and on the other hand this function is leading to exceptions, no matter if I'm trying to access values in the request header or the response header.

I already digged a little bit into the code of nghttp and found support for headers of up to 80 KB, so the long URL of about 510 characters should not be the problem.

If you need a URL with redirection to such a long URL, you can use https://github.com/pathob/esp32-ws2812-light/releases/download/0.0.7/app.bin

If you want me to contribute some example code, please let me know.

pathob, 2018-11-20:
Thank you @tuanpmt that solved the issue with the long URL.

Now, I still need the location I was directed to. What do you think about a getter `esp_http_client_get_location`? I could contribute that...

tuanpmt, 2018-11-23:
Hi @pathob 
The short code below can help you with this. We will send MR to the IDF as soon as possible
```c
esp_err_t esp_http_client_follow_redirection(esp_http_client_handle_t client)
{
    if (client == NULL) {
        return ESP_ERR_INVALID_ARG;
    }
    if (client->location == NULL) {
        return ESP_ERR_INVALID_ARG;
    }
    return esp_http_client_set_url(client, client->location);
}
```

pathob, 2019-04-04:
Hi @tuanpmt 

any progress on this? The code is already there and could be helpful for others.

jitin17, 2019-04-09:
@pathob API for handling URL redirection is under review in the internal work queue and will be available on Github soon.

mahavirj, 2019-04-29:
Fixed with https://github.com/espressif/esp-idf/commit/cf9f2aa466d0e81b44b2acb9a114f7c28101f4d9.

closed 2019-04-29 by mahavirj