espressif/esp-idf #2713: How many DHCP requests does it take?

tve, 2018-11-18:
## Environment

- Development Kit: Adafruit Huzzah32
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v3.1.1
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: linux
- Power Supply: usb plus LiPo (Huzzah32 has LiPo charger & connector)

## Problem Description

When the module starts and connects to the AP I see a total of 4 DHCP requests even though the DHCP server responds immediately. Yes "it works" but in low-power scenarios (e.g. periodic wake-up from light sleep or deep sleep) this represents 4x the number of necessary high-power TX periods.

### Expected Behavior
A single DHCP request being sent.
### Actual Behavior
tcpdump on a linux box that has the AP (hostapd) and serves DHCP:
```
12:33:07.798427 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 30:ae:a4:23:86:64, length 308
12:33:07.798506 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 30:ae:a4:23:86:64, length 308
12:33:07.798818 IP 192.168.12.1.67 > 192.168.12.116.68: BOOTP/DHCP, Reply, length 300
12:33:07.807030 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 30:ae:a4:23:86:64, length 308
12:33:07.807062 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 30:ae:a4:23:86:64, length 308
12:33:07.807322 IP 192.168.12.1.67 > 192.168.12.116.68: BOOTP/DHCP, Reply, length 300
```

### Steps to reproduce

1. compile wifi/simple_wifi example
2. run tcpdump on dhcp server (`sudo tcpdump -n -i ap0 -s 0 port bootps`)
3. watch multiple dhcp requests at boot time

## Debug Logs

```
I (844) wifi: mode : sta (30:ae:a4:23:86:64)
I (854) simple wifi: wifi_init_sta finished.
I (854) simple wifi: connect to ap SSID:OC1 password:
I (974) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (974) wifi: state: init -> auth (b0)
I (974) wifi: state: auth -> assoc (0)
I (984) wifi: state: assoc -> run (10)
I (984) wifi: connected with OC1, channel 1
I (984) wifi: pm start, type: 1

I (1614) event: sta ip: 192.168.12.116, mask: 255.255.255.0, gw: 192.168.12.1
I (1614) simple wifi: got ip:192.168.12.116
```

negativekelvin, 2018-11-18:
Usually this is because the DHCP server is delaying its response until it completes an arp scan to detect address conflicts.

https://github.com/espressif/esp-idf/issues/799

jhinklejr, 2018-11-21:
My 2 cents from my experience.

I've seen it take over 12 sometimes.

The number is not up to you -- it seems to be up to the device asking for an IP.

I have documented an IPAD -- asking for an IP over 12 times (each time the DHCP Server sends an ACK) before it stops asking.

If the LWIP DHCP Server can do something - I too would like to know

jhinklejr, 2018-11-21:
Please ignore my last post -- I originally thought you were asking as the AP and a client was requesting an IP.

liuzfesp, 2018-11-21:
HI @tve, it depends whether the DHCP ACK is received or not. If ESP32 STA gets a DHCP ACK for the first DHCP request, it stops to send DHCP request, otherwise it retries until getting the IP.

liuzfesp, 2018-11-21:
HI @tve according to the timestamp in your log, I guess the 4 DHCP request is actual the same DHCP request because the delayed interval between each DHCP request is very short. 

If DHCP request retry happens in DHCP module, the delayed interval should be at least 500ms or 250ms. So the multiple DHCP requests should be the retry by WiFi. 

According to WiFi Specification, for each transmitting unicast data, if the Wifi doesn't receive the WiFi Ack from the receiver, WiFi will do retry. In other words, this behavior is expected and consistent with WiFi specification.

tve, 2018-11-21:
I see, so you are saying that what happens is:
- Req1: AP receives packet, esp does not receive wifi ack
- Req2: wifi rexmit, AP receives packet, esp does not receive wifi ack
- Reply 1: esp does not receive packet
- Req3: wifi rexmit, AP receives packet, esp does not receive wifi ack
- Req4: wifi rexmit, AP receives packet, esp may or may not receive wifi ack
- Reply 2: esp receives packet, end of DHCP

I guess without dumping packets in the esp32 I can't really tell, but it's certainly a plausible explanation. Thank you.

closed 2018-11-21 by tve

liuzfesp, 2018-11-21:
@tve, each Wifi packet has a WiFi sequence, if the packets have same WiFi sequence, it means the they are retried by WiFi, if the packets have different WiFi sequence, it means the packets are retried by application.

tve, 2018-11-22:
FYI, I refreshed my memory about the DHCP protocol and I captured a wifi sequence using a second adapter in monitor mode. What I see:
- The esp32 always sends two DHCP Discovery frames. The second is not a retransmission. The first one has a sequence number 0, the second a non-zero one, which changes, e.g. 2057.
- It similarly sends two DHCP Request frames, one with se no 1, the other 2 higher than the second discovery frame.
- All this is down in the noise, from the first DHCP Discovery frame to the last gratuitous ACK about 6 seconds and 89 packets are sent/received by the esp32 !!! (Not counting beacons.) It's quite shocking.