espressif/esp-idf #2715: SPI half-duplex and dummy phase

michprev, 2018-11-18:
## Problem Description

When using dummy phase in half-duplex mode SPI stops working.
Your statement

> Full duplex mode is not compatible with the dummy bit workaround, hence the frequency is limited. See dummy bit speed-up workaround.

on https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#known-issues is wrong. SPI gets broken when using dummy bits together with mosi phase (SPI_USR_DUMMY and SPI_USR_MOSI in SPI_USER_REG are set).

Expected: 0x8F 0x00 0xAB 0xCD (some 8 bits) (some 8 bits) (CS goes high)

[image]

### Code to reproduce this issue
https://gist.github.com/michprev/091c02efe181cd012d3ba388293a984a

ginkgm, 2020-07-06:
Hi @michprev ,

Unfortunately the dummy bit (workaround) is not compatible with the case where both write and read phases are enabled, no matter full duplex or half duplex. We are updating the documents to reflect this limitation.

Alvin1Zhang, 2020-08-24:
Thanks for reporting and sorry for slow turnaround. We have a fix under internal reviewing, and the issue will be closed automatically when the fix is available on GitHub. Thanks.

closed 2020-09-16 by espressif-bot