espressif/esp-idf #2716: [BUG] send() sometimes gets stuck ?

Lir10, 2018-11-19:
As i have tested tcp client example i ran it for a couple of hours untill send() got stuck and didn't return any value

I'm calling it this way : `send(sock, data_to_send, strlen(data_to_send), 0);` 

This bug is repreducable

liuzfesp, 2018-11-20:
Hi @Lir10 , could you help to provide more information about this issue, such as:
- IDF commit ID
- Log
- How long does TCP client get stuck?  Does it recoverable?

Lir10, 2018-11-20:
Hi @liuzfesp ,
> 
> 
> Hi @Lir10 , could you help to provide more information about this issue, such as:
> 
>     * IDF commit ID
**fb7ba1b**
> 
>     * Log - I can only see that the task is stuck - nothing special in the log
> 
>     * How long does TCP client get stuck?  Does it recoverable?  - if i don't reboot the ESP it stays like  that indefinitely

liuzfesp, 2018-11-21:
Hi @Lir10 I don't think the task is stuck on send(), look at the code in tcp_client_task()::

         while (1) {
            int err = send(sock, payload, strlen(payload), 0);
            if (err < 0) {
                ESP_LOGE(TAG, "Error occured during sending: errno %d", errno);
                break;
            }
           int len = recv(sock, rx_buffer, sizeof(rx_buffer) - 1, 0);
            // Error occured during receiving
            if (len < 0) {
                ESP_LOGE(TAG, "recv failed: errno %d", errno);
                break;
            }
            // Data received
            else {
                rx_buffer[len] = 0; // Null-terminate whatever we received and treat like a string
                ESP_LOGI(TAG, "Received %d bytes from %s:", len, addr_str);
                ESP_LOGI(TAG, "%s", rx_buffer);
            }
            vTaskDelay(2000 / portTICK_PERIOD_MS);
        }

The task may be blocked by recv() just because the timeout is not configured for it and there is nothing to receive.  If so, it's expected behavior.

Lir10, 2018-11-22:
Hi @liuzfesp 

The task looks stuck on `send()` - I have added prints before and after `send()`.

I can see the print before `send() `but not after - i'm sure it's stuck.

Also, i'm sending and receiving without any problem - it works like that for hours until it gets stuck

zhangyanjiaoesp, 2019-01-02:
Hi @Lir10

I have tested with the commit `fb7ba1b` for hours, but I did not reproduce your problem. 
I use the `tcp_client` example, sending and receving data with the tcp server, it works well all the time.
Do you still have this problem?

Lir10, 2019-01-02:
Hi @zhangyanjiaoesp 

Thanks for checking it out
Currently I can still see the problem

As a workaround i'm rebooting the esp if the task gets stuck - it's working fine for hours until it happens

freakyxue, 2019-11-13:
@Lir10 
We will debug the problem as soon as possible

freakyxue, 2019-11-25:
@Lir10 
Test tcp_client example according to customer feedback, hang up for 2 days, no recurrence problem,do you still have the problem

freakyxue, 2019-11-25:
Test tcp_client example, hang up for 2 days, no recurrence problem,do you still have the problem

mikisch81, 2020-05-06:
Hello, I am encountering a similar issue.

I am running a WROVER device - ESP32D0WDQ5 (revision 1), ESP-IDF official version 3.3.1.

When sending a large data buffer (> 1MB) from PSRAM every 30 seconds to a server which resides on an UBUNTU machine I get this behavior after a couple of iterations.

I also used the tcp-client example, modified for this scenario.

This is the modified `tcp_client_task()` function:
```
#define MAX_SEND_SIZE        CONFIG_TCP_SND_BUF_DEFAULT
#define PSRAM_START_ADDR     (0x3F800000)
#define LARGE_DATA_START     PSRAM_START_ADDR
#define LARGE_DATA_SIZE      (1295701)

static int send_large_data(int sock, uint8_t *buffer, size_t size)
{
    size_t remaining_size = size;
    size_t total_sent = 0;
    uint8_t *curr = buffer;

    ESP_LOGW(TAG, "Going to send total of %u bytes", size);
    while (remaining_size > 0) {
        size_t send_size = (remaining_size > MAX_SEND_SIZE) ? MAX_SEND_SIZE : remaining_size;
        int sent = send(sock, curr, send_size, 0);
        if (sent < 0) {
            ESP_LOGE(TAG, "Error occurred during sending: errno %d", errno);
            return -1;
        }
        remaining_size -= sent;
        total_sent += sent;
        ESP_LOGW(TAG, "Sent %u bytes from 0x%08x (%u/%u)", sent, (uint32_t)curr, total_sent, size);
        curr += sent;
    }
    return total_sent;
}

static void tcp_client_task(void *pvParameters)
{
    char rx_buffer[128];
    char addr_str[128];
    int addr_family;
    int ip_protocol;

    while (1) {
#ifdef CONFIG_EXAMPLE_IPV4
        struct sockaddr_in destAddr;
        destAddr.sin_addr.s_addr = inet_addr(HOST_IP_ADDR);
        destAddr.sin_family = AF_INET;
        destAddr.sin_port = htons(PORT);
        addr_family = AF_INET;
        ip_protocol = IPPROTO_IP;
        inet_ntoa_r(destAddr.sin_addr, addr_str, sizeof(addr_str) - 1);
#else // IPV6
        struct sockaddr_in6 destAddr;
        inet6_aton(HOST_IP_ADDR, &destAddr.sin6_addr);
        destAddr.sin6_family = AF_INET6;
        destAddr.sin6_port = htons(PORT);
        addr_family = AF_INET6;
        ip_protocol = IPPROTO_IPV6;
        inet6_ntoa_r(destAddr.sin6_addr, addr_str, sizeof(addr_str) - 1);
#endif

        int sock =  socket(addr_family, SOCK_STREAM, ip_protocol);
        if (sock < 0) {
            ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
            break;
        }
        ESP_LOGI(TAG, "Socket created");

[... 141 lines cut ...]
I (3099) example: Socket created
I (3449) example: Successfully connected
W (3449) example: Going to send total of 1295701 bytes
W (3449) example: Sent 5744 bytes from 0x3f800000 (5744/1295701)
W (3529) example: Sent 5744 bytes from 0x3f801670 (11488/1295701)
W (3559) example: Sent 5744 bytes from 0x3f802ce0 (17232/1295701)
W (3609) example: Sent 5744 bytes from 0x3f804350 (22976/1295701)
...
...
W (6529) example: Sent 5744 bytes from 0x3f93a200 (1292400/1295701)
W (6539) example: Sent 3301 bytes from 0x3f93b870 (1295701/1295701)
I (6539) example: Going to sleep for 30 seconds
...
...
W (1261749) example: Sent 5744 bytes from 0x3f87ccf0 (516960/1295701)
W (1261749) example: Sent 5744 bytes from 0x3f87e360 (522704/1295701)
W (1261799) example: Sent 5744 bytes from 0x3f87f9d0 (528448/1295701)
W (1261819) example: Sent 5744 bytes from 0x3f881040 (534192/1295701)
W (1262099) example: Sent 5744 bytes from 0x3f8826b0 (539936/1295701)
W (1264899) example: Sent 5744 bytes from 0x3f883d20 (545680/1295701)
W (1295539) example: Sent 5744 bytes from 0x3f885390 (551424/1295701)
E (2446359) example: Error occurred during sending: errno 113
E (2446359) example: Error occured during sending: errno 113
E (2446359) example: Shutting down socket and restarting...
```

freakyxue, 2020-05-07:
hi @mikisch81 
I will use your code to reproduce this problem. Thank you for your feedback

mikisch81, 2020-05-10:
Hi @freakyxue did you manage to reproduce it?

freakyxue, 2020-05-11:
hi @mikisch81 
LWIP_IP_FRAG
Is this enabled in menuconfig????

mikisch81, 2020-05-11:
@freakyxue No, it is not set, My sdkconfig.defaults looks like this:
```
CONFIG_WIFI_SSID=<MY_SSID>
CONFIG_WIFI_PASSWORD=<MY_PASSWORD>
CONFIG_EXAMPLE_IPV4_ADDR=<MY SERVER_IP>

CONFIG_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_BOOT_INIT=y
```

So the generated sdkconfig does not set `LWIP_IP_FRAG`.

freakyxue, 2020-05-11:
make menuconfig
Component config-->LWIP->ENABLE fragment outgoing ip packets

mikisch81, 2020-05-11:
Is this what is causing the issue?

freakyxue, 2020-05-11:
yes , Enable this option, test it, thank you

mikisch81, 2020-05-11:
No, it didn't help

freakyxue, 2020-05-19:
hi @mikisch81 
I didn't reproduce the problem locally,
From the log view, the error "route problem" is returned, which leads to the sending failure, right?
If so, I think during the transmission process, WiFi may be disconnected.

mikisch81, 2020-05-19:
@freakyxue If you are referring to these lines:
```
W (1295539) example: Sent 5744 bytes from 0x3f885390 (551424/1295701)
E (2446359) example: Error occurred during sending: errno 113

```
Then how would you explain the huge time difference between them?

freakyxue, 2020-05-22:
hi @mikisch81 
Because the length of each transmission you send is equal to TCP_ Wind, after sending data, you need to wait for TCP_ Ack to release TCP_ Wind, can carry on data transmission

The last log shows that the sending data stops. I think the device end does not receive the ACK sent by the service end, or this loss causes the send_ Wind was not released, resulting in the send function never returning

I recommend that you do not send data each time with a length equal to TCP_ Wind, the best length is equal to TCP_ MSS

closed 2020-11-02 by Alvin1Zhang