espressif/esp-idf #2719: nvs_partition_gen.py

milamber-ls, 2018-11-20:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version DevKitC: v4
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    v3.2-beta1-151-g960c24057
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Linux] centos 7 64 bit
- Power Supply:         [USB]

## Problem Description
running the example 

` python nvs_partition_gen.py  --encrypt True --keygen True sample_singlepage_blob.csv sample_singlepage_blob.bin  0x3000`
### Expected Behavior
I would expect the encrypted partition to be created
### Actual Behavior
returns error `ValueError: Invalid key size (512) for AES.
`
### Steps to repropduce

execute ` python nvs_partition_gen.py  --encrypt True --keygen True sample_singlepage_blob.csv sample_singlepage_blob.bin  0x3000`
in `~/esp/esp-idf/components/nvs_flash/nvs_partition_generator/testdata/ ` directory

All python requirements are met
## Debug Logs

```
`Traceback (most recent call last):
  File "nvs_partition_gen.py", line 775, in <module>
    main()
  File "nvs_partition_gen.py", line 770, in main
    nvs_part_gen(input_filename, output_filename, input_size, key_gen, is_encrypt_data, key_file, version_no)
  File "nvs_partition_gen.py", line 692, in nvs_part_gen
    write_entry(nvs_obj, row["key"], row["type"], row["encoding"], row["value"])
  File "nvs_partition_gen.py", line 611, in write_entry
    nvs_instance.write_namespace(key)
  File "nvs_partition_gen.py", line 508, in write_namespace
    self.cur_page.write_primitive_data(key, self.namespace_idx, "u8", 0,self)
  File "nvs_partition_gen.py", line 449, in write_primitive_data
    self.write_entry_to_buf(entry_struct, 1,nvs_obj)
  File "nvs_partition_gen.py", line 208, in write_entry_to_buf
    encr_data_ret = self.encrypt_data(data, entrycount,nvs_obj)
  File "nvs_partition_gen.py", line 194, in encrypt_data
    encr_data_ret = self.encrypt_entry(data_val, tweak_val, encr_key_input)
  File "nvs_partition_gen.py", line 127, in encrypt_entry
    cipher = Cipher(algorithms.AES(encr_key), modes.XTS(tweak), backend=backend)
  File "/usr/lib64/python2.7/site-packages/cryptography/hazmat/primitives/ciphers/algorithms.py", line 30, in __init__
    self.key = _verify_key_size(self, key)
  File "/usr/lib64/python2.7/site-packages/cryptography/hazmat/primitives/ciphers/algorithms.py", line 17, in _verify_key_size
    len(key) * 8, algorithm.name
ValueError: Invalid key size (512) for AES.
`
```

milamber-ls, 2018-11-20:
success!

milamber-ls, 2018-11-20:
it worked on the Linux machine, but in my windows machine (same idf and compiler) I get another error
```
Traceback (most recent call last):
  File "nvs_partition_gen.py", line 775, in <module>
    main()
  File "nvs_partition_gen.py", line 770, in main
    nvs_part_gen(input_filename, output_filename, input_size, key_gen, is_encryp                                                                                                                                                                                                 t_data, key_file, version_no)
  File "nvs_partition_gen.py", line 692, in nvs_part_gen
    write_entry(nvs_obj, row["key"], row["type"], row["encoding"], row["value"])
  File "nvs_partition_gen.py", line 611, in write_entry
    nvs_instance.write_namespace(key)
  File "nvs_partition_gen.py", line 508, in write_namespace
    self.cur_page.write_primitive_data(key, self.namespace_idx, "u8", 0,self)
  File "nvs_partition_gen.py", line 449, in write_primitive_data
    self.write_entry_to_buf(entry_struct, 1,nvs_obj)
  File "nvs_partition_gen.py", line 208, in write_entry_to_buf
    encr_data_ret = self.encrypt_data(data, entrycount,nvs_obj)
  File "nvs_partition_gen.py", line 194, in encrypt_data
    encr_data_ret = self.encrypt_entry(data_val, tweak_val, encr_key_input)
  File "nvs_partition_gen.py", line 123, in encrypt_entry
    backend = default_backend()
  File "C:/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backen                                                                                                                                                                                                 ds/__init__.py", line 15, in default_backend
    from cryptography.hazmat.backends.openssl.backend import backend
  File "C:/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backen                                                                                                                                                                                                 ds/openssl/__init__.py", line 7, in <module>
    from cryptography.hazmat.backends.openssl.backend import backend
  File "C:/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/backen                                                                                                                                                                                                 ds/openssl/backend.py", line 53, in <module>
    from cryptography.hazmat.bindings.openssl import binding
  File "C:/msys32/mingw32/lib/python2.7/site-packages/cryptography/hazmat/bindin                                                                                                                                                                                                 gs/openssl/binding.py", line 13, in <module>
    from cryptography.hazmat.bindings._openssl import ffi, lib
ImportError: DLL load failed: The specified module could not be found.
```

milamber-ls, 2018-11-20:
and this is the result of the command you suggested on the windows machine

``` python -m pip install --user 'cryptography>=2.1.4'
Requirement already satisfied: cryptography>=2.1.4 in c:/msys32/mingw32/lib/pyth                                                                                                                                                                                                 on2.7/site-packages (2.3.1)
Requirement already satisfied: idna>=2.1 in c:/msys32/mingw32/lib/python2.7/site                                                                                                                                                                                                 -packages (from cryptography>=2.1.4) (2.7)
Requirement already satisfied: asn1crypto>=0.21.0 in c:/msys32/mingw32/lib/pytho                                                                                                                                                                                                 n2.7/site-packages (from cryptography>=2.1.4) (0.24.0)
Requirement already satisfied: six>=1.4.1 in c:/msys32/mingw32/lib/python2.7/sit                                                                                                                                                                                                 e-packages (from cryptography>=2.1.4) (1.11.0)
Requirement already satisfied: cffi!=1.11.3,>=1.7 in c:/msys32/mingw32/lib/pytho                                                                                                                                                                                                 n2.7/site-packages (from cryptography>=2.1.4) (1.10.0)
Requirement already satisfied: enum34 in c:/msys32/mingw32/lib/python2.7/site-pa                                                                                                                                                                                                 ckages (from cryptography>=2.1.4) (1.1.6)
Requirement already satisfied: ipaddress in c:/msys32/mingw32/lib/python2.7/site                                                                                                                                                                                                 -packages (from cryptography>=2.1.4) (1.0.22)
Requirement already satisfied: pycparser in c:/msys32/mingw32/lib/python2.7/site                                                                                                                                                                                                 -packages (from cffi!=1.11.3,>=1.7->cryptography>=2.1.4) (2.19)```

dobairoland, 2018-11-20:
On Windows you need to update also the openssl library. This can be done with:
```
pacman -Sy mingw-w64-i686-openssl
```

If the above command doesn't work then maybe you will have to update all of the packages:
```
pacman -Syu
```

If you have an older pacman installed then the last command need to be run twice.

Please let me know if the first step was enough for you to solve the issue.