espressif/esp-idf #2721: Unable to connect to WPA2 Enterprise PEAP MSCHAPV2

spiritonline, 2018-11-20:
## Environment

- Development Kit: Adafruit feather Huzzah ESP32
- Module or chip used:  ESP32-WROOM-32
- IDF version: v3.2-beta1-34-gfb7ba1baa
- Build System:         [Make|CMake]
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:  macOS
- Power Supply:         [USB|external 5V

## Problem Description
After first trying the Arduino core I was unable to connect the ESP32 to a WPA2 enterprise hotspot network ran by a [national ISP. ](https://www.ziggo.nl/klantenservice/wifi/wifispots/instellen/) (Google translate version [here](https://translate.google.com/translate?sl=auto&tl=en&js=y&prev=_t&hl=en&ie=UTF-8&u=https%3A%2F%2Fwww.ziggo.nl%2Fklantenservice%2Fwifi%2Fwifispots%2Finstellen%2F%23verder%2FB%2FZiggo_WifiSpots%2FAndroid%2FOverige%2FVerder%2FVerder%2FNee%2FVerder%2FVerder%2FVerder%2FVerder%2FVerder%2FIk%2520krijg%2520een%2520pop-up%2520melding%2FVerder%2FVerder%2FNee&edit-text=))
I switched to the IDF SDK in hopes of having better luck. I've configured everything in the make menuconfig although the instructions from the ISP are to leave "anonymous identify" blank. 
The ISP proved a self signed [.crt root certificate](https://www.ziggo.nl/content/dam/www.ziggo.nl/tools/WifiSpots%20certificaten/WifiSpots%20certificaat%20Android%20en%20Windows%20LGI.crt) which I assume is a CA client certificate. 
I can connect without issue to this network on my Android phone or Mac. On both systems it will also connect without a certificate (tough I have to specify in Android not to validate) 

In the serial monitor it seems to Authenticate, Associate but then gets stuck on:
"wpa: SSL: Need 2908 bytes more input data"

### Expected Behavior
Connect to WPA2 Enterprise PEAP/MSCHAPV2 
### Actual Behavior
wpa2 FAILED
### Steps to repropduce

1. Load WPA2 enterprise example in the same manner the hello world example is loaded in the docs
2. Use make menuconfig to configure the project with username,password etc.
3. Convert the crt certificate provided by the network admin to "wpa2_ca.pem" and "keywpa2_client.key using openssl command prompt 
4. Run make flash to build and upload 

### Code to reproduce this issue

*Example code* 

## Debug Logs

```
13:12:16.694 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:12:16.694 -> flash read err, 1000
13:12:16.694 -> ets_main.c 371 
ets Jun  8 2016 00:22:57
13:12:17.086 -> 
13:12:17.086 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:12:17.086 -> configsip: 0, SPIWP:0xee
13:12:17.086 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:12:17.086 -> mode:DIO, clock div:2
13:12:17.086 -> load:0x3fff0018,len:4
13:12:17.086 -> load:0x3fff001c,len:6172
13:12:17.086 -> load:0x40078000,len:10168
13:12:17.086 -> load:0x40080400,len:6552
13:12:17.086 -> entry 0x40080764
[0;32mI (28) boot: ESP-IDF v3.2-beta1-34-gfb7ba1baa 2nd stage bootloader[0m
13:12:17.158 -> [0;32mI (28) boot: compile time 14:33:55[0m
13:12:17.158 -> [0;32mI (29) boot: Enabling RNG early entropy source...[0m
13:12:17.158 -> [0;32mI (35) boot: SPI Speed      : 40MHz[0m
13:12:17.158 -> [0;32mI (39) boot: SPI Mode       : DIO[0m
13:12:17.158 -> [0;32mI (43) boot: SPI Flash Size : 4MB[0m
13:12:17.158 -> [0;32mI (47) boot: Partition Table:[0m
13:12:17.158 -> [0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[0m
13:12:17.158 -> [0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
13:12:17.158 -> [0;32mI (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
13:12:17.158 -> [0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00100000[0m
13:12:17.158 -> [0;32mI (80) boot: End of partition table[0m
13:12:17.158 -> [0;32mI (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x19254 (102996) map[0m
[0;32mI (129) esp_image: segment 1: paddr=0x0002927c vaddr=0x3ff80000 size=0x00000 (     0) load[0m
13:12:17.264 -> [0;32mI (130) esp_image: segment 2: paddr=0x00029284 vaddr=0x3ff80000 size=0x00000 (     0) load[0m
13:12:17.264 -> [0;32mI (136) esp_image: segment 3: paddr=0x0002928c vaddr=0x3ffb0000 size=0x03128 ( 12584) load[0m
13:12:17.264 -> [0;32mI (150) esp_image: segment 4: paddr=0x0002c3bc vaddr=0x3ffb3128 size=0x00000 (     0) load[0m
13:12:17.264 -> [0;32mI (154) esp_image: segment 5: paddr=0x0002c3c4 vaddr=0x40080000 size=0x00400 (  1024) load[0m
13:12:17.264 -> [0;32mI (163) esp_image: segment 6: paddr=0x0002c7cc vaddr=0x40080400 size=0x03844 ( 14404) load[0m
[... 66 lines cut ...]
13:12:29.749 -> [0;32mI (12396) example: MASK:0.0.0.0[0m
13:12:29.749 -> [0;32mI (12396) example: GW:0.0.0.0[0m
13:12:29.749 -> [0;32mI (12396) example: ~~~~~~~~~~~[0m
[0;32mI (14396) example: ~~~~~~~~~~~[0m
13:12:31.753 -> [0;32mI (14396) example: IP:0.0.0.0[0m
13:12:31.753 -> [0;32mI (14396) example: MASK:0.0.0.0[0m
13:12:31.753 -> [0;32mI (14396) example: GW:0.0.0.0[0m
13:12:31.753 -> [0;32mI (14396) example: ~~~~~~~~~~~[0m
[0;32mI (16396) example: ~~~~~~~~~~~[0m
13:12:33.780 -> [0;32mI (16396) example: IP:0.0.0.0[0m
13:12:33.780 -> [0;32mI (16396) example: MASK:0.0.0.0[0m
13:12:33.780 -> [0;32mI (16396) example: GW:0.0.0.0[0m
13:12:33.780 -> [0;32mI (16396) example: ~~~~~~~~~~~[0m
[0;32mI (18396) example: ~~~~~~~~~~~[0m
13:12:35.745 -> [0;32mI (18396) example: IP:0.0.0.0[0m
13:12:35.745 -> [0;32mI (18396) example: MASK:0.0.0.0[0m
13:12:35.745 -> [0;32mI (18396) example: GW:0.0.0.0[0m
13:12:35.745 -> [0;32mI (18396) example: ~~~~~~~~~~~[0m
[0;32mI (20396) example: ~~~~~~~~~~~[0m
13:12:37.783 -> [0;32mI (20396) example: IP:0.0.0.0[0m
13:12:37.783 -> [0;32mI (20396) example: MASK:0.0.0.0[0m
13:12:37.783 -> [0;32mI (20396) example: GW:0.0.0.0[0m
13:12:37.783 -> [0;32mI (20396) example: ~~~~~~~~~~~[0m
I (21546) wpa: >>>>>wpa2 FAILED
```

liuzfesp, 2019-03-20:
@spiritonline is it possible to provide the captured packets because we just can't reproduce this issue.

Alvin1Zhang, 2019-05-16:
@spiritonline Thanks for reporting the issue. Would you please help share more details or logs or capture packets? Thanks.

Alvin1Zhang, 2019-05-22:
@spiritonline Thanks for reporting the issue. Will close the issue due to short of details, logs or captured packets. Feel free to reopen the issue if it happens, and please help provide more details. Thanks.

closed 2019-05-22 by Alvin1Zhang