espressif/esp-idf #2724: wifi + ble crash from time to time

chegewara, 2018-11-21:
I dont know if this is esp-idf issue or just my app require some optimization. It not always happen but when im processing http client request (download file) and at the same time in my other task is starting bluetooth le (init, star GATT attributes etc) my app crash with those logs:
```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Core 0 register dump:
PC      : 0x40094408  PS      : 0x00060034  A0      : 0x80092e7e  A1      : 0x3ffbe1f0
0x40094408: uxPortCompareSetExtram at D:/msys32/home/imper/esp/esp-idf/components/freertos/por
t.c:452

A2      : 0x3f80195c  A3      : 0xb33fffff  A4      : 0x3ffbe220  A5      : 0xbad00bad
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000002  A11     : 0x3ffe38d0  A12     : 0x80098e81  A13     : 0x3ffe38c0
A14     : 0x3ffae270  A15     : 0x3ffe38f0  SAR     : 0x00000019  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x4009b7fc  LEND    : 0x4009b82a  LCOUNT  : 0xffffffff
0x4009b7fc: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/x
tensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.
S:168

0x4009b82a: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/x
tensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.
S:201

Core 0 was running in ISR context:
EPC1    : 0x40062230  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40094408
0x40094408: uxPortCompareSetExtram at D:/msys32/home/imper/esp/esp-idf/components/freertos/por
t.c:452

Backtrace: 0x40094408:0x3ffbe1f0 0x40092e7b:0x3ffbe220 0x4008b7ca:0x3ffbe250 0x4008d1e5:0x3ffb
e270 0x4008c8fb:0x3ffbe2b0 0x4008fbb3:0x3ffbe2d0 0x40084dbd:0x3ffbe300 0x4006222d:0x00000000
0x40094408: uxPortCompareSetExtram at D:/msys32/home/imper/esp/esp-idf/components/freertos/por
t.c:452

0x40092e7b: vPortCPUAcquireMutexIntsDisabledExtram at D:/msys32/home/imper/esp/esp-idf/compone
nts/freertos/tasks.c:4466
 (inlined by) vPortCPUAcquireMutexIntsDisabled at D:/msys32/home/imper/esp/esp-idf/components/
freertos/portmux_impl.h:95
 (inlined by) vTaskEnterCritical at D:/msys32/home/imper/esp/esp-idf/components/freertos/tasks
.c:4258

0x4008b7ca: wifi_int_disable_wrapper at D:/msys32/home/imper/esp/esp-idf/components/esp32/wifi
_os_adapter.c:187

0x4008d1e5: pp_post at ??:?

0x4008c8fb: lmacProcessRxSucData at ??:?

0x4008fbb3: wDev_ProcessFiq at ??:?

0x40084dbd: _xt_lowint1 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_vectors
.S:1105
```
This is log from normal run (i hope its easy enough to read):
```
I (3819) HTTP_CLIENT: 512 ---> 33365
I (3823) HTTP_CLIENT: 512 ---> 33877
I (3834) HTTP_CLIENT: 512 ---> 34389
W (3835) SampleServer: custom gatts event handler, event: 0
I (3838) HTTP_CLIENT: 512 ---> 34901
W (3842) SampleServer: custom gatts event handler, event: 7
I (3846) HTTP_CLIENT: 512 ---> 35413
W (3852) SampleServer: custom gatts event handler, event: 9
I (3854) HTTP_CLIENT: 512 ---> 35925
W (3859) SampleServer: custom gatts event handler, event: 10
I (3863) HTTP_CLIENT: 512 ---> 36437
W (3870) SampleServer: custom gatts event handler, event: 9
I (3875) HTTP_CLIENT: 512 ---> 36949
W (3884) SampleServer: custom gatts event handler, event: 12
I (3887) BLEDevice: create advertising
I (3888) HTTP_CLIENT: 112 ---> 37061
I (3901) HTTP_CLIENT: 512 ---> 37573
W (3901) SampleServer: custom gap event handler, event: 0
I (3904) HTTP_CLIENT: 512 ---> 38085
W (3909) SampleServer: custom gap event handler, event: 6
I (3915) HTTP_CLIENT: 512 ---> 38597
I (3925) HTTP_CLIENT: 512 ---> 39109
I (3928) HTTP_CLIENT: 512 ---> 39621
I (3933) HTTP_CLIENT: 512 ---> 40133
I (3937) HTTP_CLIENT: 512 ---> 40645
```
```
$ git show
commit fb7ba1baa068a2267ef54b7105cf3c8a004fe4eb (HEAD -> master, origin/master, origin/HEAD)
Merge: 4f4bed5e6 84f1eb5ea
```

liuzfesp, 2018-11-21:
The system crashed because uxPortCompareSetExtram() accesses external memory in ISR handling, it has two possibilities:
*  We have bug in uxPortCompareSetExtram() 
*  The memory has been corrupted by application

@igrr

chegewara, 2018-11-21:
@liuzfesp Do you want elf or bin file?

liuzfesp, 2018-11-21:
Sure, it's helpful to examine the exact instruction that causes the crash.

chegewara, 2018-11-22:
But i have to warn you it is a mess in code, because i am using the same app to test multiple scenarios and just switching which function is started at the moment:
https://drive.google.com/open?id=150qEwDdulo8b1vlySYaSmVwsLwN7gasO

Im getting more strange issues like coex error in app that in theory is not using wifi, only ble:
```
0x4008ec67: rcUpdateTxDoneAmpdu2 at ??:?

0x40090855: coex_timer_ts_end_disalarm at ??:?

0x40082266: ipc_task at D:/msys32/home/imper/esp/esp-idf/components/esp32/ipc.c:60

0x4008fe45: coex_core_ts_start at ??:?
```
But i think its for another issue.

liuzfesp, 2018-11-22:
@chegewara, I can't open your link.

chegewara, 2018-11-22:
[hello-world.zip](https://github.com/espressif/esp-idf/files/2606258/hello-world.zip)

chegewara, 2018-11-22:
I would like to say that coexist never before works so good, except that crash. Really great job about that.

My coexist settings are:
- turn on sw coexist in menuconfig
- coexist priority wifi

TianHao-Espressif, 2018-11-22:
@chegewara , about the "coex err" in BLE, could paste the complete log? I want to know the scenario when error happens and the error type. Thanks.

chegewara, 2018-11-22:
@TianHao-Espressif Sure, but this is only one time case and i think its more my app fault rather tan esp-idf issue:
```
I (32) boot: ESP-IDF v3.2-beta1-34-gfb7ba1baa-dirty 2nd stage bootloader
I (32) boot: compile time 13:43:57
I (33) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : DIO
I (47) boot: SPI Flash Size : 4MB
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (69) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (77) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  3 ota_0            OTA app          00 10 00010000 001c2000
I (92) boot:  4 ota_1            OTA app          00 11 001e0000 001c2000
I (99) boot: End of partition table
I (103) boot: No factory image, trying OTA 0
I (108) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x510f8 (332024) map
I (183) esp_image: segment 1: paddr=0x00061120 vaddr=0x3ff80000 size=0x00000 (     0) load
I (183) esp_image: segment 2: paddr=0x00061128 vaddr=0x3ff80000 size=0x00000 (     0) load
I (189) esp_image: segment 3: paddr=0x00061130 vaddr=0x3ffbdb60 size=0x03404 ( 13316) load
I (201) esp_image: segment 4: paddr=0x0006453c vaddr=0x3ffc0f64 size=0x00000 (     0) load
I (207) esp_image: segment 5: paddr=0x00064544 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_ve
ctors.S:1685

I (216) esp_image: segment 6: paddr=0x0006494c vaddr=0x40080400 size=0x0b6c4 ( 46788) load
I (235) esp_image: segment 7: paddr=0x00070018 vaddr=0x400d0018 size=0xc8110 (819472) map
0x400d0018: _flash_cache_start at ??:?

I (398) esp_image: segment 8: paddr=0x00138130 vaddr=0x4008bac4 size=0x0c348 ( 49992) load
0x4008bac4: ieee80211_output_process at ??:?

I (409) esp_image: segment 9: paddr=0x00144480 vaddr=0x400c0000 size=0x00000 (     0) load
I (409) esp_image: segment 10: paddr=0x00144488 vaddr=0x50000000 size=0x00000 (     0) load
I (416) esp_image: segment 11: paddr=0x00144490 vaddr=0x50000000 size=0x00000 (     0) load
I (429) boot: Loaded app from partition at offset 0x10000
I (431) boot: Disabling RNG early entropy source...
E (438) spiram: SPI RAM enabled but initialization failed. Bailing out.
I (444) cpu_start: Failed to init external RAM; continuing without it.
I (451) cpu_start: Pro cpu up.
I (454) cpu_start: Starting app cpu, entry point is 0x400813e0
0x400813e0: call_start_cpu1 at D:/msys32/home/imper/esp/esp-idf/components/esp32/cpu_start.c:2
45

I (0) cpu_start: App cpu up.
D (465) memory_layout: Checking 12 reserved memory ranges:
D (470) memory_layout: Reserved memory range 0x3f800000 - 0x3fc00000
D (477) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (483) memory_layout: Reserved memory range 0x3ffae6e0 - 0x3ffaff10
D (489) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6388
D (496) memory_layout: Reserved memory range 0x3ffb8000 - 0x3ffb9a20
D (502) memory_layout: Reserved memory range 0x3ffbdb28 - 0x3ffbdb5c
D (509) memory_layout: Reserved memory range 0x3ffbdb60 - 0x3ffcc840
D (515) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (521) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (528) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (534) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_ve
ctors.S:1685

D (541) memory_layout: Reserved memory range 0x40080000 - 0x40097e09
0x40080000: _WindowOverflow4 at D:/msys32/home/imper/esp/esp-idf/components/freertos/xtensa_ve
ctors.S:1685

0x40097e09: rtc_clk_cpu_freq_to_pll_mhz at D:/msys32/home/imper/esp/esp-idf/components/soc/esp
32/rtc_clk.c:474

D (547) memory_layout: Building list of available memory regions:
D (553) memory_layout: Available memory region 0x3ffaff10 - 0x3ffb0000
[... 153 lines cut ...]

0x4008fe45: coex_core_ts_start at ??:?

Core 1 register dump:
PC      : 0x4008ec67  PS      : 0x00060134  A0      : 0x80090858  A1      : 0x3ffb9d20
0x4008ec67: rcUpdateTxDoneAmpdu2 at ??:?

A2      : 0x3ffb78fc  A3      : 0x0000abab  A4      : 0xb33fffff  A5      : 0x00000001
A6      : 0x00060120  A7      : 0x0000cdcd  A8      : 0x0000cdcd  A9      : 0x3ffb9d20
A10     : 0x00000003  A11     : 0x00060123  A12     : 0x00060120  A13     : 0x00000001
A14     : 0x00060123  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4008ec67:0x3ffb9d20 0x40090855:0x3ffb9d50 0x40082266:0x3ffb9d90 0x4008fe45:0x3ffb
9db0
0x4008ec67: rcUpdateTxDoneAmpdu2 at ??:?

0x40090855: coex_timer_ts_end_disalarm at ??:?

0x40082266: ipc_task at D:/msys32/home/imper/esp/esp-idf/components/esp32/ipc.c:60

0x4008fe45: coex_core_ts_start at ??:?

CPU halted.
```

liuzfesp, 2018-11-22:
HI @chegewara we need time to debug this issue...

chegewara, 2018-11-22:
@chegewara Cool, take your time. This is nothing that bothers me now.

enricop, 2019-10-17:
still present on release/v4.0 while working with BLE GATT and Wifi is on.
please push master updates also on this branch.
thanks for your support

liuzfesp, 2019-10-21:
Hi @enricop, by looking at the history of this issue, it related to several different crashes. Could you let me know the exact crash log of your issues? It will be helpful if you can provide more information about your issue, such as the steps to reproduce this issue etc.

enricop, 2019-10-21:
Hi @liuzfesp , it happened before connecting to the GATT with a BLE application while also the Wifi was running.

Here is the log:

> Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
> Core 0 register dump:
> PC      : 0x40014ebb  PS      : 0x00060634  A0      : 0x80088524  A1      : 0x3ffbf630  
> A2      : 0x3ffb3264  A3      : 0x3ffb3180  A4      : 0x07ffffff  A5      : 0x04000000  
> A6      : 0x0000028c  A7      : 0x00000000  A8      : 0x01b3ab01  A9      : 0x07ffffff  
> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x0000f10e  A13     : 0x01b49c0f  
> A14     : 0x04000000  A15     : 0x3ffbef30  SAR     : 0x00000017  EXCCAUSE: 0x00000005  
> EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
> Core 0 was running in ISR context:
> EPC1    : 0x40081e02  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40014ebb
> 0x40081e02: coex_bt_release_wrapper at /home/marco/esp-homekit-sdk-beta/esp-idf/components/esp32/esp_adapter.c:477
> 
> 
> ELF file SHA256: dd36f2f373e56d54378d364c42c531e2de6f7c30d620c32efbb58d1d98d7ef30
> 
> Backtrace: 0x40014eb8:0x3ffbf630 |<-CORRUPTED
> 
> Rebooting...
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:6640
> ho 0 tail 12 room 4
> load:0x40078000,len:14432
> load:0x40080400,len:4296
> entry 0x400806e4

the sdkconfig
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/3751465/sdkconfig.txt)

enricop, 2019-10-21:
Another one:

> Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
> Core 0 register dump:
> PC      : 0x40098c49  PS      : 0x00060d34  A0      : 0x80081ac4  A1      : 0x3ffbf5a0  
> 0x40098c49: xQueueReceiveFromISR at /home/marco/esp-homekit-sdk/esp-idf/components/freertos/queue.c:1645
> 
> A2      : 0x3f813674  A3      : 0x00000000  A4      : 0x3ffbf5e0  A5      : 0x3f8136c0  
> A6      : 0x00060921  A7      : 0x50000000  A8      : 0xbad00bad  A9      : 0x3ffbf580  
> A10     : 0x3f8136c0  A11     : 0x3ffb8d40  A12     : 0x00000008  A13     : 0x4008aa20  
> 0x4008aa20: r_lld_evt_end_isr at ??:?
> 
> A14     : 0x3ffbfc84  A15     : 0x3ffcdbe0  SAR     : 0x00000016  EXCCAUSE: 0x00000007  
> EXCVADDR: 0x00000000  LBEG    : 0x40093030  LEND    : 0x40093046  LCOUNT  : 0xffffffff  
> 0x40093030: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:146
> 
> 0x40093046: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:161
> 
> Core 0 was running in ISR context:
> EPC1    : 0x40099693  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40098c49
> 0x40099693: vTaskEnterCritical at /home/marco/esp-homekit-sdk/esp-idf/components/freertos/tasks.c:4190
> 
> 0x40098c49: xQueueReceiveFromISR at /home/marco/esp-homekit-sdk/esp-idf/components/freertos/queue.c:1645
> 
> 
> ELF file SHA256: ad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0baad0bd0ba
> 
> Backtrace: 0x40098c46:0x3ffbf5a0 0x40081ac1:0x3ffbf5c0 0x40083836:0x3ffbf5e0 0x40083c96:0x3ffbf620 0x400837dd:0x3ffbf660 0x40081add:0x3ffbf680 0x4008aabb:0x3ffbf6a0 0x4008bdfe:0x3ffbf6c0 0x4008cc73:0x3ffbf6e0 0x4008645a:0x3ffbf700 0x4000c050:0x3ffcda60 |<-CORRUPTED
> 0x40098c46: xQueueReceiveFromISR at /home/marco/esp-homekit-sdk/esp-idf/components/freertos/queue.c:1645
> 
> 0x40081ac1: semphr_take_from_isr_wrapper at /home/marco/esp-homekit-sdk/esp-idf/components/esp32/esp_adapter.c:261
> 
> 0x40083836: coex_bb_reset_unlock at ??:?
> 
> 0x40083c96: coex_core_release at ??:?
> 
> 0x400837dd: coex_bt_release at ??:?
> 
> 0x40081add: coex_bt_release_wrapper at /home/marco/esp-homekit-sdk/esp-idf/components/esp32/esp_adapter.c:474
> 
> 0x4008aabb: r_lld_evt_end_isr at ??:?
> 
> 0x4008bdfe: r_rwble_isr at ??:?
> 
> 0x4008cc73: r_rwbtdm_isr_wrapper at intc.c:?
> 
> 0x4008645a: _xt_lowint1 at /home/marco/esp-homekit-sdk/esp-idf/components/freertos/xtensa_vectors.S:1153
> 
> 
> Rebooting...

liuzfesp, 2019-10-22:
Hi @TianHao-Espressif please help to take a look at this issue

enricop, 2019-10-29:
with latest fixes on `release/v4.0` the situation seems to be improved regarding the above errors

TianHao-Espressif, 2019-11-01:
For the following crash, it's fixed in latest master, release/v4.0, release/v3.3. For example, in master branch, this commit 4987a5ad904d7ab469853f28bc2345f9248bc872 fix the bug.

> Guru Meditation Error: Core 0 panic'ed (Cache disabled but cached memory region accessed)
> Core 0 register dump:
> PC : 0x40098c49 PS : 0x00060d34 A0 : 0x80081ac4 A1 : 0x3ffbf5a0
> 0x40098c49: [...]

Alvin1Zhang, 2020-01-21:
@enricop Thanks for sharing the updates. Would you please help try the commit shared by @TianHao-Espressif and help share if the issue gets resolved at your side? Thanks.

enricop, 2020-01-21:
we will try to re-enable `CONFIG_BTDM_CTRL_FULL_SCAN_SUPPORTED` and have a try. thank you

enricop, 2020-01-22:
We saw that commit https://github.com/espressif/esp-idf/commit/4987a5ad904d7ab469853f28bc2345f9248bc872 is not on `release/v4.0` yet.

We can't afford to switch to master for our project. We ask if it is backported. Thank you

enricop, 2020-01-23:
hi @Alvin1Zhang ,

we are using this commit since it was released and no crash occurred. For us you can close the ticket.

Alvin1Zhang, 2020-02-18:
@enricop Thanks for updates. Feel free to reopen if the issue still happens. Thanks.

closed 2020-02-18 by Alvin1Zhang