espressif/esp-idf #2725: Classical Bluetooth SPP send throughput is very slow to Android

copercini, 2018-11-21:
When sending data using `esp_spp_write()` at more than ~ 5KB/s to and Android phone, it triggers a lot of `ESP_SPP_CONG_EVT` events and the communication slow down, this problem doesn't occur in other systems, like a Windows PC with the same ESP code

Commenting out [this line](https://github.com/espressif/esp-idf/blob/master/components/bt/bluedroid/stack/rfcomm/rfc_utils.c#L468) as proposed by Markus Becker, this issue seems less frequent and no more ESP_SPP_CONG_EVT events with a high throughput

More discussion about this problem, and workaround: https://www.esp32.com/viewtopic.php?t=5613

Wth-Esp, 2020-11-02:
Hi @copercini 

Please pull the latest IDF and have a try. We have merged two MR related to SPP flow control. It will take several days to sync on Github.

Thanks
Weitianhua

Alvin1Zhang, 2020-11-25:
Thanks for reporting and sorry for slow turnaround, the fix has been available at https://github.com/espressif/esp-idf/commit/8e4b919391eb5592f11438e76202265e6820abe0, feel free to reopen if the issue still happens. Thanks.

closed 2020-11-25 by Alvin1Zhang