espressif/esp-idf #2727: esptool error

pilnikov, 2018-11-22:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs on the monitor](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses) or the [coredump](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Provide more items under **Other items if possible** can help us better locate your problem.
5. Use markup (buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to repropduce

1. step1
2. ...

// It helps if you attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{
    
}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

projectgus, 2018-11-22:
Hi @pilnikov ,

Is this an open source project? If so, are you able to attach the sdkconfig and the KaRadio32.elf file from the build directory? If you can't share them it's OK, can figure out another approach.

igrr, 2018-11-22:
backlink https://github.com/karawin/Ka-Radio32/issues/65

pilnikov, 2018-11-22:
Yes of course. Its  karadio from https://github.com/karawin/Ka-Radio32
[1.ZIP](https://github.com/espressif/esp-idf/files/2607075/1.ZIP)

projectgus, 2018-11-22:
Thanks for posting those.

Something seems to have gone wrong in the linker script for both v3.1.1 and v3.2-beta1.

The ELF files both have hundreds of loadable sections when they should have about 5 (you can look at the output with `xtensa-esp32-elf-objdump -h ELFFILE`).

For v3.1.1, there are 254 loadable sections so the .bin file is generated - however the generated .bin file won't boot as the bootloader can only handle up to 16 loadable sections.

For v3.2, there are more than 255 loadable sections so the esptool error comes when this value is packed into a single byte.

I'll fix the error message to be more informative, but the problem is with the ELF file not the elf2binary step.

I can't reproduce the ELF problem with ESP-IDF examples or with the Ka-Radio32 project. Is it possible you've edited some of the linker script files in the IDF repository itself which is causing a bad link?

projectgus, 2018-11-23:
> Is it possible you've edited some of the linker script files in the IDF repository itself which is causing a bad link?

One other question, can you please post the output of `git describe --tags --dirty` when run in the IDF directory? Thanks.

pilnikov, 2018-11-23:
[image]

pilnikov, 2018-11-23:
> I can't reproduce the ELF problem with ESP-IDF examples or by building the Ka-Radio32 project locally. Is it possible you've edited some of the linker script files in the IDF repository itself which is causing a bad link?

Tru make without paremeter
e.g.
make -j4

pilnikov, 2018-11-23:
[image]

pilnikov, 2018-11-23:
after git pull & git submodule update
[image]

pilnikov, 2018-11-23:
[2.zip](https://github.com/espressif/esp-idf/files/2609806/2.zip)
Files after make all with updated 3.2 beta1
Size .elf = 6m

projectgus, 2018-11-23:
The binary seems to be building correctly now, so it appears some change on master branch has fixed this.

One thing I noticed in your screenshots is you are running MSYS2 via "msys.exe", not "mingw32.exe". This is not supported - if you look in the build output you'll see a warning about it. The root cause may be something to do with that, and using an MSYS rather than a MINGW32 Python. But I'm just guessing.

(BTW, I can't explain why your copy of git describes commit 4c881708d as being based on v3.2-beta1-182-g4c881708d. This commit is actually v3.3-dev, and if I clone from github on Windows it's described correctly. Strange! Maybe a `git fetch` will correct this.)

pilnikov, 2018-11-23:
[image]

projectgus, 2018-11-25:
Everything looks 100% correct now (both binary and version)! Are you experiencing any more issues with ESP-IDF?

closed 2018-11-27 by pilnikov