espressif/esp-idf #2728: WIFI+BT+BLE

akshar001, 2018-11-22:
Hello,
We are using BT+BLE+WIFI in one of our project,
BT is specifically used for SPP server to view log over BT.
But There is some RF calibration error showing when all runs at same time?

muratsplat, 2018-11-22:
@akshar001  Do you want to use BT+BLE at same time ? I don't know that they can work at same time.

TianHao-Espressif, 2018-11-23:
@akshar001 , could your describe the error with pasting a log?

akshar001, 2018-11-25:
@TianHao-Espressif i will post all of the things that we will need to analyze. @muratsplat espressif have mentioned that esp32 can run at both mode bt and wifi at the same time. Yes there might not have been achieved or might be but esp32 is capable of that! That's a beauty of it

Alvin1Zhang, 2018-12-03:
@akshar001 Could you help share any logs regarding this issue? Thanks.

Alvin1Zhang, 2019-03-15:
Could you help share any logs regarding this issue? Thanks.

akshar001, 2019-03-15:
Hello @Alvin1Zhang what type of log you want?

Alvin1Zhang, 2019-08-06:
@akshar001 Thanks for reporting the issue, will close due to short of feedback, feel free to reopen if you still have the issue, thanks.

closed 2019-08-06 by Alvin1Zhang