espressif/esp-idf #2732: Cannot connect after abnormal BLE disconnection

meriororen, 2018-11-23:
Hi, sorry I am not sure whether it is wise to open a new issue or just comment on #1185 so I decided to do both..

I can still reproduce the bug mentioned in #1185 with the latest SDK with gatts_demo example. Below is the patch for gatts_server example and the necessary client script (based on nodejs "noble" library).

[gatts_demo.patch.txt](https://github.com/espressif/esp-idf/files/2610084/gatts_demo.patch.txt)
[test_disconnect.js.txt](https://github.com/espressif/esp-idf/files/2610085/test_disconnect.js.txt)

Notes:

esp-idf release v3.1.1
put 'noble' directory with the same directory as 'test_disconnect.js'
https://github.com/noble/noble
run command (bash) :
for i in {1..1000}; do echo "repeat $i"; sudo ./test_disconnect.js <ble_mac_address>; sleep 1; done
In n-th (n sometimes as high as >100 but sometimes less than 20) times, the esp32 will have a TIMEOUT disconnect (0x8), and then failed to advertise (sometimes, advertisement present, but cannot connect).

Please advise..

If this is hard to resolve, can you suggest a way to recover from abnormal disconnection like this? reset the controller probably?

Here's (part of) the logs:
[gatts_log.txt](https://github.com/espressif/esp-idf/files/2610088/gatts_log.txt)
[client_log.txt](https://github.com/espressif/esp-idf/files/2610091/client_log.txt)

meriororen, 2018-11-23:
Original issue has been reopened.

closed 2018-11-23 by meriororen

Alvin1Zhang, 2018-11-23:
@meriororen Since the bug you mentioned could be reproduced for #513  , for consistency, I have reopened #513 for better tracking and will close this issue. Would you also paste the logs in the bottom of this issue to #513 ? Thanks.

meriororen, 2018-11-23:
Yes. Thank you very much @Alvin1Zhang