espressif/esp-idf #2736: Custom build dir failing

Elektrik1, 2018-11-24:
On latest commit, setting up BUILD_DIR_BASE custom build path in Makefile does no longer works and produces such error:

> AR <ANY_PATH>/xtensa-debug-module/libxtensa-debug-module.a
> make: *** No rule to make target '<ANY_PATH>/esp32/esp32.common.ld', needed by '<ANY_PATH>/<PROJECT_NAME>.elf'.  Stop.

Steps to reproduce, any example project, include BUILD_DIR_BASE :=<ANY_PATH> and then make.

Elektrik1, 2019-01-24:
No news ?

Elektrik1, 2019-02-03:
No news ? Build to external directory is still broken.

I have found commit https://github.com/espressif/esp-idf/commit/63411fc556fb2e71985db50dacbf42a6532b3993 broke this. Tried to altering CMakeLists.txt files, with no luck. 

Basically issue is that esp32/esp32.common.ld file is not generated on external build dir.

projectgus, 2019-02-03:
Hi @Elektrik1 ,

Sorry noone got back to you earlier about this. 

> Tried to altering CMakeLists.txt files, with no luck.

Are you using the GNU Make based build system (ie the [default setup steps in the Getting Started Guide](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html))? If so, the CMakeLists.txt files are not used. [CMake based build system](https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/) is only used if you run are running `cmake` or `idf.py` to start the build.

Assuming you're usingthe GNU make build system, I just attempted to reproduce this on Linux with latest master branch commit (d4d3ccf6d) two ways, but couldn't make it fail. I tried one time like this:

```
make BUILD_DIR_BASE=/path/to/other/build_dir
```

and a second time by editing the project template Makefile like this:

```
PROJECT_NAME := app-template

BUILD_DIR_BASE := /path/to/other/build_dir

include $(IDF_PATH)/make/project.mk
```

Both builds succeeded, including generation of the esp32.common.ld file.

Can you please provide all of the [environment details asked for in the issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md#environment), to give us some information to try and reproduce?

Also, please provide details of how you set `BUILD_DIR_BASE`, what build command you run, and the build output (output from `make V=1` would be the best, you may need to post this on an external site like https://gist.github.com or https://pastebin.com ). Plus any other details you think may be relevant.

Thanks,

Angus

Elektrik1, 2019-02-04:
Thank you for your response, I do use regular make (msys32) to make the project. Now I understand why CMakeList did not alter anything, thou that was the only file referenced to esp32.common.ld file I have found and tried to enter paths manually.

Package was gathered here: https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20181001.zip, used in Windows 10 x64 machine.

Logs after trying to compile hello_world project:

```
MINGW32 /home/esp-idf/examples/get-started/hello_world
$ make BUILD_DIR_BASE=D:/Test
....
CC /d/Test/xtensa-debug-module/eri.o
AR /d/Test/xtensa-debug-module/libxtensa-debug-module.a
make: *** No rule to make target '/d/Test/esp32/esp32.common.ld', needed by 'D:/Test/hello-world.elf'.  Stop.
```

Here is verbose of rebuild cycle: https://pastebin.com/6qUFfFx0
Here is verbose of clean build: https://gist.github.com/Elektrik1/49371385b7500ae1e0accf24eb8a7a7e

Thanks. Meanwhile, will try CMake one.

projectgus, 2019-02-04:
Thanks for the extra details.

Does it work if you do `make BUILD_DIR_BASE=/d/Test` ? (this would still be a bug, just a more subtle one).

Elektrik1, 2019-02-04:
Haha, it works fine with unix-like path, shame I didn't try it that way for months :)

```
MINGW32 /home/esp-idf/examples/get-started/hello_world
$ make BUILD_DIR_BASE=/d/Test
...
Generating esp32.common.ld
LD /d/Test/hello-world.elf
esptool.py v2.6
```

Thank you for support!

P.s. I think it's time to move all projects builds to CMake, will try to deal with all undefined references.

projectgus, 2019-02-04:
Glad you got it working.

Have reopened because this is still something that should be handled in the Make build system. I think probably it will be with an error message asking you to set the Unix-style path (as we can't otherwise remap a Make-level variable override), but this is still better than simply not working).

closed 2021-04-01 by espressif-bot