espressif/esp-idf #2744: Error (active)E0070:	incomplete type is not allowed

YordanYanakiev, 2018-11-26:
- [x ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x ] I have searched the issue tracker for a similar issue and not found a similar issue.

## Environment

- Development Kit:     ESP32-Wrover-Kit
- Kit version (for WroverKit): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

I am using Visual Studio 2017 to build projects for ESP32, anyway it keeps me warning me about the following possible error :
**STRUCT_AFIELD_A(char,1,XCHAL_CP1_SA_ALIGN,CS_SA_,cp1,XCHAL_CP1_SA_SIZE)**

Error (active)	E0070	incomplete type is not allowed	Sketch2	c:\Users\Yordan\AppData\Local\arduino15\packages\esp32\hardware\esp32\1.0.0\tools\sdk\include\esp32\xtensa\xtruntime-core-state.h	166	

and this reapeats for all of the follows :
STRUCT_AFIELD_A(char,1,XCHAL_CP0_SA_ALIGN,CS_SA_,cp0,XCHAL_CP0_SA_SIZE)
STRUCT_AFIELD_A(char,1,XCHAL_CP1_SA_ALIGN,CS_SA_,cp1,XCHAL_CP1_SA_SIZE)
STRUCT_AFIELD_A(char,1,XCHAL_CP2_SA_ALIGN,CS_SA_,cp2,XCHAL_CP2_SA_SIZE)
STRUCT_AFIELD_A(char,1,XCHAL_CP3_SA_ALIGN,CS_SA_,cp3,XCHAL_CP3_SA_SIZE)
STRUCT_AFIELD_A(char,1,XCHAL_CP4_SA_ALIGN,CS_SA_,cp4,XCHAL_CP4_SA_SIZE)
STRUCT_AFIELD_A(char,1,XCHAL_CP5_SA_ALIGN,CS_SA_,cp5,XCHAL_CP5_SA_SIZE)
STRUCT_AFIELD_A(char,1,XCHAL_CP6_SA_ALIGN,CS_SA_,cp6,XCHAL_CP6_SA_SIZE)
STRUCT_AFIELD_A(char,1,XCHAL_CP7_SA_ALIGN,CS_SA_,cp7,XCHAL_CP7_SA_SIZE)

The projects compile without problem, but this is sitting there all the time.
I am just posting it in just-in-case scenario.

### Expected Behavior
unknown
### Actual Behavior
unknown
### Steps to repropduce
1. Install Arduino Studio.
2. Install Visual Studio 2017.
3. install "VisualMicro" plugin. 
4. Try to build any project and check the Error list.

brianignacio5, 2020-06-09:
This seems more related to your Visual Studio configuration for C/C++ than anything else. 

We do not officially support Visual Studio, so you should raise the question to the VisualMicro team instead, but if is any help here is how we [configure C/C++ properties](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) in our [ESP-IDF Visual Studio Code extension](https://github.com/espressif/vscode-esp-idf-extension)

VisualMicro, 2020-12-07:
Hi, the Visual Studio intellisense is seperate to the build process for Arduino/ESP. The errors you have reported look like intellisense errors. It is possible to alter the combo box above the errors list to "Build Only". Then you will only see the more accurate build errors. Alternatively, above the "errors list" is another combo that can be set to Project Only, that will exclude the underlying sources from intellisense "errors list" reporting. 

Why? Visual Studio intellisense is not 100% perfect, it caters for projects pre and post build so it is more flexible but less accurate than the build process. 

tip: Visual Micro provides a "Verbose" output option on the "vMicro" menu gives the Arduino style build output that the Admins of this site will understand. 

For any other questions about Arduino in Visual Studio or to see if we can help resolve intellisense errors, please join our forum at https://www.visualmicro.com.forums.

Thank you.

closed 2021-02-03 by Alvin1Zhang