espressif/esp-idf #2745: Get PROJECT_NAME in sources and from build tool

pathob, 2018-11-26:
Hi,

I'm currently implementing updates over the air and noticed that it is not possible to use the configured `PROJECT_NAME` in the sources or from the build tool.

The application should be able to tell with which `PROJECT_NAME` it has been compiled with in order to be able to select the appropriate binary on a server. This could be done by adding a define using the `CFLAGS`.

And the build tool should be able to output the `PROJECT_NAME` without all the usual noise above:

```bash
Toolchain path: /path/to/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /path/to/esp-idf/requirements.txt are satisfied.
```

So only:

```bash
$ make project_name
my_project_name
```

I know that I also can grep the `PROJECT_NAME` from the Makefile / CMakeLists.txt but I would prefer a clean way for doing this.

And when already improving the usage of the `PROJECT_NAME`, I would suggest to define a default name for the case that `PROJECT_NAME` is not set. Currently this leads to the fact that no binary is build at all and that the esptool instruction is weird:

```bash
To flash all build output, run 'make flash' or:
python [...] 0x10000 /path/to/project/build/.bin [...]
```

I could try to contribute all that. Please let me know what you think about it.

igrr, 2018-11-26:
Hi @pathob, we do have an upcoming feature which will add some metadata to the application binary, such as PROJECT_NAME, version (custom value or from `git describe`), IDF version, compilation date/time and so on. It should appear on Github in the next few days.

pathob, 2018-11-26:
Hi @igrr , thank you, sounds great. What about the output of the `PROJECT_NAME` from the build tool? I need that to deploy the right binary (not bootloader, partition etc.).

igrr, 2018-11-26:
One way this can be done is by defining a custom target which will write $(PROJECT_NAME) into a file. Then you can get the contents of this file from your build tool. I think relying on specific output of 'make' is not very robust; even if we silence all the information messages in normal case, we might still print out warnings if misconfiguration is detected.

pathob, 2018-11-26:
Good point regarding the warnings, but I'm not convinced from writing the result to a file. Then I would suggest we also keep the information messages and everyone who want's to use the `PROJECT_NAME` has to use the custom target together with `tail` like so:

```
make project_name | tail -1
```

projectgus, 2018-11-26:
Hi @pathob,

If you're open to using CMake, another approach is to use the metadata files that CMake writes out:
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system-cmake.html#build-system-metadata
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system-cmake.html#flash-arguments

The `project_description.json` file contains `project_name` as well as a number of other project properties, and is intended for build scripts and IDEs to be able to parse this info. To use it in a shell script you'll need to parse the JSON, but this can be done with a one-liner in your favourite scripting language, for example:

```
python -c 'import json; print(json.load(open("build/project_description.json"))["project_name"])'
```

Or, if you install the [command line tool jq](https://stedolan.github.io/jq/), it's possible to do:

```
jq .project_name build/project_description.json
```

Unlike make standard output which can often vary due to `make` warnings and version/flavour inconsistencies, the CMake build metadata files are designed to be machine-parsed.

Depending on what you're using the project name for, the "flasher args" files may also work for your goals (see docs link earlier in this comment).

projectgus, 2018-12-03:
@pathob Great!

Will leave this issue open until the app metadata feature is merged to GitHub.

igrr, 2019-02-26:
App metadata feature has been added in 3b9cb25fe18c5a6ed64ddd6a1dc4d0ce0b6cdc2a with some follow-up tweaks in subsequent commits.

closed 2019-02-26 by igrr