espressif/esp-idf #2746: wifi: rx ampdu: duplicated addba request, potential compitability bug

AxelLin, 2018-11-26:
## Environment

- Module or chip used: 
    ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    ESP-IDF v3.3-dev-236-gfa59b1b1c
- Build System:         [Make|CMake]
    Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     [Windows|Linux|macOS]
   Linux
- Power Supply:         [USB|external 5V|external 3.3V|Battery]
   USB

## Problem Description

Testing with master tree, sometimes I got below output:
W (220437) wifi: rx ampdu: duplicated addba request, potential compitability bug
W (220438) wifi: discard, ba not setup

The device looks still working but I'm not sure if this is harmless.

My device is running as STA mode.
I'm testing mqtt using a while loop to keep publishing data to server.
But above message seems not related to mqtt at all, I also got the same message
when test http.

liuzfesp, 2018-11-26:
HI @AxelLin each addba request transmitted by the AP has a unique token ID, if ESP32 STA receive different addba request with the same token, it will accept the duplicated addba request and print this warning message. It's not a bug in ESP32, it may be a bug in AP side. Just ignore it.

BTW, could you let me know the type of the AP?  How frequency does this message is printed?

AxelLin, 2018-11-26:
The AP is Tenda W309R.
It seems happens at random time, I saw the message several times in a day.

Cell 01 - Address: C8:3A:35:27:70:38
          Channel:1
          Frequency:2.412 GHz (Channel 1)
          Quality=65/70  Signal level=-45 dBm  
          Encryption key:on
          ESSID:"Tenda_277038"
          Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                    24 Mb/s; 36 Mb/s; 54 Mb/s
          Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
          Mode:Master
          Extra:tsf=000000bc470b2a86
          Extra: Last beacon: 1908ms ago
          IE: Unknown: 000C54656E64615F323737303338
          IE: Unknown: 010882840B162430486C
          IE: Unknown: 030101
          IE: Unknown: 2A0100
          IE: Unknown: 2F0100
          IE: IEEE 802.11i/WPA2 Version 1 
              Group Cipher : CCMP
              Pairwise Ciphers (1) : CCMP
              Authentication Suites (1) : PSK 
          IE: Unknown: 32040C121860
          IE: Unknown: 2D1AFC181BFFFF000000000000000000000000000000000000000000
          IE: Unknown: 3D1601080400000000000000000000000000000000000000
          IE: Unknown: DD090010180203F02C0000
          IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00

liuzfesp, 2018-11-27:
@AxelLin, I think it's AP's bug, just ignore the log message. Currently, we don't have AP with same type, will purchase one and then double confirm my conclusion.

cyberman54, 2018-11-27:
@liuzfesp I may have similar problem. Code which already did run, does now run on error 

> W (6755) wifi: discard, ba not setup

since latest updates of esp32-idf and arduino-esp32:

```
I (711) wifi: wifi driver task: 3ffd4984, prio:23, stack:3584, core=0
I (1605) wifi: wifi firmware version: abbd416
I (1605) wifi: config NVS flash: enabled
I (1607) wifi: config nano formating: disabled
I (1628) wifi: Init dynamic tx buffer num: 32
I (1628) wifi: Init data frame dynamic rx buffer num: 10
I (1628) wifi: Init management frame dynamic rx buffer num: 10
I (1632) wifi: Init static rx buffer size: 1600
I (1636) wifi: Init static rx buffer num: 4
I (1640) wifi: Init dynamic rx buffer num: 10
I (1702) wifi: mode : sta (d8:a0:1d:60:af:b4) + softAP (d8:a0:1d:60:af:b5)
I (1704) wifi: Init max length of beacon: 752/752
I (1705) wifi: Init max length of beacon: 752/752
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
I (1725) wifi: mode : sta (d8:a0:1d:60:af:b4)
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 14 - AP_STOP
[I][ota.cpp:77] start_ota_update(): Trying to connect to MyHomeWifi
I (1866) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (2599) wifi: state: init -> auth (b0)
I (2603) wifi: state: auth -> assoc (0)
I (2610) wifi: state: assoc -> run (10)
I (2717) wifi: connected with MyHomeWifi, channel 1
I (2786) wifi: pm start, type: 1
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:386] _eventCallback(): STA IP: 192.168.8.246, MASK: 255.255.255.0, GW: 192.168.8.1
I (6747) wifi: state: run -> init (0)
I (6748) wifi: pm stop, total sleep time: 3690337 us / 3962505 us
I (6749) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[I][ota.cpp:82] start_ota_update(): Connected to MyHomeWifi
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:357] _eventCallback(): Reason: 8 - ASSOC_LEAVE
W (6755) wifi: discard, ba not setup
```

liuzfesp, 2018-11-28:
Hi @cyberman54, how frequency is this message displayed? 
It's not a bug, just debug log. We will change the log level for this message to debug level.

cyberman54, 2018-11-28:
@liuzfesp the message appears with my AccessPoint (Beagle Board with OpenWRT) maybe 20% each connect trial.

Alvin1Zhang, 2019-06-30:
@AxelLin Thanks for reporting the issue, feel free to reopen if it still exists. Thanks.

closed 2019-06-30 by Alvin1Zhang