espressif/esp-idf #2750: ESP32 IDF extend I2C buffer size

avgustine13, 2018-11-27:
Hi to everyone.
I have such a problem.
As I understand the I2C buffer (master mode) in ESP32 is limited to 32 bytes.
Actually I need to read 60 bytes from a sensor and I get only 32.
I need to increase it for more big amounts of data (up to 64 bytes or even 128).

How can I implement this?

Kind regards.
Ivan.

avgustine13, 2018-11-28:
Hello Chuck.
Can you give a source file?
I need this only for read function.

stickbreaker, 2018-11-28:
It part of the arduino-esp32
Here: https://github.com/espressif/arduino-esp32/blob/7e9afe8c5ed7b5bf29624a5cd6e07d431c027b97/cores/esp32/esp32-hal-i2c.h and https://github.com/espressif/arduino-esp32/blob/7e9afe8c5ed7b5bf29624a5cd6e07d431c027b97/cores/esp32/esp32-hal-i2c.c

avgustine13, 2018-11-30:
Hello Chuck.
Can you give me some your API functions order how to init i2c master for read/write operations.
(without using it in arduino's Wire)
I rewrite the lib under esp-idf format.

stickbreaker, 2018-11-30:
These are the important functions from esp32-hal-i2c.h:
```c++
struct i2c_struct_t;
typedef struct i2c_struct_t i2c_t;

i2c_t * i2cInit(uint8_t i2c_num, int8_t sda, int8_t scl, uint32_t clk_speed);
void i2cRelease(i2c_t *i2c); // free ISR, Free DQ, Power off peripheral clock.  Must call i2cInit() to recover
i2c_err_t i2cWrite(i2c_t * i2c, uint16_t address, uint8_t* buff, uint16_t size, bool sendStop, uint16_t timeOutMillis);
i2c_err_t i2cRead(i2c_t * i2c, uint16_t address, uint8_t* buff, uint16_t size, bool sendStop, uint16_t timeOutMillis, uint32_t *readCount);
i2c_err_t i2cFlush(i2c_t *i2c);
i2c_err_t i2cSetFrequency(i2c_t * i2c, uint32_t clk_speed);
uint32_t i2cGetFrequency(i2c_t * i2c);
uint32_t i2cGetStatus(i2c_t * i2c); // Status register of peripheral
```
you use it like this:
```c++
i2c_t * i2c = i2cInit(0,17,18,100000); // use the first I2c peripherial on sda=17,scl=18, at 100KHz
if(i2c != NULL){
  uint8_t buffer[500];
  uint16_t numRead;
  i2c_err_t err = i2cRead(i2c, 0x50, buffer, 290, true, 50, &numRead);
/*
// External Wire.h equivalent error Codes
typedef enum {
    I2C_ERROR_OK=0,  //success
    I2C_ERROR_DEV, // problem with interrupt or hardware
    I2C_ERROR_ACK, // slave did not acknowledge
    I2C_ERROR_TIMEOUT, // slave stretched SCL longer than timeout
    I2C_ERROR_BUS, // bus arbitration fault.
    I2C_ERROR_BUSY, // other Master on bus is active
    I2C_ERROR_MEMORY, // out of heap
    I2C_ERROR_CONTINUE, // ReSTART operation pending, need Write or Read with STOP==TRUE
    I2C_ERROR_NO_BEGIN // error code for Wire()
} i2c_err_t;
*/ 

  if (err == I2C_ERROR_OK){ // command successfully completed, data is in buffer
  } else { // problem, numRead contains count of bytes transfered, 
      // evaluate err for actual problem,
   }
// to use ReSTART i2c sequence With Write then Read
// using 24LC64 EEPROM to read 267 bytes at offset 0x01AB
  data[0] = 0x01;
  data[1] = 0xAB;
  err = i2cWrite(i2c, 0x50,data,2,false,50);
  if(err==I2C_ERROR_CONTINUE){ // ReSTART operation, everything is fine
   // write command has not actually been execute, it is queued until a read/write command with 
   // STOP== TRUE is encountered

    err = i2cRead(i2c,0x50,data,267,true,50,&numRead); // actually execute the queued Write, then this Read
    if(err == I2C_ERROR_OK){ // success, data now contains numRead(267) bytes
    } else { // numRead contains actual number or read byte(0..267)
        // evaluate err
    i2cFlush(i2c); // clear out pending data queue after error
    } 

// done with i2c
  i2cReleaase(i2c);
  i2c = null;
} else { // bus did not init, hardware fault
}
```

I think all you will have to do to make it directly compatible with IDF is to change pin allocations, and logging, delays

chuck

Alvin1Zhang, 2019-03-21:
@avgustine13 Hi, would you help share if any updates for the issue? Thanks.

avgustine13, 2019-03-21:
@Alvin1Zhang 
I' ve solved my problems completely moving to Arduino-ESP32 platform.
There I2C worked exactly fine for my tasks.

closed 2019-03-21 by Alvin1Zhang