espressif/esp-idf #2751: .gitignore modification request

IBeRyUS, 2018-11-27:
Can you add 
```
        tools/kconfig/conf-idf.exe
        tools/kconfig/mconf-idf.exe
```
those files to the .gitignore file? Those files compiling on pc and not part sdk download.

Changing and updating to stable release needs discarding this files all the time and needs to recompile again. 

Because of changing files sdk version always has --dirty flag. We can't distinguish is it our modified version or just .exe files version.

```
D:\esp\esp-idf>git status
HEAD detached at v3.1.1
Untracked files:
  (use "git add <file>..." to include in what will be committed)

        tools/kconfig/conf-idf.exe
        tools/kconfig/mconf-idf.exe

nothing added to commit but untracked files present (use "git add" to track)

D:\esp\esp-idf>
```

Thanks.

projectgus, 2018-11-28:
Thanks for letting us know about this, will fix.

closed 2018-11-28 by igrr

IBeRyUS, 2019-01-03:
this issue still exist on v3.1.2

projectgus, 2019-01-03:
> $ git tag --contains f781167
> v3.3-beta1

Looks like I didn't mark this fix for backporting, so it's fixed in master but not backported to releases. Have put in backports for it now, it should be in the next v3.1 release.

If you want to ignore them for now without modifying ESP-IDF's .gitignore, this can be done by adding them to the [global git excludes list](https://git-scm.com/docs/gitignore).