espressif/esp-idf #2752: atol function in ROM is bugged

X-Ryl669, 2018-11-27:
Using latest master, I've found this code that fails:
```
unsigned u = atol("3172936133"); 
if (u != 3172936133U)
   explode();
```
Since atol is supposed to return a long, it should work for any 32 bits number, but it returns `0x7FFFFFFF` instead (in fact I guess `atoi` and `atol` do the same, while the former is supposed to accept `signed integer`, and the later `long`)

I know there is probably nothing you can do about this (since it's in the ROM, it's not software), but at least there's an issue here so other will find it and not get crazy when their code is perfectly fine.
BTW, `strtoul` seems to work.

projectgus, 2018-11-28:
Hi @X-Ryl669,

I don't think this is a bug. On 32-bit systems like `xtensa-esp32-elf-gcc`, `long int` is a 32-bit signed integer. So atoi and atol are equivalent here.

`3172936133` is `0xbd1f29c5` so can only be represented correctly as an unsigned int (or unsigned long). 

Signed integer overflow is undefined behaviour in C, so atol() and atoi() are free to return whatever value they like if the argument overflows. I just tested with gcc on 32-bit x86 and glibc also returns -1 (unsigned value 0x7FFFFFFF) for atol("3172936133").

Using `strtoul` is the correct solution here.

X-Ryl669, 2018-11-28:
You're right. The 2 functions presence is misleading since they both perform the same.
When porting code (from 64 bits system), this is commonly found and works correctly for unsigned int (since long is 64 bits here).
So it's a bug that's not obvious to spot (took me some time to understand where it failed) since the same code was working fine on my computer.
Anyway, this is not a true issue, by itself. I've opened it so other could find it and have a solution.

closed 2018-11-28 by X-Ryl669